espressif/esp-idf #3032: Http Server

snahmad, 2019-02-06:
Is this Http Server better than mongoose. I have having issues with mongoose web server on ESP32 platform. Loading web files of total size=2.7 Mb from SD Card  to server web pages includes 20 files with images takes few seconds.

https://github.com/espressif/esp-idf/blob/abea9e4c02bb17e86298aec4e299780399e4789f/examples/protocols/http_server/file_serving/main/main.c

Alvin1Zhang, 2019-03-21:
@snahmad Thanks for reporting this issue.

mahavirj, 2019-03-21:
@snahmad Difficult to comment on this, but if you run into any issues with http server from IDF please feel to file new issue. Closing this one.

closed 2019-03-21 by mahavirj

snahmad, 2019-03-21:
ok, not using http server. Mongoose is working for me. Also mongoose is full networking library include TCP and UDP works on other platforms as well. better for our use case.