espressif/esp-idf #3033: HID : Set the device as a remote control

Marc-Aurele, 2019-02-07:
Hello,

I'm using the HID project example. All is working fine but my device seems to be recognized as a keyboard. I would like to know if it is possible to set it as a remote control so that my phone stop asking me to configure the physical keyboard.
If you have any idea, feel free to share.

Best regards,

Marc

Marc-Aurele, 2019-02-07:
I tried to use the following code : 
`                esp_bt_cod_t cod = {.major = 5, .minor = 3};
                esp_bt_gap_set_cod(cod, ESP_BT_SET_COD_MAJOR_MINOR);`

but without success. I'm still being asked to configure the keyboard when i pair the device to a phone ...

chegewara, 2019-02-07:
@Marc-Aurele You are mixing 2 things here, i am guessing that you are running ble hid example, but your second post is about classic bluetooth.

What you need to do is to change appearance in advertising.

Marc-Aurele, 2019-02-07:
Hello,

Thank you very much for the reply. Yes you are right, it tried the previous code because the solution you propose doesn't work. It tried different appearance but without any change. I don't know exactly what to put (i thought i should refer to this page [https://www.bluetooth.com/specifications/assigned-numbers/baseband](url) ) but every appearance make the device recognized as a keyboard.
I'm surely missing something.

chegewara, 2019-02-07:
Could you post link to hid example you are trying to use?

chegewara, 2019-02-07:
This is appearance table for ble:
https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.gap.appearance.xml

Marc-Aurele, 2019-02-07:
Here is the project example link i'm using : [https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/ble_hid_device_demo](url)

chegewara, 2019-02-07:
like i said, you are using ble hid example but referring to classic blutooth.

> I tried to use the following code :
> 
> ```
> esp_bt_cod_t cod = {.major = 5, .minor = 3};
> esp_bt_gap_set_cod(cod, ESP_BT_SET_COD_MAJOR_MINOR);
> ```
> but without success. I'm still being asked to configure the keyboard when i pair the device to a phone ...

This is setup for classic bluetooth.
Change this line:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/ble_hid_device_demo/main/ble_hidd_demo_main.c#L79

with 0x03c4 for example?

Marc-Aurele, 2019-02-07:
I removed the part of code related to classic bluetooth but i'm still having the issue even if i set the appearance with the following value : 0x03c4
What i have identified is that it is when i change the device name. If i use the following one : 

> #define DEVICE_NAME                    "HID"

All is working fine, but for every other device name, when i pair the device with my phone i can see the following on my phone : 

[image]

Marc-Aurele, 2019-02-08:
Hello,

As soon as, i want to change the device name with something different from "HID", i get this message on my phone : "Configure physical keyboard", whatever the value of the appearance ...
If anyone knows how to put his own device name without getting this message, feel free to share.

Regards,

Marc

Marc-Aurele, 2019-02-08:
Hi @chegewara
I think i have finally understood why this message is displayed on my phone. This is because of the hidReportMap. Indeed, if i remove the keyboard part, the message is no longer displayed. Actually, i would like to keep the "consumer devices" part but if i'm doing this, i can play or pause the music but "next track" and "previous track" become respectively up and down for the volume. 
 Do you have an idea of what i'm missing ?

Here is my report map : 

```
static const uint8_t hidReportMap[] = {
    //
    0x05, 0x0C,   // Usage Pg (Consumer Devices)
    0x09, 0x01,   // Usage (Consumer Control)
    0xA1, 0x01,   // Collection (Application)
    0x85, 0x03,   // Report Id (3)
    0x09, 0x02,   //   Usage (Numeric Key Pad)
    0xA1, 0x02,   //   Collection (Logical)
    0x05, 0x09,   //     Usage Pg (Button)
    0x19, 0x01,   //     Usage Min (Button 1)
    0x29, 0x0A,   //     Usage Max (Button 10)
    0x15, 0x01,   //     Logical Min (1)
    0x25, 0x0A,   //     Logical Max (10)
    0x75, 0x04,   //     Report Size (4)
    0x95, 0x01,   //     Report Count (1)
    0x81, 0x00,   //     Input (Data, Ary, Abs)
    0xC0,         //   End Collection
    0x05, 0x0C,   //   Usage Pg (Consumer Devices)
    0x09, 0x86,   //   Usage (Channel)
    0x15, 0xFF,   //   Logical Min (-1)
    0x25, 0x01,   //   Logical Max (1)
    0x75, 0x02,   //   Report Size (2)
    0x95, 0x01,   //   Report Count (1)
    0x81, 0x46,   //   Input (Data, Var, Rel, Null)
    0x09, 0xE9,   //   Usage (Volume Up)
    0x09, 0xEA,   //   Usage (Volume Down)
    0x15, 0x00,   //   Logical Min (0)
    0x75, 0x01,   //   Report Size (1)
    0x95, 0x02,   //   Report Count (2)
    0x81, 0x02,   //   Input (Data, Var, Abs)
    0x09, 0xE2,   //   Usage (Mute)
    0x09, 0x30,   //   Usage (Power)
    0x09, 0x83,   //   Usage (Recall Last)
    0x09, 0x81,   //   Usage (Assign Selection)
    0x09, 0xB0,   //   Usage (Play)
    0x09, 0xB1,   //   Usage (Pause)
    0x09, 0xB2,   //   Usage (Record)
    0x09, 0xB3,   //   Usage (Fast Forward)
    0x09, 0xB4,   //   Usage (Rewind)
    0x09, 0xB5,   //   Usage (Scan Next)
    0x09, 0xB6,   //   Usage (Scan Prev)
    0x09, 0xB7,   //   Usage (Stop)
    0x15, 0x01,   //   Logical Min (1)
    0x25, 0x0C,   //   Logical Max (12)
    0x75, 0x04,   //   Report Size (4)
    0x95, 0x01,   //   Report Count (1)
    0x81, 0x00,   //   Input (Data, Ary, Abs)
    0x09, 0x80,   //   Usage (Selection)
    0xA1, 0x02,   //   Collection (Logical)
    0x05, 0x09,   //     Usage Pg (Button)
    0x19, 0x01,   //     Usage Min (Button 1)
    0x29, 0x03,   //     Usage Max (Button 3)
    0x15, 0x01,   //     Logical Min (1)
    0x25, 0x03,   //     Logical Max (3)
    0x75, 0x02,   //     Report Size (2)
    0x81, 0x00,   //     Input (Data, Ary, Abs)
    0xC0,           //   End Collection
    0x81, 0x03,   //   Input (Const, Var, Abs)
    0xC0,            // End Collection
    0x06, 0xFF, 0xFF, // Usage Page(Vendor defined)
    0x09, 0xA5,       // Usage(Vendor Defined)
    0xA1, 0x01,       // Collection(Application)
    0x85, 0x04,   // Report Id (4)
    0x09, 0xA6,   // Usage(Vendor defined)
    0x09, 0xA9,   // Usage(Vendor defined)
    0x75, 0x08,   // Report Size
    0x95, 0x7F,   // Report Count = 127 Btyes
    0x91, 0x02,   // Output(Data, Variable, Absolute)
    0xC0,         // End Collection
};
```

chegewara, 2019-02-09:
Hi,
i didnt work for quite some time with hid, but i think you need to change a bit this report map.
I dont know if you need numeric keypad:
```
    0x85, 0x03,   // Report Id (3)
    0x09, 0x02,   //   Usage (Numeric Key Pad)
    0xA1, 0x02,   //   Collection (Logical)
    0x05, 0x09,   //     Usage Pg (Button)
    0x19, 0x01,   //     Usage Min (Button 1)
    0x29, 0x0A,   //     Usage Max (Button 10)
    0x15, 0x01,   //     Logical Min (1)
    0x25, 0x0A,   //     Logical Max (10)
    0x75, 0x04,   //     Report Size (4)
    0x95, 0x01,   //     Report Count (1)
    0x81, 0x00,   //     Input (Data, Ary, Abs)
    0xC0,         //   End Collection
```
If you do, i would make separate report id for keypad and for track control keys, easier to manage it, even if you have to have additional characteristic.
Also, im pretty sure this report map is broken and windows for example would not parse it (android would forgive it, dont know about iPhone). You have only 2 start collection and 4 end collection instructions. Just start with minimum report map, or try to find on internet exactly what you need.

Marc-Aurele, 2019-02-11:
Hi @chegewara,
I have followed your advises and make my HID report map as simple as possible by reading the documentation. I'm starting to understang every field in the HID map and thanks to that i finally come to this : 
```
    0x05, 0x0C,     /*      Usage Page (Consumer Devices) */
    0x09, 0x01,     /*      Usage (Consumer Control) */
    0xA1, 0x01,     /*      Collection (Application) */
    0x85, 0x01,     /*      Report ID=1 */
    0x05, 0x0C,     /*      Usage Page (Consumer Devices) */
    0x15, 0x00,     /*      Logical Minimum (0) */
    0x25, 0x01,     /*      Logical Maximum (1) */
    0x75, 0x01,     /*      Report Size (1) */
    0x95, 0x0B,     /*      Report Count (11) */
    
    0x09, 0xe2,     /*      USAGE (Mute)                0x01 */
    0x09, 0xe9,     /*      USAGE (Volume Up)           0x02 */
    0x09, 0xea,     /*      USAGE (Volume Down)         0x03 */
    0x09, 0xcd,     /*      USAGE (Play/Pause)          0x04 */
    0x09, 0xb7,     /*      USAGE (Stop)                0x05 */
    0x09, 0xb6,     /*      USAGE (Scan Previous Track) 0x06 */
    0x09, 0xb5,     /*      USAGE (Scan Next Track)     0x07 */
    0x09, 0xB1,     /*      Usage (Pause)               0x08 */
    0x09, 0xB0,     /*      Usage (Play)                0x09 */
    0x09, 0x83,     /*      Usage (Recall Last)         0x0A */
    0x09, 0x30,     /*      Usage (Power)               0x0B */
    0x81, 0x62,     /*      INPUT (Data,Var,Abs,NPrf,Null) */

    0x75, 0x01,     /*      Report Size (1) */
    0x95, 0x05,     /*      Report Count (5) */
    0x81, 0x01,     /*      Input (Constant) */
    0xC0,           /*      End Collection */
```
Every commands seems to work except the "Scan Previous Track" (6), "Scan Next Track" (7) and the "Play" (9)
I do not understand why they are not accepted in my case whereas with your example, all is working fine. I have also noticed that with my ubuntu, most of the commands are not taken into account. 

My purpose is to get what you have in your project example without the keyboard interface. I don't think it is mandatory to have the keyboard if i want all is working ?

I'm runing out of ideas about this issue, if you have any indication to give me, d'ont hesitate.

Regards,

Marc-Aurele, 2019-02-11:
I have the impression that i need to implement the keyboard report map if i want the "Scan Previous Track" and  the "Scan Next Track" working ...

Marc-Aurele, 2019-02-11:
Could my issue be related with this : https://www.hidglobal.fr/node/31461#xiaomi

Marc-Aurele, 2019-02-11:
Is this issue can be reopened pls, i made a mistake :(

closed 2019-02-13 by Marc-Aurele

Webunity, 2019-03-13:
@Marc-Aurele did you eventually got this to work?

Marc-Aurele, 2019-03-13:
Hello,

Not i didn't succeed to make it work without keyboard in the report map. I'm still having some weird behaviors with some phones when the report map doesn't embed a keyboard ...

Regards,

chegewara, 2019-03-15:
Hi @Marc-Aurele 
just for test i am using your report map and it works with my samsung 9+. Look at comments in report map:
```
0x05, 0x0C,     /*      Usage Page (Consumer Devices) */
    0x09, 0x01,     /*      Usage (Consumer Control) */
    0xA1, 0x01,     /*      Collection (Application) */
    0x85, 0x01,     /*      Report ID=1 */
    0x05, 0x0C,     /*      Usage Page (Consumer Devices) */
    0x15, 0x00,     /*      Logical Minimum (0) */
    0x25, 0x01,     /*      Logical Maximum (1) */
    0x75, 0x01,     /*      Report Size (1) */
    0x95, 0x0B,     /*      Report Count (11) */
    
    0x09, 0xe2,     /*      USAGE (Mute)                0x00 */
    0x09, 0xe9,     /*      USAGE (Volume Up)           0x01 */
    0x09, 0xea,     /*      USAGE (Volume Down)         0x02 */
    0x09, 0xcd,     /*      USAGE (Play/Pause)          0x03 */
    0x09, 0xb7,     /*      USAGE (Stop)                0x04 */
    0x09, 0xb6,     /*      USAGE (Scan Previous Track) 0x05 */
    0x09, 0xb5,     /*      USAGE (Scan Next Track)     0x06 */
    0x09, 0xB1,     /*      Usage (Pause)               0x07 */
    0x09, 0xB0,     /*      Usage (Play)                0x08 */
    0x09, 0x83,     /*      Usage (Recall Last)         0x09 */
    0x09, 0x30,     /*      Usage (Power)               0x0A */
    0x81, 0x62,     /*      INPUT (Data,Var,Abs,NPrf,Null) */

    0x75, 0x01,     /*      Report Size (1) */
    0x95, 0x05,     /*      Report Count (5) */
    0x81, 0x01,     /*      Input (Constant) */
    0xC0,           /*      End Collection */
```
Lets take play for example:
`    0x09, 0xB0,     /*      Usage (Play)                0x08 */`
0x08 means that you have to send value with bit 8 set. Because HID client expects 2 byte values in little endian order you have to send notification with value:
`{0x00, 0x01}`

Marc-Aurele, 2019-03-15:
Hello @chegewara ,

Thank you very much for your help. 

Finally, the report map attached in my previous was wrong as you noticed, here the last one i tested : 

```
    0x05, 0x0C,         //  Usage Page (Consumer Devices)
    0x09, 0x01,         //  Usage (Consumer Control)
    0xA1, 0x01,         //  Collection (Application)
    0x85, 0x01,         //      Report ID=1
    0x05, 0x0C,         //      Usage Page (Consumer Devices)
    0x15, 0x00,         //      Logical Minimum (0)
    0x25, 0x01,         //      Logical Maximum (1)
    0x75, 0x01,         //      Report Size (1)
    0x95, 0x10,         //      Report Count (16)
    0x09, 0xe2,         //      USAGE (Mute)                    00 01
    0x09, 0xe9,         //      USAGE (Volume Up)               00 02
    0x09, 0xea,         //      USAGE (Volume Down)             00 04
    0x09, 0xcd,         //      USAGE (Play/Pause)              00 08
    0x09, 0xb7,         //      USAGE (Stop)                    00 10
    0x09, 0xb6,         //      USAGE (Scan Previous Track)     00 20
    0x09, 0xb5,         //      USAGE (Scan Next Track)         00 40
    0x09, 0xCF,         //      USAGE (voice assist)            00 80
    0x09, 0xB8,         //      USAGE (Eject)                   01 00
    0x09, 0xB9,         //      USAGE (Random)                  02 00
    0x0a, 0x23, 0x02,   //      USAGE (www home)                04 00
    0x0a, 0x04, 0x02,   //      USAGE (www exit)                08 00
    0x0A, 0xAE, 0x01,   //      USAGE (AL Keyboard Layout)      10 00
    0x0a, 0x25, 0x02,   //      USAGE (www forward)             20 00
    0x0a, 0x24, 0x02,   //      USAGE (www back)                04 00
    0x0a, 0x21, 0x02,   //      USAGE (www search)              08 00
    0x81, 0x62,         //      INPUT (Data,Var,Abs,NPrf,Null)
    0xC0,               //  End Collection
```

But even with that one, all is working fine with a lot of phone but i have issues with my xiaomi mix 2 with next and previous that are replaced with volume bar (this is not volume up and down, the volume bar is just displayed). It is very weird. 
The only purpose to not have the keyboard in my report map is to not to be identified as a keyboard by the phone, as a result, the virtual keyboard is still display when it is needed on the phone.

Regards,

Aurélien

chegewara, 2019-03-15:
I can only guessing that it may be issue with xiaomi firmware.

Marc-Aurele, 2019-03-15:
So, you don't see any inconvenience to remove the keyboard from the report map ?

chegewara, 2019-03-15:
I am using only report map from you and nothing more:
```
0x05, 0x0C,     /*      Usage Page (Consumer Devices) */
    0x09, 0x01,     /*      Usage (Consumer Control) */
    0xA1, 0x01,     /*      Collection (Application) */
    0x85, 0x01,     /*      Report ID=1 */
    0x05, 0x0C,     /*      Usage Page (Consumer Devices) */
    0x15, 0x00,     /*      Logical Minimum (0) */
    0x25, 0x01,     /*      Logical Maximum (1) */
    0x75, 0x01,     /*      Report Size (1) */
    0x95, 0x0B,     /*      Report Count (11) */
    
    0x09, 0xe2,     /*      USAGE (Mute)                0x00 */
    0x09, 0xe9,     /*      USAGE (Volume Up)           0x01 */
    0x09, 0xea,     /*      USAGE (Volume Down)         0x02 */
    0x09, 0xcd,     /*      USAGE (Play/Pause)          0x03 */
    0x09, 0xb7,     /*      USAGE (Stop)                0x04 */
    0x09, 0xb6,     /*      USAGE (Scan Previous Track) 0x05 */
    0x09, 0xb5,     /*      USAGE (Scan Next Track)     0x06 */
    0x09, 0xB1,     /*      Usage (Pause)               0x07 */
    0x09, 0xB0,     /*      Usage (Play)                0x08 */
    0x09, 0x83,     /*      Usage (Recall Last)         0x09 */
    0x09, 0x30,     /*      Usage (Power)               0x0A */
    0x81, 0x62,     /*      INPUT (Data,Var,Abs,NPrf,Null) */

    0x75, 0x01,     /*      Report Size (1) */
    0x95, 0x05,     /*      Report Count (5) */
    0x81, 0x01,     /*      Input (Constant) */
    0xC0,           /*      End Collection */
```

Marc-Aurele, 2019-03-15:
Finally, if i remove the keyboard from the report map, i'm also having issues with Iphone. In my case, event such as AC Home are no longer interpreted if there is no keyboard in the report map. As a consequence, i can't launch a voice assist anymore without AC Home ...

chegewara, 2019-03-16:
@Marc-Aurele could you provide full report map you are using? It can be send on email.

chegewara, 2019-03-16:
Try with this value for AC Home in your report map, it works for me:
` 0x0a, 0x23, 0x02`

Marc-Aurele, 2019-03-18:
Hello @chegewara ,

> @Marc-Aurele could you provide full report map you are using? It can be send on email.

Here is the report i use and that is failing for some commands : 

```
static const uint8_t hidReportMap[] = {
    0x05, 0x0C,         //  Usage Page (Consumer Devices)
    0x09, 0x01,         //  Usage (Consumer Control)
    0xA1, 0x01,         //  Collection (Application)
    0x85, 0x01,         //      Report ID=2
    0x05, 0x0C,         //      Usage Page (Consumer Devices)
    0x15, 0x00,         //      Logical Minimum (0)
    0x25, 0x01,         //      Logical Maximum (1)
    0x75, 0x01,         //      Report Size (1)
    0x95, 0x10,         //      Report Count (16)
    0x09, 0xe2,         //      USAGE (Mute)                    00 01
    0x09, 0xe9,         //      USAGE (Volume Up)               00 02
    0x09, 0xea,         //      USAGE (Volume Down)             00 04
    0x09, 0xcd,         //      USAGE (Play/Pause)              00 08
    0x09, 0xb7,         //      USAGE (Stop)                    00 10
    0x09, 0xb6,         //      USAGE (Scan Previous Track)     00 20
    0x09, 0xb5,         //      USAGE (Scan Next Track)         00 40
    0x09, 0xCF,         //      USAGE (voice assist)            00 80
    0x09, 0xB1,         //      USAGE (Pause)                   01 00
    0x09, 0xB9,         //      USAGE (Random)                  02 00
    0x0a, 0x23, 0x02,   //      USAGE (www home)                04 00
    0x0a, 0x04, 0x02,   //      USAGE (www exit)                08 00
    0x0a, 0x26, 0x02,   //      USAGE (www stop)                10 00
    0x0a, 0x25, 0x02,   //      USAGE (www forward)             20 00
    0x0a, 0x24, 0x02,   //      USAGE (www back)                04 00
    0x0a, 0x21, 0x02,   //      USAGE (www search)              08 00
    0x81, 0x62,         //      INPUT (Data,Var,Abs,NPrf,Null)
    0xC0,               //  End Collection
};
```

> Try with this value for AC Home in your report map, it works for me:
> ` 0x0a, 0x23, 0x02`
> 
> PS this is great website to check report map
> https://eleccelerator.com/usbdescreqparser/

This is exactly the event i use for AC Home. My report map seems ok when i use the website you give. Indeed, this is a greatsite, i wish i could have known this site before.

Regards,

chegewara, 2020-05-10:
This topic is closed for some time now, so i put here only short clue about using AC and AL hid commands:
>Selector (Sel)
These Usages are Selectors, which belong to "Application Launch Buttons" (Usage: 0x0180)
> - AL Consumer Control Configuration
> - AL Desktop
This Usage is also a Selector, which belongs to "Generic GUI Application Controls" (Usage:  0x0200)
> - AC Exit
Selectors are defined in a separate Named Array collection.
Refer to "A.4 Named Array Field" section of Hut1_12v2.pdf

https://www.microchip.com/forums/m786291.aspx