espressif/esp-idf #3034: Errors in kkonfig

pilnikov, 2019-02-07:
after last update 

//First---------------------------------------------------------------------------------
file esp-idf/components/bt/Kconfig

error in line

  config BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN_EFF
            int
            default \
                BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN if BTDM_CONTROLLER_MODE_BR_EDR_ONLY || BTDM_CONTROLLER_MODE_BTDM
            default 0

        config BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN_EFF
            int
            default \
                BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN if BTDM_CONTROLLER_MODE_BR_EDR_ONLY || BTDM_CONTROLLER_MODE_BTDM
            default 0

//Second--------------------------------------------------------------------
file esp-idf/components/esp32/Kconfig

error in line

        range 0 27000 if ESP32_RTC_CLOCK_SOURCE_EXTERNAL_CRYSTAL || ESP32_RTC_CLOCK_SOURCE_EXTERNAL_OSC \
            || ESP32_RTC_CLOCK_SOURCE_INTERNAL_8MD256
        range 0 32766 if ESP32_RTC_CLOCK_SOURCE_INTERNAL_RC

pilnikov, 2019-02-08:
Z:\esp-idf\components\bt\Kconfig:63:warning: ignoring unsupported character '\'
Z:\esp-idf\components\bt\Kconfig:64: syntax error
Z:\esp-idf\components\bt\Kconfig:63: invalid option
Z:\esp-idf\components\bt\Kconfig:64: unknown option "BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN"
Z:\esp-idf\components\bt\Kconfig:69:warning: ignoring unsupported character '\'
Z:\esp-idf\components\bt\Kconfig:70: syntax error
Z:\esp-idf\components\bt\Kconfig:69: invalid option
Z:\esp-idf\components\bt\Kconfig:70: unknown option "BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN"
Z:\esp-idf\components\esp32\Kconfig:809:warning: ignoring unsupported character '\'
Z:\esp-idf\components\esp32\Kconfig:809:warning: ignoring unsupported character '|'
Z:\esp-idf\components\esp32\Kconfig:809:warning: ignoring unsupported character '|'
Z:\esp-idf\components\esp32\Kconfig:809:warning: ignoring unsupported character ' '
Z:\esp-idf\components\esp32\Kconfig:811: syntax error
Z:\esp-idf\components\esp32\Kconfig:810: unknown option "ESP32_RTC_CLOCK_SOURCE_INTERNAL_8MD256"

projectgus, 2019-02-08:
Hi @pilnikov ,

Thanks for reporting this. I think it's the same issue as #3012. 

A fix has been merged to our private repo but it hasn't been pushed to GitHub yet, I'll check on why.

pilnikov, 2019-02-08:
> I think it's the same issue as #3012.

Yes, it's true

closed 2019-02-08 by pilnikov