espressif/esp-idf #3035: Enabling UART 2 with UART 1 enabled breaks gettimeofday function assertion

menesesleonardo, 2019-02-07:
## Environment

- Module or chip used:  [ESP32-WROOM-32|ESP32-WROVER-I]
- IDF version:3.1.2
- Build System:         [Make]
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description
My application uses UART 1 to connect to a GSM module (simcom sim800l). This connection has the  following configuration:
-Bdr: 115200
-8bits
-no parity
-1bit stop
-hw control disable

When connection is made, time is updated via SNTP protocol and the device is connected to an MQTT broker. Until this part everything works fine (72hrs+).

however, the problem comes when a second UART interface is created to communicate with a GPS module with this configuration
-Bdr: 9600
-8bits
-no parity
-1bit stop
-hw control disable

continously, the time is requested and eventually (around 5min), an assertion is failed when MQTT task calls gettimeofday function.

MQTT task has nothing to do because I tried making a loop of requesting time every second and this fails after just 30 seconds.

## Debug Logs
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4008df2a  PS      : 0x00060930  A0      : 0x80085406  A1      : 0x3ffe59d0  
0x4008df2a: xQueueGenericReceive at /home/user/esp/esp-idf/components/freertos/queue.c:2037

A2      : 0x0a0d3937  A3      : 0x00000000  A4      : 0xffffffff  A5      : 0x00000000  
A6      : 0x00000002  A7      : 0x00000001  A8      : 0x8008e328  A9      : 0x3ffe59e0  
A10     : 0x00000003  A11     : 0x00060923  A12     : 0x00060920  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0a0d3977  LBEG    : 0x400978d0  LEND    : 0x400978db  LCOUNT  : 0x00000000  
0x400978d0: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:142

0x400978db: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:152

Backtrace: 0x4008df2a:0x3ffe59d0 0x40085403:0x3ffe5a10 0x400854d5:0x3ffe5a40 0x400e9629:0x3ffe5a60 0x40085231:0x3ffe5a80 0x40164c99:0x3ffe5aa0 0x40131879:0x3ffe5ac0 0x4012ffc5:0x3ffe5af0
0x4008df2a: xQueueGenericReceive at /home/user/esp/esp-idf/components/freertos/queue.c:2037

0x40085403: lock_acquire_generic at /home/user/esp/esp-idf/components/newlib/locks.c:157

0x400854d5: _lock_acquire at /home/user/esp/esp-idf/components/newlib/locks.c:165

0x400e9629: get_adjusted_boot_time at /home/user/esp/esp-idf/components/newlib/time.c:220

0x40085231: _gettimeofday_r at /home/user/esp/esp-idf/components/newlib/time.c:220

0x40164c99: gettimeofday at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/syscalls/../../../.././newlib/libc/syscalls/sysgettod.c:12

0x40131879: platform_tick_get_ms at /home/user/esp/projects/scooter/emot-scooter/components/espmqtt/lib/platform_esp32_idf.c:30

0x4012ffc5: esp_mqtt_task at /home/user/esp/projects/scooter/emot-scooter/components/espmqtt/mqtt_client.c:835

CPU halted.

```

## Other items

- sdkconfig file is provided

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2843022/sdkconfig.txt)

igrr, 2019-02-08:
It seems that you are facing heap corruption. A2=0x0a0d3937 was supposed to be a valid pointer, but in fact it looks like the end of a string buffer (0x37 0x39 CR LF), and it has overwritten part of the mutex structure. So I would say that somewhere in your application you are writing more data into a buffer than its size.

Please check the following section of the programming guide, it may help you debug such an issue. https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/heap_debug.html#finding-heap-corruption

menesesleonardo, 2019-02-10:
After 24hrs+ working I can say that solved it. Thanks a lot!

closed 2019-02-10 by menesesleonardo