espressif/esp-idf #3036: Compiler incorrectly uses lsi & ssi to load and store floats from/to IRAM addresses

fknrdcls, 2019-02-08:
## Environment

- Development Kit:   none
- Module or chip used:  ESP32-WROOM-32 & ESP32-WROVER
- IDF version: v3.3-beta1-179-ge931fe9f5
- Build System:         Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     macOS
- Power Supply:         USB || external 5V

## Problem Description

Any application code that tries to perform single precision floating point operations on IRAM (MALLOC_CAPS_32BIT) will crash with a load store error. 

### Expected Behavior

The floating point operation should succeed with a reduction in performance:
```cpp
        l32i.n a5, a2, 0         <----  Register is loaded with the correct instruction
	l32i.n a6, a2, 4   
	wfr    f2, a5            <----  Float is copied from register
	wfr    f3, a6      
	mul.s  f4, f3, f2  
	rfr    a7, f4      
	s32i.n a7, a2, 8   
```

### Actual Behavior

LoadStoreError occurs:
```cpp
        lsi	f1, a2, 0        <---- Core  X panic'ed (LoadStoreError). Exception was unhandled.
	lsi	f0, a2, 4
	mul.s	f0, f1, f0
	ssi	f0, a2, 8
```
### Steps to repropduce

1. Open fresh app template or hello_world example and run this code:

### Code to reproduce this issue

```cpp
    float *f;

    f = heap_caps_malloc(12, MALLOC_CAP_32BIT);  // MALLOC_CAP_32BIT
    f[0] = 1.5;
    f[1] = 2.2;
    f[2] = 0.0;
    // This works
    ESP_LOGI("BEFORE ASM: ", "f[0] %f, f[1] %f, f[2], %f : ptrs: %p, %p, %p", f[0], f[1], f[2], f, f+1, f+2);
    asm (\
        "#hand coded mul.s from/to IRAM\n\t"\
        "l32i.n a5, %0, 0   \n\t"\
        "l32i.n a6, %0, 4   \n\t"\
        "wfr    f2, a5      \n\t"\
        "wfr    f3, a6      \n\t"\
        "mul.s  f4, f3, f2  \n\t"\
        "rfr    a7, f4      \n\t"\
        "s32i.n a7, %0, 8   "\
        :    :  "a" (f) : "f2", "f3", "f4", "a5", "a6", "a7", "memory");
    ESP_LOGI("AFTER ASM: ", "f[0] %f, f[1] %f, f[2], %f : ptrs: %p, %p, %p", f[0], f[1], f[2], f, f+1, f+2);
    // This doesn't
    f[2] = 0.0;
    vTaskDelay(1000/portTICK_RATE_MS);
    f[0] = rand();
    if (f[0] > f[1]) f[0] = 4.5;    //Crash with LoadStoreError
    f[2] = f[1];                    //Crash with LoadStoreError
    f[2] = f[0] * f[1];             //Crash with LoadStoreError
```

## Debug Logs

```
Float using IRAM test
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
I (100) BEFORE ASM: : f[0] 1.500000, f[1] 2.200000, f[2], 0.000000 : ptrs: 0x4007801c, 0x40078020, 0x40078024
I (110) AFTER ASM: : f[0] 1.500000, f[1] 2.200000, f[2], 3.300000 : ptrs: 0x4007801c, 0x40078020, 0x40078024
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d1d49  PS      : 0x00060030  A0      : 0x800d0891  A1      : 0x3ffb3830  
0x400d1d49: app_main at /Users/amead/esp/iram_fpu_test/main/hello_world_main.c:62

A2      : 0x4007801c  A3      : 0x40078020  A4      : 0x40078024  A5      : 0x00000000  
A6      : 0x0000006e  A7      : 0x40533334  A8      : 0x800d1d46  A9      : 0x3ffb3810  
A10     : 0x5851f42d  A11     : 0xffffdeec  A12     : 0x3ffb0064  A13     : 0x00000000  
A14     : 0x00000003  A15     : 0x3ffb2120  SAR     : 0x0000001f  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40078020  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

Backtrace: 0x400d1d49:0x3ffb3830 0x400d088e:0x3ffb38a0
```

## Other items if possible
This behaviour may be expected to developers who have access to the ISA documentation. Unfortunately there isn't even one iota about the FPU in the ESP32 tech docs. Either the docs need updating or the compiler should be fixed.

fknrdcls, 2019-06-30:
I'm a professional technical writer - does anyone at espressif want me to help out with documentation?

krzychb, 2019-07-02:
Hi @fknrdcls yes, please mail me krzysztof 📫  espressif com

colonelwatch, 2021-08-22:
I noticed a while ago that this was one of the few places where this problem was documented, and I also really needed to store floats in the IRAM. I don't know any assembly and I don't know how to fix the compiler, but I came up with a decent workaround for the problem in C++. It is a class that can be used generally be used in place of a float but can also be stored in arrays allocated in IRAM with no errors. It uses volatile and reinterpret_cast to force floats to be represented as an unsigned int to the ESP32, and thus the compiler naturally uses l32i and s32i.

```c++
// iram_float_t
// A float that must be stored in IRAM as an integer. The result is forcing the compiler 
//  to use the l32i/s32i instructions (using other instructions causes a LoadStoreError).
// Original issue (and assembly solution): https://github.com/espressif/esp-idf/issues/3036
// This code is released into the public domain or under CC0

class iram_float_t{
    public:
        iram_float_t(float value = 0) 
            : _value(*reinterpret_cast<volatile unsigned int*>(&value)) {}

        void* operator new[] (size_t size){ // Forces allocation from IRAM
            return heap_caps_malloc(size, MALLOC_CAP_32BIT);
        }

        operator float() const {
            uint32_t a_raw = _value;
            return *reinterpret_cast<float*>(&a_raw);
        }
    private:
        volatile unsigned int _value;
};
```

I have a working example using this class in my project, [ESP32-fluid-simulation](https://github.com/colonelwatch/ESP32-fluid-simulation), though it's written slightly differently in this post because I'm not sure how Arduino code translates to esp-idf.