espressif/esp-idf #3037: [BLE Mesh] How to create mesh proxy node communication with GATT client and servers

marquesn, 2019-02-08:
## Environment

- Development Kit: ESP32 DevKitC
- Kit version: V4
- Module or chip used: ESP32-WROOM-32
- IDF version: v3.2-dev-525-g2f565635
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a5
- Operating System: Windows
- Power Supply: USB

## Problem Description

Hello, I am fairly new to BLE Mesh network development (and coming from the Arduino-ESP32 to the ESP-IDF environment). I've been doing some reading/experimenting with a few ESP32 devices that I have, and I want to implement a mesh network within my current setup, which is a simple communication between a GATT client (raspberry Pi), and two GATT servers (ESP32). A diagram of this is shown below, with the current setup that I have running on the left hand side, and the same one but with a BLE mesh on the right hand side:

[image]
 
I have already ran the ble_mesh_node example from the BLE mesh branch, and succeeded connecting two ESP32 as nodes by provisioning them with the Silicone Labs Mesh app. However, I am not sure how to integrate a GATT server and a GATT client into this Mesh?

And from what I undertood (although I maybe wrong) after reading the BLE Mesh documentation, the purpose of proxy nodes is to communicate with non mesh devices such as GATT clients and servers. But then more questions arise such as:
- Can a node be both a relay and a proxy node?
- Can a node be both a friend and a proxy node?
- Would I be able to have the Raspberry Pi start as a provisioner to provision the mesh, and once that is done, having it run as a GATT client in that same mesh in order to communicate with the GATT servers on the other end of the mesh?
- Is there any example code in the BLE mesh branch that establishes a GATT server and GATT client connection with proxy nodes that are already in a mesh?

I am also open to any suggestions on how this architecture might be changed in order to achieve the end goal of having BLE servers communicate with a BLE client on opposite ends of the mesh.

chegewara, 2019-02-08:
Hi,
proxy node purpose is to allow co-exist in the same mesh nodes that support GATT bearing and does not support advertising bearing. You cant connect regular (non mesh) GATT client/server to mesh. This is why you have provisioning, its security purpose, so only authenticated mesh nodes that are provisioned could participate in mesh network.

llymaximus, 2019-02-13:
1. A node can be both a relay and a proxy node.
2. A node can be both a friend and a proxy node.
3. You mean the node acting as a GATT Server is far away from the Provisioner? Using Raspberry Pi as a Provisioner and after it provision one BLE Mesh device successfully, it will scan the Mesh Node Identity/Network ID advertising packets which is nearby and Mesh Proxy connection can be created.
4. Currently only the BLE Mesh node can be a GATT Server after provisioned, and Provisioner only support communicate with the node using advertising bearer. APIs used by Provisioner to create connection with BLE Mesh nodes will be provided later.
5. Recommend to create connection with only one BLE Mesh Node nearby for Provisioner and this BLE Mesh node communicates with other BLE Mesh nodes using advertising bearer.

Campou, 2019-02-13:
* A node can be both a relay and a proxy node.

* A node can be both a friend and a proxy node.

* You mean the node acting as a GATT Server is far away from the Provisioner? Using Raspberry Pi as a Provisioner and after it provision one BLE Mesh device successfully, it will scan the Mesh Node Identity/Network ID advertising packets which is nearby and Mesh Proxy connection can be created.

* Currently only the BLE Mesh node can be a GATT Server after provisioned, and Provisioner only support communicate with the node using advertising bearer. APIs used by Provisioner to create connection with BLE Mesh nodes will be provided later.

* Recommend to create connection with only one BLE Mesh Node nearby for Provisioner and this BLE Mesh node communicates with other BLE Mesh nodes using advertising bearer.

Alvin1Zhang, 2019-03-14:
@marquesn Hi, could you help share if your issue has been resolved or not? Thanks.

marquesn, 2019-03-14:
@Alvin1Zhang Sort of... I've changed the architecture a bit to have the RPi using the meshctl tool from Bluez 5.50 as a provisioner (not GATT) for the other nodes. Although I haven't particularly achieved this yet, you may mark this issue as resolved, for the issues I'm having now might be considered different to the initial proposition of this thread.

Alvin1Zhang, 2019-03-14:
@marquesn Thanks, and feel free to raise a new issue if necessary, thanks.

closed 2019-03-14 by Alvin1Zhang