espressif/esp-idf #3039: cannot register level 5 interrupt

hoene, 2019-02-09:
## Environment

- Development Kit:      self made
- Module or chip used:  ESP32-WROOM-32
- IDF version:  v3.3-beta1-223-ga62cbfe
- Build System:         CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-73-ge28a011) 5.2.0
- Operating System:     Linux
- Power Supply:         external 5V

## Problem Description

I wrote my own level 5 assembly interrupt for RMI processing. However, I cannot register it with

esp_intr_alloc(ETS_RMT_INTR_SOURCE, ESP_INTR_FLAG_LEVEL5 | ESP_INTR_FLAG_IRAM, NULL, NULL, &isr_handle));
 
### Expected Behavior

My interrupt function starting with

    .section .iram1,"ax"
    .global     xt_highint5
    .type       xt_highint5,@function
    .align      4
xt_highint5:

should be called.

### Actual Behavior

However, it is not.

### Steps to reproduce

1. Write an assembly interrupt routine
2. Register it as above
3. Wait for it being called.

Dazza0, 2019-02-22:
`esp_intr_alloc()` is generally used to register C callable interrupts. However, high level interrupts are not supposed to be C callable in the Xtensa architecture thus cannot be registered using `esp_intr_alloc()`. To use the level 5 interrupt, you need override the weakly defined [default `xt_highint5`](https://github.com/espressif/esp-idf/blob/master/components/freertos/xtensa_vector_defaults.S#L107)  by first implementing your own `xt_highint5`, then adding a linker flag to ensure that your own implementation is linked in. See [High-Level Interrupts Guide](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/hlinterrupts.html) for more details.

hoene, 2019-02-22:
Hello Dazza0,

that is all clear, has been done, and is working. All that I am asking is to accept the pull request (https://github.com/espressif/esp-idf/pull/3040) so that 
 the function call to esp_intr_alloc(ETS_RMT_INTR_SOURCE, ESP_INTR_FLAG_LEVEL5 | ESP_INTR_FLAG_IRAM, NULL, NULL, &isr_handle));
is working.

Currently, it does not work.

Thank you

projectgus, 2019-03-01:
@hoene Yes, you're right - it should be possible to call esp_intr_alloc as shown to change the interrupt source. Have cherry-picked your PR (it had some accidental submodule changes in the commit so could not be merged as-is). Will merge ASAP.

hoene, 2019-03-01:
@projectgus Thank you!

closed 2019-03-08 by igrr