espressif/esp-idf #3041: ASSERT_PARAM(2051 0) in llc_util.c causes random application crashes during BLE scanning

pvachon, 2019-02-10:
## Environment

- Development Kit:  none (this is a custom board, but could reproduce the issue with a ESP32-WROOM-based board too)
- Module or chip used: `ESP32-D0WD`
- IDF version: `v3.3-dev-207-g78051e932`
- Build System: `Make`
- Compiler version: `xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0`
- Operating System: `Linux`
- Power Supply:  external 5V or USB

## Problem Description

With a BLE client application that periodically connects to multiple devices in parallel (between 3-6; when limiting the max devices to even 2, this issue shows up as well), my application occasionally panics during BLE device scanning, in code that is shipped pre-compiled with the SDK, with the following assertion:

```
ASSERT_PARAM(2051 0), in llc_util.c at line 289
```

The issue seems to be in the pre-compiled library used by the BT stack, so I can't provide much more in the way of diagnostics. The panic can happen within minutes of the device starting, or can take several hours to manifest.

I have disabled Bluetooth modem sleep, and have ensured the CPU core only runs at 240MHz to rule out any sort of power management or CPU scaling issues. The issue still is reproducible.

### Expected Behaviour
The BLE stack should not panic when scanning.

### Actual Behaviour
The application will panic after seemingly non-deterministic intervals, generally during BLE scans.

### Steps to reproduce

Run an application that periodically scans for BLE devices, connects to multiple devices in parallel, interrogates the GATT characteristics and services exposed by the devices, disconnects from the devices, then resumes scanning for BLE devices once all the devices have disconnected.

## Debug Logs

The actual panic trace is as follows (from `make monitor`):
```
ASSERT_PARAM(2051 0), in llc_util.c at line 289
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008429e  PS      : 0x00060b34  A0      : 0x80046350  A1      : 0x3ffe12e0  
0x4008429e: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffbdbf8  A7      : 0x60008050  A8      : 0x80084299  A9      : 0x3ffe12c0  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400841d1  LEND    : 0x400841d8  LCOUNT  : 0x00000000  
0x400841d1: r_assert_param at ??:?

0x400841d8: r_assert_param at ??:?

Backtrace: 0x4008429e:0x3ffe12e0 0x4004634d:0x3ffe1300 0x4010c195:0x3ffe1320 0x40019d11:0x3ffe1360 0x40055b4d:0x3ffe1380 0x401071df:0x3ffe13a0 0x401076f5:0x3ffe13c0
0x4008429e: r_assert_param at ??:?

0x4010c195: r_lld_evt_deffered_elt_handler at ??:?

0x401071df: r_rw_schedule at ??:?

0x401076f5: btdm_controller_task at ??:?
```

The GDB backtrace:

```
(gdb) bt
#0  0x4008429e in r_assert_param ()
#1  0x40046350 in ?? ()
#2  0x4010c198 in r_lld_evt_deffered_elt_handler ()
#3  0x40019d14 in ?? ()
#4  0x40055b50 in ?? ()
#5  0x401071e2 in r_rw_schedule ()
#6  0x401076f8 in btdm_controller_task ()
```

Weijian-Espressif, 2019-02-11:
@pvachon, We have fixed this bug, please update to the latest IDF and try, thanks.

pvachon, 2019-02-11:
@Weijian-Espressif I just tried the latest IDF, and my app eventually died with a crash in the BT ISR, due to the watchdog timing out:
```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40014f5c  PS      : 0x00060c34  A0      : 0x80083fd0  A1      : 0x3ffbe800  
A2      : 0x00000005  A3      : 0x68db8bad  A4      : 0x000004e2  A5      : 0x00000000  
A6      : 0x00005067  A7      : 0x04000000  A8      : 0xfffffffd  A9      : 0x00000005  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000578  A13     : 0xffffffff  
A14     : 0x00000002  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x40093e17  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40014f5c
0x40093e17: vTaskEnterCritical at /home/pvachon/esp/esp-idf/components/freertos/tasks.c:3537

Backtrace: 0x40014f5c:0x3ffbe800 0x40083fcd:0x3ffbe820 0x40086786:0x3ffbe850 0x40086252:0x3ffbe870 0x400875c5:0x3ffbe890 0x4008838f:0x3ffbe8b0 0x400826aa:0x3ffbe8d0 0x4000bfed:0x3ffba210 0x400d2c62:0x3ffba220 0x400d2c80:0x3ffba240 0x400d2d39:0x3ffba260 0x400942ac:0x3ffba280
0x40083fcd: r_ea_elt_insert at ??:?

0x40086786: r_lld_evt_end at ??:?

0x40086252: r_lld_evt_end_isr at ??:?

0x400875c5: r_rwble_isr at ??:?

0x4008838f: r_rwbtdm_isr_wrapper at intc.c:?

0x400826aa: _xt_lowint1 at /home/pvachon/esp/esp-idf/components/freertos/xtensa_vectors.S:1154

0x400d2c62: esp_pm_impl_idle_hook at /home/pvachon/esp/esp-idf/components/esp32/pm_esp32.c:452

0x400d2c80: esp_pm_impl_waiti at /home/pvachon/esp/esp-idf/components/esp32/pm_esp32.c:483

0x400d2d39: esp_vApplicationIdleHook at /home/pvachon/esp/esp-idf/components/esp32/freertos_hooks.c:86

0x400942ac: prvIdleTask at /home/pvachon/esp/esp-idf/components/freertos/tasks.c:3537

```

I did a clean build, updated the configuration for my app. I'm guessing this is unrelated to my prior crash (and can submit a separate bug report). However, this is very reliably reproducible.

pvachon, 2019-02-11:
Ahh, upon disabling `CONFIG_BTDM_CONTROLLER_FULL_SCAN_SUPPORTED`, the crash seems to have vanished. I'll leave this in testing for the day and let you know if my other crash is fixed.

Weijian-Espressif, 2019-02-12:
@pvachon, I will try to reproduce this issue. Please help test if `ASSERT_PARAM(2051 0)` is resolved, thanks.

pvachon, 2019-02-12:
@Weijian-Espressif I've been testing for 24 hours now with one of my devices using the new IDF, and it has not crashed yet. This is looking promising. I'm now testing with a batch of 4 other devices to validate the change -- if that works, I'll be happy to close this.

pvachon, 2019-02-13:
@Weijian-Espressif Unfortunately, after a bit over 36 hours of running, I ended up with the same panic.

Weijian-Espressif, 2019-02-15:
@pvachon,  do you reproduce  `ASSERT_PARAM(2051 0), in llc_util.c at line 289` ?

Weijian-Espressif, 2019-02-15:
@pvachon do you modify the `interrupt watchdog timeout` value in menuconfig file.  we set default value to 300ms.

pvachon, 2019-02-15:
@Weijian-Espressif I have been able to reproduce that exact assertion with the latest IDF. It took about 48 hours to manifest, but the backtrace is identical to the backtrace in the original issue (https://github.com/espressif/esp-idf/issues/3041#issue-408535930). However, it seems much more reliable than it was before (it used to crash after a few minutes, maybe an hour of use).

Also, I did not modify the interrupt watchdog timeout value. It is set to the default. I would prefer not to raise that limit though, it would seem strange an interrupt would take that long to complete handling.

pvachon, 2019-02-16:
@Weijian-Espressif Just confirming again -- this seems to take about 24-48 hours to manifest, but I consistently get the same assertion failure:
```
ASSERT_PARAM(2051 0), in llc_util.c at line 289
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
```

Weijian-Espressif, 2019-02-27:
Hi @pvachon ,
We have fixed the bug, this week the code will be merged into IDF.

pvachon, 2019-02-28:
Thank you! Let me know when it lands and I'll start testing.

arunbm123, 2019-03-03:
hello ESP Geeks 
Even I am facing some problems. I have 100 ibeacons. Esp32 scans for 10mins and crashes

0x4008405d: r_assert_param at ??:?

0x40084064: r_assert_param at ??:?

0x40084125: r_assert_param at ??:?

0x40123a5d: r_platform_reset at ??:?

0x4012c54b: r_llm_le_adv_report_ind at ??:?

0x4012ce49: r_llm_pdu_defer at ??:?

0x4012a48b: r_lld_pdu_check at ??:?

0x40128a8e: r_lld_evt_deffered_elt_handler at ??:?

0x40123dbf: r_rw_schedule at ??:?

0x401242c9: btdm_controller_task at ??:?

0x40093f79: vPortTaskWrapper at C:/msys32/home/renu/esp/esp-idf/components/freertos/port.c:403

Any insights ?

Weijian-Espressif, 2019-03-04:
@pvachon, please update to the latest IDF, master(9a1bbcfa1a5607c02d457e07cdce44a47bb2c23b), v3.2(b2e4af4ed07ef360e044ab693bfab8dd943e4bc4) or v3.1(43dc03659e6eb3d571c6f31cd339ecbdb9749bae) and try, we have fixed ASSERT_PARAM(2051 0) .

Weijian-Espressif, 2019-03-04:
@arunbm123,   controller memory may run out.
please enable duplicate scan , when you set `esp_ble_gap_set_scan_params()`
```
static esp_ble_scan_params_t ble_scan_params = {
    .scan_type              = BLE_SCAN_TYPE_ACTIVE,
    .own_addr_type          = BLE_ADDR_TYPE_PUBLIC,
    .scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
    .scan_interval          = 0x50,
    .scan_window            = 0x30,
    .scan_duplicate         = **BLE_SCAN_DUPLICATE_ENABLE**
};
```

pvachon, 2019-03-04:
@Weijian-Espressif I've started a test now, I'll let you know how the outcomes are.

I've also noticed, while having multiple device connections in parallel, that I run into errors when querying the GATT characteristics and services on the device. It seems the local GATT DB might get overwritten at some point? If I only enable one connection at a time, this issue goes away. Happy to open another issue if that's appropriate.

Weijian-Espressif, 2019-03-05:
@pvachon ,
Only one can be connected at a time

pvachon, 2019-03-06:
@Weijian-Espressif Understood, so this means that I can only connect to another device after I've identified the services and their attributes I'm interested in for the previous device?

Also, I'm finding this version much more unstable. I seem to see hangs when connecting to devices, I'll try to capture logs. However, this version seems to hang within several hours now, rather than running for 24-48 hours.

Alvin1Zhang, 2019-03-13:
@pvachon Hi, please help create a new ticket if you have new issue, thanks.

closed 2019-03-13 by Alvin1Zhang