espressif/esp-idf #3043: E (178) esp_core_dump: Failed to mmap core dump data (260)!

lucascoxBAF, 2019-02-11:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [external 3.3V]

## Problem Description

Whenever I try to enable coredump over uart, I get the following error:

E (178) esp_core_dump: Failed to mmap core dump data (260)!

I enabled core dump to uart from idf menuconfig, and I set the following partition table: 

```
# Name,   Type, SubType, Offset,  Size
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,           ,  0x6000
phy_init, data, phy,           ,  0x1000
factory,  app,  factory,       ,  1M
coredump, data, coredump,      ,  64K

```

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6288
ho 0 tail 12 room 4
load:0x40078000,len:11164
load:0x40080400,len:6672
entry 0x40080764
I (29) boot: ESP-IDF v3.3-beta1-326-gd7a7a6835 2nd stage bootloader
I (29) boot: compile time 12:53:20
I (29) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 2MB
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot:  3 coredump         Unknown data     01 03 00110000 00010000
I (89) boot: End of partition table
I (93) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x23af0 (146160) map
I (153) esp_image: segment 1: paddr=0x00033b18 vaddr=0x3ffbdb60 size=0x02e90 ( 11920) load
I (158) esp_image: segment 2: paddr=0x000369b0 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/lucas/Projects/rESPite/esp-idf/components/freertos/xtensa_vectors.S:1779

I (160) esp_image: segment 3: paddr=0x00036db8 vaddr=0x40080400 size=0x09258 ( 37464) load
I (184) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x7dc60 (515168) map
0x400d0018: _flash_cache_start at ??:?

I (365) esp_image: segment 5: paddr=0x000bdc80 vaddr=0x40089658 size=0x08bb0 ( 35760) load
0x40089658: r_lld_evt_end at ??:?

I (390) boot: Loaded app from partition at offset 0x10000
I (390) boot: Disabling RNG early entropy source...
I (391) cpu_start: Pro cpu up.
I (394) cpu_start: Application information:
I (399) cpu_start: Project name:     gatt_client_demo
I (405) cpu_start: App version:      v3.3-beta1-326-gd7a7a6835
I (411) cpu_start: Compile time:     12:53:18
I (416) cpu_start: Compile date:     Feb 11 2019
I (422) cpu_start: ESP-IDF:          v3.3-beta1-326-gd7a7a6835
I (428) cpu_start: Starting app cpu, entry point is 0x40081108
0x40081108: call_start_cpu1 at /home/lucas/Projects/rESPite/esp-idf/components/esp32/cpu_start.c:265

I (420) cpu_start: App cpu up.
I (439) heap_init: Initializing. RAM available for dynamic allocation:
I (446) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (452) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (458) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (464) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (470) heap_init: At 3FFC9D28 len 000162D8 (88 KiB): DRAM
I (476) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (483) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (489) heap_init: At 40092208 len 0000DDF8 (55 KiB): IRAM
I (495) cpu_start: Pro cpu start user code
I (178) esp_core_dump: Init core dump to UART
E (178) esp_core_dump: Failed to mmap core dump data (260)!
I (179) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)

https://gist.github.com/lucascoxBAF/4f4f4801746275bf68c992abdd27082d

gerekon, 2019-02-12:
Actually this is not a problem. 

This message just tells that there is no valid core dump data in flash. This has no any impact on proper operation of the device or core dump functionality. 

Possible solution is to remove this check when core dump is printed to UART and/or increase verbosity level of the message to warning.

gerekon, 2019-02-12:
We will hide this message in the next release of core dump.

lucascoxBAF, 2019-02-12:
@gerekon, good to know. Thanks!

closed 2019-02-12 by lucascoxBAF