espressif/esp-idf #3044: TAG V3.1 - esp_wifi_scan_start() in STA mode

caipiblack, 2019-02-12:
Hi,

We are using the TAG V3.1

When we call esp_wifi_scan_start() in STA mode (connected to an acces point) we get this error:
`Now is connecting, user scan invalid now!`

Is it possible to use this functions in STA mode (connected to an AP) ?

liuzfesp, 2019-02-15:
Hi @caipiblack, ESP32 supports two kinds of scan: the connecting scan which is triggered by esp_wifi_connect() and the user scan which is triggered by esp_wifi_scan_start(). The connecting scan has higher priority than the user scan, in other words, the esp_wifi_scan_start() will fail if the station now is connecting to the AP and following log will be displayed:

> W (167973) wifi: Now is connecting, user scan invalid now!

So it's expected behavior. 

The most simple way to handle this cause is to delay sometime and call esp_wifi_scan_start() again. However this method doesn't make sense in following scenario:

1.  esp_wifi_connect() is called and ESP32 station is connecting to an non-exist AP.
2. WiFi disconnected event is raised because the AP is not found, the WiFi event handler simply calls esp_wifi_connect() to reconnect the non-exist AP.
3. step 1/2 is repeated again and again.
4. esp_wifi_scan_start() is called.

If this is your case, you need to call esp_wifi_disconnect() to stop WiFi connection before esp_wifi_scan_start() is called, otherwise, esp_wifi_scan_start() will always fails.

For the return value ESP_ERR_WIFI_TIMEOUT, **I think it's not accurate,** we need to modify it, maybe it's better to return ESP_ERR_WIFI_FAIL etc.

You can refer to: [wifi_doc](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/wifi.html#scan-when-wi-fi-is-connecting) for more info about scan.

BTW, are you use the block scan: esp_wifi_scan_start(config, true)?

Alvin1Zhang, 2019-03-14:
@caipiblack Hi, would you help share if any updates for the issue? @liuzfesp how is the modification of **return ESP_ERR_WIFI_FAIL etc**? Thanks.

Alvin1Zhang, 2019-05-16:
@caipiblack Thanks for reporting the issue. Would you please help share if any updates for the issue? Has your issue been fixed? Thanks.

Alvin1Zhang, 2020-05-29:
Thanks for reporting, the fix is available at https://github.com/espressif/esp-idf/commit/c05a39de0280f896d7fcd7a8852d7e760e850393, feel free to reopen if the issue still happens. Thanks.

closed 2020-05-29 by Alvin1Zhang