espressif/esp-idf #3045: ASSERT_PARAM(64 0), in rwble.c at line 230

lucascoxBAF, 2019-02-12:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version: v3.3-beta1-328-gabea9e4c0
- Build System:         [CMake]
- Compiler version: crosstool-ng-1.22.0-73-ge28a011
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

I have an application that acts as both a ble central and peripheral, so it switches between advertising and scanning quickly. It also stands up a wifi access point. I randomly get "ASSERT_PARAM(64 0), in rwble.c at line 230." It always seems to happen while starting advertising. No reboot. No core dump. Just the error. Some of my application continues to run, but as you can see from the log below, there is a definite change after this assert. I know this is an issue that keeps popping up occasionally, but it's been pretty obnoxious to deal with. Especially since there is seemingly no way to trigger a reboot. 

## Debug Logs
```
I (4727630) Ble Driver: event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
I (4727630) Ble Driver: scan start success
I (4728630) Ble Driver: event: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT
I (4728630) Ble Driver: scan stop success
I (4728630) Esp Ble Server: advertisingParams: 32, 64, 7, 0.
I (4728630) Ble Driver: event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
I (4728640) Ble Driver: Advertising Succeeded!
I (4729630) Ble Driver: event: ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT
I (4729630) Ble Driver: Stop adv successfully

I (4729630) Ble Driver: event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
I (4729630) Ble Driver: scan start success
I (4730630) Ble Driver: event: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT
I (4730630) Ble Driver: scan stop success
I (4730630) Esp Ble Server: advertisingParams: 32, 64, 7, 0.
I (4730630) Ble Driver: event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
I (4730640) Ble Driver: Advertising Succeeded!
I (4731630) Ble Driver: event: ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT
I (4731630) Ble Driver: Stop adv successfully

I (4731630) Ble Driver: event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
I (4731630) Ble Driver: scan start success
I (4732630) Ble Driver: event: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT
I (4732630) Ble Driver: scan stop success
I (4732630) Esp Ble Server: advertisingParams: 32, 64, 7, 0.
I (4732630) Ble Driver: event: ESP_GASSERT_PARAM(64 0), in rwble.c at line 230
AP_BLE_ADV_START_COMPLETE_EVT
I (4732640) Ble Driver: Advertising Succeeded!
I (4738630) Esp Ble Server: advertisingParams: 32, 64, 7, 0.
I (4744630) Esp Ble Server: advertisingParams: 32, 64, 7, 0.
I (4750630) Esp Ble Server: advertisingParams: 32, 64, 7, 0.
I (4756630) Esp Ble Server: advertisingParams: 32, 64, 7, 0.
```

## Other items if possible

- [x] sdkconfig file: https://gist.github.com/lucascoxBAF/53d89497b224aea423c37fd895080e48

lucascoxBAF, 2019-02-12:
[image]

Above is a screenshot of the callstack when this error occurs. I hope it isn't a red herring. I've noticed sometimes the callstack isn't quite right.

Weijian-Espressif, 2019-02-13:
@lucascoxBAF  I have tested it for more than 5 hours according to your description and have not reproduced this issue. Can you provide your test code? So that we can solve this problem quickly, thanks.

Weijian-Espressif, 2019-02-13:
We can't open this `https://gist.github.com` in China. Can you provide the config file?

lucascoxBAF, 2019-02-13:
@Weijian-Espressif, The problem appears to be quite intermittent. Sometimes it happens quickly, other times it takes 12+ hours. 

Update. I changed my idf to version to v3.3-beta1 and got the following debug info and core dump. You can see that the core dump happens right after the ASSERT_PARAM(64 0) in the debug log. 

[moreErrorInfo.txt](https://github.com/espressif/esp-idf/files/2860782/moreErrorInfo.txt)

I'd love to be able to supply my code, but our source code is proprietary. Any idea if Espressif would be willing to sign an NDA? If so, I'm sure we'd be able to supply our source code to you. 

Here is the sdkconfig file
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2860781/sdkconfig.txt)

lucascoxBAF, 2019-02-13:
@Weijian-Espressif, my boss just reminded me that we actually do have an NDA signed with Espressif over the Homekit SDK, so I can send you our source code. Where would you like me to send our source code?

Weijian-Espressif, 2019-02-14:
Hi @lucascoxBAF  My mail address iszhiweijian@espressif.com,  please send me an email,  thanks.

Weijian-Espressif, 2019-02-15:
@lucascoxBAF,  please send the source code by email, my email is zhiweijian@espressif.com .

lucascoxBAF, 2019-02-15:
@Weijian-Espressif , thank you. Sorry, I forgot to send it yesterday. I'll bundle it and send it today.

no1seman, 2019-03-09:
Hi,

if there any ideas about: ASSERT_PARAM(131072 0), in rwble.c at line 230?
I'm facing the same error on v3.3-beta1-543-g40d23b6d5 IDF version. I've played with different parameters of SDKCONFIG, but still didn't find working combination.

Thanks in advance

lucascoxBAF, 2019-04-18:
@Weijian-Espressif , did you ever have any luck fixing this? I still run into it frequently. It's a very obnoxious error to deal with.

lucascoxBAF, 2019-04-18:
@no1seman, do you see the issue if you disable all logging except errors? Just curious.

no1seman, 2019-04-19:
@lucascoxBAF, I can't test it now, I'am away for some days, but problems with ble stack still peresent, it's a fact. In last tests i'm facing with the following errors:
ASSERT_PARAM(64 0), in rwble.c at line 23**5** and ASSERT_PARAM(9 10), in lld_pdu.c at line 527
I can't share the code, but the logic of programm is the foolowing:
advertise Eddystone TLM with a period 200 ms, and simulteneous BLE scan, all scanned data sends via mqtt client to mqtt server. I know that it's not good, but all work is going on in esp_gap_cb handler. So, if I set max frequency to 80 Mhz instead of 160Mhz in menuconfig or if using DFS mentioned above errors still happenning. On next week I'll try to investigate it more detail.

Weijian-Espressif, 2019-04-21:
@lucascoxBAF ,Sorry, I did not reproduce this issue with your test code. I have no idea why, where is the difference.

lucascoxBAF, 2019-04-22:
@Weijian-Espressif , The only difference I could think of would be that I have a very noisy environment. There are tons and tons of ble devices in my office.

no1seman, 2019-05-03:
@lucascoxBAF, @Weijian-Espressif ,
so... after a days of problem investigation I should say that i'm not facing any more with ASSERT_PARAM(64 0), in rwble.c at line 230 and ASSERT_PARAM(9 10), in lld_pdu.c at line 527 errors on latest master (v4.0-dev-464-ge7a7d216b). 
Here is my findings:
You can easily find that esp_gap_cb() handler is called from "Btc_task", so if don't wanna face with above problems you need the following:
1. Do not use any long operations in esp_gap_cb() switch-case
2. Do not intensively use stack i.e. don't call function which calls another function and so on
3. The best way to do is to copy esp_ble_gap_cb_param_t->scan_rst to queue and perform all needed calculations in a separate task
4. Do not use huge mallocs/callocs for queue especially in external PSRAM which too much expensive, for now I'm using static preallocated one
5. Do not use mutexes, delays and any other functions that may delay execution of code within esp_gap_cb() callback handler

As for ASSERT_PARAM(131072 0), in rwble.c at line 235 - i'll open a separate issue soon

lucascoxBAF, 2019-05-06:
@no1seman Thanks for the tips. Were you able to use your curl commands in #3415 to quickly reproduce the ASSERT_PARAM(64 0) before making changes 1-4? If so, that may be all you need to reproduce the issue, @Weijian-Espressif.

no1seman, 2019-05-06:
@lucascoxBAF,  #3415 is about smth other. If look in the code you'll find that BLE stack works fine until programm begins to call  uxTaskGetSystemState() and vice versa , if disable BLE - ps_handler works also fine without any errors. The problem is in BLE || tasks.h || my code (may be I'm doing smth in a wrong way). In any ways i'm not facing ASSERT_PARAM(64 0), in rwble.c at line 230 at that moment even in 'noisy' envioronment (I'm talking about latest master v4.0-dev-485-g537a477d6, not v3.3......).

FFtust, 2019-07-22:
Hi,
I have randomly  got the log "ASSERT_PARAM(1024 0), in rwble.c at line 230" too, any idea about this problem?

**Environment**:
    Module: [ESP32-WROOM] -- SPIRAM(4M)
    IDF version:   a18f92b09b2e87a3c0424d

**Problem Description**:
     It only cames when Bluetooth(run as a device) is adverstising,   following with a system restart, no backtrace.  But after bluetooth is connected, nothing happend.

axa88, 2019-08-22:
I am facing this issue as well.

```
ASSERT_PARAM(64 0), in rwble.c at line 235
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40085ad2  PS      : 0x00060034  A0      : 0x80089648  A1      : 0x3ffbe310  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffbdd48  A7      : 0x60008050  A8      : 0x80085acd  A9      : 0x3ffbe2f0  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40085a05  LEND    : 0x40085a0c  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x401af64a  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40085ad2

Backtrace: 0x40085ad2:0x3ffbe310 0x40089645:0x3ffbe330 0x4008a237:0x3ffbe350 0x400819e1:0x3ffbe370 0x401af647:0x3ffbbff0 0x40152ecf:0x3ffbc010 0x4008f8f1:0x3ffbc030 0x40090a45:0x3ffbc050

Core 1 register dump:
PC      : 0x40083a37  PS      : 0x00060d34  A0      : 0x80083796  A1      : 0x3ffca420  
A2      : 0xda8fb64a  A3      : 0x00000000  A4      : 0xa0000000  A5      : 0x00000000  
A6      : 0x00000020  A7      : 0x3ffc8580  A8      : 0x00000000  A9      : 0xda8fb661  
A10     : 0x00000000  A11     : 0xfa000000  A12     : 0x00000050  A13     : 0x3ffca300  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000142d  LEND    : 0x4000143a  LCOUNT  : 0xfffffffd  

Backtrace: 0x40083a37:0x3ffca420 0x40083793:0x3ffca440 0x40081453:0x3ffca460 0x400d3853:0x3ffca480 0x400d4754:0x3ffca4a0 0x400dbd01:0x3ffca4c0 0x40090a45:0x3ffca4e0
```

I don't like to confuse things add objective evidence, but my use also suggests that being in an very noisy environment may contribute to this issue.  To quantify 'noisy' as being in an electronics design test facility with hundreds of wireless devices in a similar range...
I have yet to see this issue at home where its very 'quiet'.

no1seman, 2019-08-22:
After upddating to: https://github.com/espressif/esp32-bt-lib/commit/717f0c6ec71a016a0b292acffeacf239d007b8ff I have't seen that error anymore

Alvin1Zhang, 2020-07-30:
Thanks for reporting and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/2f6f842d0f2a0e57541ab8a71a1ad5ba382a956c. Feel free to reopen if the issue still happens. Thanks.

closed 2020-07-30 by Alvin1Zhang