espressif/esp-idf #3047: BUG - v3..3 beta : esp_mesh_set_router doesn't work

leenowell, 2019-02-12:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|
- IDF version (run ``git describe --tags`` to find it): v3.3-beta1-223-ga62cbfe
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-73-ge28a011
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

Having started the mesh with an invalid SSID but valid password and channel, the mesh starts but naturally fails to connect to the router.   On the "no parent found" event, calling esp_mesh_set_router with the correct information the device fails to connect to the router.  Running this on IDF v3.1.2 the same code works fine and connects to the router. 

For further description I had logged this on the forum here too before establishing this was indeed a bug.

https://www.esp32.com/viewtopic.php?f=21&t=9098

Note - the call to esp_mesh_set_self_organized was included as per suggestion from the forum but actually doesn't make a difference.

### Expected Behavior
The node should connect to the router with the correct information

### Actual Behavior
The node fails to connect to the router

### Steps to repropduce

1. Start the mesh with an incorrect SSID
2. On no parent found event call esp_mesh_set_router with the correct values

### Code to reproduce this issue
In the on no parent event found handler add the following code but replacing the SSID and password with the correct values

```cpp
mesh_router_t newinfo;
strcpy((char*)newinfo.ssid, "XXXXXXX");
strcpy((char*)newinfo.password,"YYYYYY" );
newinfo.ssid_len = strlen((char*) newinfo.ssid); 
    
ESP_LOGI(HOMEIOT_TAG, "About to call set router org ssid[%s] password[%s] len[%d]", newinfo.ssid, newinfo.password, newinfo.ssid_len);

esp_err_t err = esp_mesh_set_router(&newinfo);
ESP_LOGI(HOMEIOT_TAG, "Finished set router err [%d = %s]", err, esp_err_to_name(err));
err = esp_mesh_set_self_organized(true,true);
ESP_LOGI(HOMEIOT_TAG, "Finished set self organised [%d = %s]", err, esp_err_to_name(err));

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

log running v3.3 beta
```
[0;32mI (22368) mesh_node: MeshNodeEvent: <MESH_EVENT_NO_PARENT_FOUND>scan times:60[0m

[0;32mI (22378) HomeIoT: About to call set router org ssid[XXXXXX] password[YYYYYY] len[6][0m

[0;32mI (22388) HomeIoT: Finished set router err [0 = ESP_OK][0m

I (22388) mesh: [IO]enable self-organizing, search parent<adaptive>

[0;32mI (22398) HomeIoT: Finished set self organised [0 = ESP_OK][0m

D (22408) event: SYSTEM_EVENT_SCAN_DONE, status:1, number:0[0m

I (22408) mesh: <SYSTEM_EVENT_SCAN_DONE>status:fail, num:0, id:188

[0;32mI (22418) wifi_node: WIFINodeEvent: Unknown event [1][0m

I (22408) mesh: <MESH_NWK_SCAN_REQ_PASSIVE>unexpected, stop previous scan by parent selection

I (22428) mesh: <MESH_NWK_SCAN_DONE>unexpected, flush scan results, request a new scan by MESH_NWK_LOOK_FOR_NETWORK

D (26648) event: SYSTEM_EVENT_SCAN_DONE, status:0, number:2[0m

I (26648) mesh: [FIND][ch:0]AP:2, otherID:0, MAP:0, idle:0, candidate:0, root:0[0e:fd:3f:0c:00:00]

[0;32mI (26648) wifi_node: WIFINodeEvent: Unknown event [1][0m

I (26648) mesh: [FIND:61]fail to find a network, channel:0, cfg<channel:1, router:XXXXXX, 0e:fd:3f:0c:00:00>

```
Log for v3.1.2
```
[0;32mI (45306) mesh_node: MeshNodeEvent: <MESH_EVENT_NO_PARENT_FOUND>scan times:0[0m

[0;32mI (45306) HomeIoT: About to call set router org ssid[XXXXXX] password[YYYYYY] len[6][0m

[0;32mI (45316) HomeIoT: Finished set router err [0 = ESP_OK][0m

I (45326) mesh: [IO]enable self-organizing, search parent<adaptive>

[0;32mI (45326) HomeIoT: Finished set self organised [0 = ESP_OK][0m

I (45336) mesh: <MESH_NWK_PARENT_SELECTION>scan_from_yield:0, connected_layer:0

D (45346) event: SYSTEM_EVENT_SCAN_DONE, status:1, number:0[0m

W (45346) wifi: scan number 0

[0;32mI (45346) wifi_node: WIFINodeEvent: Unknown event [1][0m

I (45356) mesh: [SCAN][ch:1]AP:0, otherID:0, MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:0,i:0]<>

I (45366) mesh: [FAIL][121]root:0, fail:121, normal:0, <pre>backoff:0

D (45676) event: SYSTEM_EVENT_SCAN_DONE, status:0, number:1[0m

I (45676) mesh: [SCAN][ch:1]AP:1, otherID:0, MAP:0, idle:0, candidate:1, root:0, topMAP:0[c:0,i:0]router found<>

[0;32mI (45676) wifi_node: WIFINodeEvent: Unknown event [1][0m

I (45676) mesh: 1232[SCAN]init rc[24:0a:c4:82:6f:1d,-48], mine:0, voter:0

I (45686) mesh: 1270, vote myself, router rssi:-48 > voted rc_rssi:-120

I (45696) mesh: [SCAN:1/10]rc[128][24:0a:c4:82:6f:1d,-48], self[24:0a:c4:82:6f:1c,-48,reason:0,votes:1,idle][mine:1,voter:1(1.00)percent:1.00][128,1,24:0a:c4:82:6f:1d]

```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

dongdong004, 2019-02-13:
Replied on ESP32 forum.

leenowell, 2019-02-23:
Bug confirmed on the forum together with a workaround if anyone else has this issue.  Link is here
https://www.esp32.com/viewtopic.php?f=21&t=9098

dongdong004, 2019-04-01:
The lastest IDF has modified this.

Alvin1Zhang, 2019-04-12:
@leenowell The fix has been merged, please try the latest IDF and see whether your issue has been resolved. Thanks.

Alvin1Zhang, 2019-05-16:
@leenowell Thanks for reporting the issue. Would you please help update if any updates for the issue? Has your issue been fixed? Thanks.

Alvin1Zhang, 2019-07-19:
@leenowell Thanks for reporting the issue. @dongdong004 Thanks for the fix. Feel free to reopen the issue if it still exists. Thanks.

closed 2019-07-19 by Alvin1Zhang

iottrends, 2021-09-05:
I am facing issues running the mesh/ip_internal_network code. I have two ESP32 NodeMCU v1.1. 
Both ESP are not communicating with each other..
I have taken the latest branch

---------------------------------------------------------
pi@raspberrypi:~/Desktop/work/esp/esp-idf/examples/mesh/ip_internal_network/main $ git show
commit d5f58ab13551cd883e8d8478ba367b6e4543ffec (HEAD -> master, origin/master, origin/HEAD)
Merge: 81391fb18c dd3d27a03c
Author: Mahavir Jain <mahavir@espressif.com>
Date:   Fri Sep 3 09:04:18 2021 +0000

    Merge branch 'feature/captive_portal' into 'master'

    Captive Portal Example

    Closes IDFGH-5347, IDFGH-1868, and IDF-1123

    See merge request espressif/esp-idf!14726

pi@raspberrypi:~/Desktop/work/esp/esp-idf/examples/mesh/ip_internal_network/main $
------------------------------------------------------------------------------------------------------
Logs
Node-1
I (840195) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:305
I (840195) mesh: <MESH_NWK_LOOK_FOR_NETWORK>no parent found, set ROOTLESS, scan times:60, heap:189748
I (840205) mesh: <MESH_NWK_LOOK_FOR_NETWORK>router<cfg:00:00:00:00:00:00> is allowed to switch, find routers with SSID:ROUTER_SSID
I (840205) mesh_main: <MESH_EVENT_NO_PARENT_FOUND>scan times:60
I (840225) mesh: <MESH_NWK_PARENT_DISCONNECTED>non-root, need_reselect:1
I (840235) mesh: <MESH_NWK_LOOK_FOR_NETWORK>channel<cfg:0> is allowed to switch, do a full channel scan
I (844145) mesh: [FIND][ch:0]AP:6, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]
I (844145) mesh: [FIND:305]fail to find a network, channel:0, cfg<channel:0, router:ROUTER_SSID, 00:00:00:00:00:00>
-----------------------------------------------------------------------------------------------------------------------------------
Node2 
I (1678865) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:609
I (1682785) mesh: [FIND][ch:0]AP:4, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]
I (1682785) mesh: [FIND:609]fail to find a network, channel:0, cfg<channel:0, router:ROUTER_SSID, 00:00:00:00:00:00>

I (1682795) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x1, look_for_nwk_count:610
I (1682805) mesh: <MESH_NWK_LOOK_FOR_NETWORK>no parent found, set ROOTLESS, scan times:60, heap:189604
I (1682815) mesh: <MESH_NWK_LOOK_FOR_NETWORK>router<cfg:00:00:00:00:00:00> is allowed to switch, find routers with SSID:ROUTER_SSID
I (1682815) mesh_main: <MESH_EVENT_NO_PARENT_FOUND>scan times:60
I (1682825) mesh: <MESH_NWK_PARENT_DISCONNECTED>non-root, need_reselect:1
I (1682835) mesh: <MESH_NWK_LOOK_FOR_NETWORK>channel<cfg:0> is allowed to switch, do a full channel scan
I (1684415) mesh: [FIND][ch:0]AP:0, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]<scan router>
I (1684415) mesh: [FIND:610]fail to find a network, channel:0, cfg<channel:0, router:ROUTER_SSID, 00:00:00:00:00:00>