espressif/esp-idf #3048: Abrupt reset of ESP-32 without backtrace  IDF 3-2 Beta IDF

jegajith, 2019-02-13:
Hi All I am getting following system reset , whenever **I (89416) wifi: lmac stop hw txq** this function triggered , Please advice how to fix this.  IDF Version  ESP-IDF v3.2-beta1-

----Log----
I (88396) APP_DEBUG: System Power Up, Days: 0 and Time: 05:46:59
I (88406) APP_DEBUG: Heap size ENDING OF THE CALL  39104
I (88406) APP_DEBUG: Turbine_Counter_Considered = 0 WaterUse.Todays_Gals= 0
I (88416) APP_DEBUG: Raw_Turbine_Counter_(both +ve and -ve edges) = 0 Turbine_Counter_With_(-ve edges only) = 0

I (89396) wifi: state: run -> init (0)
I (89396) wifi: pm stop, total sleep time: 68535854 us / 85784307 us

I (89396) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (89416) wifi: flush txq
I (89416) wifi: stop sw txq
E (89416) aws_iot: failed! mbedtls_ssl_write returned -0x4e
**I (89416) wifi: lmac stop hw txq**
I (11) boot: ESP-IDF v3.2-beta1-dirty 2nd stage bootloader
I (11) boot: compile time 09:49:38
I (11) boot: Enabling RNG early entropy source...
I (16) boot: SPI Speed      : 40MHz
I (20) boot: SPI Mode       : DIO
I (24) boot: SPI Flash Size : 4MB
I (28) boot: Partition Table:
I (31) boot: ## Label            Usage          Type ST Offset   Length
I (39) boot:  0 phy_init         RF data          01 01 00009000 00001000
I (46) boot:  1 aws_certs        unknown          41 00 0000a000 00001300
I (54) boot:  2 nvs              WiFi data        01 02 0000f000 0000f000
I (61) boot:  3 otadata          OTA data         01 00 0001e000 00002000

igrr, 2019-02-14:
@jegajith Are you redirecting console UART to non-default pins using menuconfig? In this case, would you be able to also capture UART output which appears between 

    **I (89416) wifi: lmac stop hw txq**

and 

    I (11) boot: ESP-IDF v3.2-beta1-dirty 2nd stage bootloader

on the standard UART pins?

jegajith, 2019-02-19:
Yes I am redirecting the UART Pins to  UART-1 for my Log .  Do you want me to Capture the Both Log during the crash ?? @igrr

jegajith, 2019-02-19:
If you see the log whenever this function calls i can see a crash in 3-2 Beta **I (89416) wifi: lmac stop hw txq**

igrr, 2019-02-26:
> Do you want me to Capture the Both Log during the crash ??

Yes, that would help, thank you.

Alvin1Zhang, 2019-05-16:
@jegajith Thanks for reporting the issue. Would you please help share if any updates for the issue? Thanks.

closed 2019-05-22 by jegajith