espressif/esp-idf #3049: [SD CARD] Writes strange addresses (offseted?!)

mtmsweb, 2019-02-13:
I have got a code that generates random buffer, sets its beginning to a phrase "sector ####" and writes buffer to next and next sectors. 
After writing it is checking, if buffer was written correctly.

When reading card on a computer, it turns out that on address 0, there is phrase "sector: 2048". Why sector address is offseted?! Test it with your cards

	char* buffer_write = heap_caps_malloc(s_card->csd.sector_size, MALLOC_CAP_DMA);
	char* buffer_read = heap_caps_malloc(s_card->csd.sector_size, MALLOC_CAP_DMA);

	esp_err_t ret;
	for (size_t sector = 2; ; sector++) {
		fill_buffer(sector, buffer_write, s_card->csd.sector_size);
		sprintf(buffer_write, "sector: %d / rand: ", sector);
		ret = sdmmc_write_sectors(s_card, buffer_write, sector, 1);
		if (ret != ESP_OK) { ESP_LOGE(TAG, "sdmmc_write_sectors returned rc=0x%x", ret); return; }

		ret = sdmmc_read_sectors(s_card, buffer_read, sector, 1);
		if (ret != ESP_OK) { ESP_LOGE(TAG, "sdmmc_read_sectors returned rc=0x%x", ret); return; }

		if (memcmp(buffer_write, buffer_read, s_card->csd.sector_size) == 0) {
			ESP_LOGI(TAG, "READ-WRITE sector %d : OK", sector);
		} else {
			ESP_LOGI(TAG, "READ-WRITE sector %d : ERROR", sector);
		}
	}

[Hex editor screen](https://i.imgur.com/QlCHzGZ.jpg)

Code was tested both in sdmmc and spi modes.
Function always result in READ-WRITE sector XX : OK (buffer read is the same as written). High sectors (it is not always 2048) are shifted, low not present when reading with hex editor on the computer!
[Question copy on esp32.com](https://esp32.com/viewtopic.php?f=13&t=9222&p=38483#p38483)

igrr, 2019-02-22:
@mtmsweb Have replied at https://esp32.com/viewtopic.php?f=13&t=9222&p=38483#p38537, please see if this addresses your issue.

Alvin1Zhang, 2019-05-20:
@mtmsweb Thanks for reporting the issue. Would you help share if any updates of the issue? Has your issue been fixed? Thanks.

igrr, 2019-06-20:
Assuming that the issue has been resolved, please reopen if you are still facing it.

closed 2019-06-20 by igrr