espressif/esp-idf #3050: BLE : How to expose 128bits UUID of a characteristic

Marc-Aurele, 2019-02-13:
Hello,

My characteristics expose only the first 16 bits instead of full UUID (128bits). Is there a way to expose the full UUID ?

```
static const uint16_t custom_svc        = 0xFF00;
static const uint16_t custom_str_uuid   = 0xFF01;
static const uint8_t custom_ccc[2]      = {0x00, 0x00};
static uint8_t  custom_str_value[10+1]  = "NULL";

#define GATTS_DEMO_CHAR_VAL_LEN_MAX 500

static const esp_gatts_attr_db_t custom_att_db[CUSTOM_IDX_NB] =
{
    /* Custom service Declaration */
    [CUSTOM_IDX_SVC]            = {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&primary_service_uuid, ESP_GATT_PERM_READ,
                                    sizeof(uint16_t), sizeof(custom_svc), (uint8_t *)&custom_svc}},

    /* Characteristic Declaration */
    [CUSTOM_IDX_CHAR]           = {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ,
                                    CHAR_DECLARATION_SIZE,CHAR_DECLARATION_SIZE, (uint8_t *)&char_prop_read_write_notify}},

    /* Characteristic Value */
    [CUSTOM_IDX_CHAR_VAL]       = {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&custom_str_uuid, ESP_GATT_PERM_WRITE|ESP_GATT_PERM_READ,
                                    GATTS_DEMO_CHAR_VAL_LEN_MAX, sizeof(custom_str_value), (uint8_t *)custom_str_value}},

    /* Client Characteristic Configuration Descriptor */
    [CUSTOM_IDX_CHAR_CFG]       = {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_client_config_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE,
                                    sizeof(uint16_t), sizeof(custom_ccc), (uint8_t *)custom_ccc}},
};
```

Regards,

Aurélien

chegewara, 2019-02-13:
Dou you want to have service with 128 bit UUID or advertise service with 128 bit UUID?

Marc-Aurele, 2019-02-13:
Hi @chegewara 
Once my device is paired, I want all my characteristics expose a 128 bits UUID if possible. For now, it seems my characteristics only expose a 16 bits UUID.
Regards,

chegewara, 2019-02-13:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_server_service_table/main/gatts_table_creat_demo.c#L92-L123

Weijian-Espressif, 2019-02-27:
@chegewara,
thanks for your help.
@Marc-Aurele ,
Has your problem been solved?

Marc-Aurele, 2019-02-27:
Hi all,

Thank you very much @chegewara for the proposition but i'm afraid to not understand exactly what you mean. Indeed, what i want is the following. Here are two devices : 

[image]

These two devices expose custum characteristics (red border)
Mine is on the left, but as you can see, it exposes only 16 bits uuid whereas the one on the right exposes the full 128 bits uuid.
Maybe i'm missing something in the link you attached to your previous post.

Regards,

chegewara, 2019-02-27:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_server_service_table/main/gatts_table_creat_demo.c#L103-L104

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_server_service_table/main/gatts_table_creat_demo.c#L120-L121

If you setup it with different UUIDs then you can advertise 2 different 128bit UUIDs at the same time.
Remember this option is to use with normal advertising, if you are using raw advertising then you need to build advertising packet byte by byte.

Marc-Aurele, 2019-03-04:
Thank you very much @chegewara ;)

closed 2019-03-13 by Alvin1Zhang

hcanak, 2021-03-02:
Hello,
I have a question about the same project (gatts_table_creat_demo.c). 
I want to "advertise" service with 128 bit UUID. In the example file line 85-89 defines 128bit UUID. See below;

_static uint8_t service_uuid[16] = {
    /* LSB <--------------------------------------------------------------------------------> MSB */
    //first uuid, 16bit, [12],[13] is the value
    0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0xFF, 0x00, 0x00, 0x00,
};_

When i compile and run the project it does NOT advertise above 128bit UUID instead it advertise 2 "0x00FF" services!!!. Please see attachment of LightBlue caps.
[image]

Now, my questions;

1) Although we supplied one 128bit UUID in line 85-89, we get responce of two 16bit UUIDs  (0x00FF). 
How can i get 128bit advertisement UUID?
2) When i change service_uuid[12] and service_uuid[13], it advertise that value as it said in the comment line 87. Why do we get two 16bit service UUIDs while we set only one 16bit UUID?

Thanks in advance.

xulongzhuang, 2021-04-06:
Hello, @hcanak , First, the two UUIDs are displayed because the broadcast data and scan response data are configured at the same time. You can comment out `esp_ble_gap_config_adv_data(&scan_rsp_data)` or `esp_ble_gap_config_scan_rsp_data_raw(raw_scan_rsp_data, sizeof(raw_scan_rsp_data))` to display only one service uuid. Second, if you use custom data to display 128-bit UUID, you should write this service ID to the `raw_scan_rsp_data` array, as follows：
```
static uint8_t raw_scan_rsp_data[] = {
        /* flags */
        0x02, 0x01, 0x06,
        /* tx power */
        0x02, 0x0a, 0xeb,
        /* service uuid */
        // 0x03, 0x03, 0xFF,0x00   // Replace this 16-bit UUID 
        0x11, 0x07, 0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0xFF, 0x00, 0x00, 0x00
};
```
It may not be defined in `raw_adv_data` because it exceeds 31 bytes with the device name and other AD structures.
If you want to use `esp_ble_gap_config_adv_data(...)` to configure 128-bit UUID, this method is complicated and you should go to the next level of tracking code. One possible way is to comment out the following code in [uuidType](https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gatt_util.c), 
```
int uuidType(unsigned char *p_uuid)
{
    int i = 0;
    int match = 0;
    int all_zero = 1;

    for (i = 0; i != 16; ++i) {
      // if (i == 12 || i == 13) {
      //      continue;
      //  }  // it will return LEN_UUID_128

        if (p_uuid[i] == BASE_UUID[i]) {
            ++match;
        }

        if (p_uuid[i] != 0) {
            all_zero = 0;
        }
    }
    if (all_zero) {
        return 0;
    }
    if (match == 12) {
        return LEN_UUID_32;
    }
    if (match == 14) {
        return LEN_UUID_16;
    }
    return LEN_UUID_128;
}
```
it will return the value of `LEN_UUID_128`,  Note that it cannot be defined together with the device name in the same `esp_ble_adv_data_t` structure

SlitheringMule, 2023-06-08:
Had to change `ESP_UUID_LEN_16` to `ESP_UUID_LEN_128` and set `static const uint8_t spp_service_uuid[16] = {0x9e,0xca,0xdc,0x24,0x0e,0xe5,0xa9,0xe0,0x93,0xf3,0xa3,0xb5,0x01,0x00,0x40,0x6e};`

HieuDAY1, 2024-12-14:
@chegewara Can you reshare the link?

chegewara, 2024-12-18:
https://github.com/espressif/esp-idf/blob/release/v3.3/examples/bluetooth/gatt_server_service_table/main/gatts_table_creat_demo.c#L92-L123

Im not sure where it is in master branch right now