espressif/esp-idf #3051: HID over GATT : report map is wrongly interpreted

Marc-Aurele, 2019-02-13:
I want to use HID over GATT on my ESP32 to handle a remote control : play/pause,scan next, scan previous, mute ... but it is not interpreted in the same way according the phone i use.

The report map i'm using is working very well with :

 - Huawei Tablet

 - Xiaomi Redmi note 5

 - An old Archos phone

It is wrongly interpreted with :

 - Xiaomi MIX 2 (the scan next and the scan previous are doing the volume up and down)

 - Ubuntu laptop : only volume up and down is working

It is not working at all with :

 - Windows labtop : HID is seen in error

I didn't test on Iphone

Here is the report map i'm using :

```
0x05, 0x0C,   // Usage Pg (Consumer Devices)
0x09, 0x01,   // Usage (Consumer Control)
0xA1, 0x01,   // Collection (Application)
0x85, 0x01,   // Report Id (1)

0x09, 0x02,   //   Usage (Numeric Key Pad)
0xA1, 0x02,   //   Collection (Logical)
0x05, 0x09,   //     Usage Pg (Button)
0x19, 0x01,   //     Usage Min (Button 1)
0x29, 0x0A,   //     Usage Max (Button 10)
0x15, 0x01,   //     Logical Min (1)
0x25, 0x0A,   //     Logical Max (10)
0x75, 0x04,   //     Report Size (4)
0x95, 0x01,   //     Report Count (1)
0x81, 0x00,   //     Input (Data, Ary, Abs)
0xC0,         //     End Collection

0x05, 0x0C,   //   Usage Pg (Consumer Devices)
0x09, 0x86,   //   Usage (Channel)
0x09, 0xE0,   //   Usage (Volume)
0x15, 0xFF,   //   Logical Min (-1)
0x25, 0x01,   //   Logical Max (1)
0x75, 0x02,   //   Report Size (2)
0x95, 0x02,   //   Report Count (2)
0x81, 0x46,   //   Input (Data, Var, Rel, Null)

0x09, 0xE2,   //   Usage (Mute)
0x09, 0x30,   //   Usage (Power)
0x09, 0x83,   //   Usage (Recall Last)
0x09, 0x81,   //   Usage (Assign Selection)
0x09, 0xB0,   //   Usage (Play)
0x09, 0xB1,   //   Usage (Pause)
0x09, 0xB2,   //   Usage (Record)
0x09, 0xB3,   //   Usage (Fast Forward)
0x09, 0xB4,   //   Usage (Rewind)
0x09, 0xB5,   //   Usage (Scan Next)
0x09, 0xB6,   //   Usage (Scan Prev)
0x09, 0xB7,   //   Usage (Stop)
0x15, 0x01,   //   Logical Min (1)
0x25, 0x0C,   //   Logical Max (12)
0x75, 0x04,   //   Report Size (4)
0x95, 0x01,   //   Report Count (1)
0x81, 0x00,   //   Input (Data, Ary, Abs)

0x09, 0x80,   //   Usage (Selection)
0xA1, 0x02,   //   Collection (Logical)
0x05, 0x09,   //     Usage Pg (Button)
0x19, 0x01,   //     Usage Min (Button 1)
0x29, 0x03,   //     Usage Max (Button 3)
0x15, 0x01,   //     Logical Min (1)
0x25, 0x03,   //     Logical Max (3)
0x75, 0x02,   //     Report Size (2)
0x95, 0x01,   //     Report Count (1)
0x81, 0x00,   //     Input (Data, Ary, Abs)
0xC0,         //     End Collection

0x75, 0x02,   //   Report Size (2)
0x95, 0x01,   //   Report Count (1)
0x81, 0x03,   //   Input (Const, Var, Abs)
0xC0,         //   End Collection
```

Do you see any error in this report map, it is inspired from the HID project example in esp-idf. I have taken a look at the HID usage table : https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf and compared with the example they give at the end but i didn't see anything wrong.

I would like my ESP32 can work with as many phones as possible and it would be great if i could make it work with my ubuntu and/or windows !

If you have any idea about my issue or if you know how i could debug it, don't hesitate to share.

Best regard,

heiye1996, 2019-02-14:
same issue here, tested on huawei p20 pro, mac book and some iphones, all works fine, but not on windows. somehow the hid example was removed from the latest release.

Marc-Aurele, 2019-02-14:
Hi @heiye1996 ,

We have tested it on Iphone, and nothing is working with these phones. Could please share your report map if it is possible or at least, if it is not possible tell me what are you embedding in this report map : mouse, keyboard, consumer control ..

Thank you in advance,

best regards,

Aurélien

Marc-Aurele, 2019-02-14:
@chegewara do you know a way to debug this on the phone to know how HID report map is interpreted ?

closed 2019-03-05 by Marc-Aurele