espressif/esp-idf #3052: esp_wifi_internal_tx randomly returns -16

marjon-halter, 2019-02-13:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit: |none]
- Kit version (for WroverKit/PicoKit/DevKitC): n/a
- Module or chip used:  [ESP32-WROVER]
- IDF version (run ``git describe --tags`` to find it): // v3.1-dev-322-gf307f41
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [external 5V]

## Problem Description

Recently the code started throwing the error code -16:
E (208931) NetInterface: Failed to tx buffer 0x3ffc7fc2, len 318, err -16

I have traced the error code to be "Illegal Argument" by searching around, but since the next layer down in the code is a library, I could not figure out why it was throwing that. The TCP/IP stack is from amazon-freertos port of the implementation.

What is weird is that, the error code is not easily reproducible. 
1. It sometimes happens quickly, sometimes takes a few minutes or maybe an hour or so.
2. Some hardware (we have several with the same components for each), have a harder time reproducing it than others.
3. Sometimes, the device recovers, sometimes it is just stuck there.

### Expected Behavior
Not throwing the error.

### Actual Behavior
Randomly throws error at different times, but error value is always "-16"

### Steps to repropduce
We compile and run our code on the device. Error is inconsistent.

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue
n/a

// the code should be wrapped in the ```cpp tag so that it will be displayed better.
n/a

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

mahavirj, 2019-02-15:
@marjon-halter Could you please provide more details on your use-case (since you mentioned that you are using FreeRTOS TCP stack)? Interim, please check that interface (either station or AP) is not down when you are enqueuing data using `esp_wifi_internal_tx`, that can result in this error.

liuzfesp, 2019-02-15:
@marjon-halter the -16 means the low-level netif is error, please check whether the WiFi is disconnected.

marjon-halter, 2019-02-15:
@liuzfesp, thank you. It would be nice if those error codes are part of the API interface. It is difficult to guess what is happening at the driver level since only the header was exposed. I will check.

marjon-halter, 2019-02-15:
@mahavirj , I was able to replicate the issue with the debugger attached. The following functions are the ones in the stack trace:
`xNetworkInterfaceOutput( pxNetworkBuffer, pdTRUE );` <-- line 228 at FreeRTOS_UDP_IP.C

Calls -->

`BaseType_t xNetworkInterfaceOutput( NetworkBufferDescriptor_t *const pxNetworkBuffer, BaseType_t xReleaseAfterSend )` <-- NetworkInterface.c

Calls --> wifi driver call to 
`esp_wifi_internal_tx()`

I think @liuzfesp  might be right, but the problem is that it is the FreeRTOS stack doing the call. How can I ensure that the wifi is connected prior to the stack doing its thing and sending data out the socket?

liuzfesp, 2019-02-15:
@marjon-halter if the application is based on sockets, all sockets related RX/TX in application should be called after the event SYSTEM_EVENT_STA_GOT_IP is raised.

marjon-halter, 2019-02-17:
As context, running the amazon-freertos version of FreeRTOS_IP/FreeRTOS_DHCP implementation.
More debug information on the item:
```C
I (88332) wifi: n:3 0, o:8 0, ap:255 255, sta:3 0, prof:1
I (89312) wifi: state: run -> auth (b0)  
I (90312) wifi: state: run -> init (2)  
I (90312) wifi: n:3 0, o:3 0, ap:255 255, sta:3 0, prof:1  
DEBUGGING: Socket established. prvSendDHCPRequest()  
E (92812) NetInterface: Failed to tx buffer 0x3ffc810e, len 325, err -16  
I (97872) wifi: n:3 0, o:3 0, ap:255 255, sta:3 0, prof:1  
I (98852) wifi: state: init -> auth (b0)  
I (98932) wifi: state: auth -> assoc (0)  
I (98952) wifi: state: assoc -> run (10)
W (98952) wifi: add mismatch
I (98972) wifi: connected with <accesspoint name>, channel 3
```
It seems the stack thinks it is already connected, and starts doing a DHCP operation. But the wifi driver has not yet reached the correct state, and does not know what to do with the data.

Is there an API for the wifi driver to read it's state (I am assuming you are running a state machine inside it).

marjon-halter, 2019-04-23:
Seems to have been fixed after amazon-freertos updated their copy of esp-idf to a newer revision.

closed 2019-04-23 by marjon-halter