espressif/esp-idf #3053: Menuconfig: BLE_SCAN_DUPLICATE, DUPLICATE_SCAN_CACHE_SIZE, BLE_MESH_SCAN_DUPLICATE_EN, MESH_DUPLICATE_SCAN_CACHE_SIZE

carlessole, 2019-02-14:
Hello,
First of all, sorry for my ignorance but I don't understand what this configuration parameters mean:

- BLE_SCAN_DUPLICATE
- DUPLICATE_SCAN_CACHE_SIZE
- BLE_MESH_SCAN_DUPLICATE_EN
- MESH_DUPLICATE_SCAN_CACHE_SIZE

Is there somewhere where it's explained with more detail? 
With the info found here (https://docs.espressif.com/projects/esp-idf/en/v3.0.7/api-reference/kconfig.html) I still don't undrestand.

Thanks

chegewara, 2019-02-15:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_gap_ble.html#_CPPv2N21esp_ble_scan_params_t14scan_duplicateE

This should be self explanatory(how many advertised device to  keep track):
* DUPLICATE_SCAN_CACHE_SIZE

Those 2 are related to ble mesh which is not implemented in master yet:
* BLE_MESH_SCAN_DUPLICATE_EN
* MESH_DUPLICATE_SCAN_CACHE_SIZE

Weijian-Espressif, 2019-03-01:
@chegewara, thanks.
@carlessole, Do you have any other questions?

closed 2019-03-13 by Alvin1Zhang