espressif/esp-idf #3054: Device does not find access point in WIFI_MODE_APSTA mode

AlexVilella, 2019-02-15:
Hi, 
I have a problem with esp32.
I have to connect a device (device A) to esp32 and at the same time esp32 has to connect to the network. The problem is that when I set the WIFI_MODE_APSTA mode, device A does not find the esp32 access point, but if I set it to WIFI_MODE_AP if it does.
I need device A to connect to the access point and esp32 to connect to the network.
I don't know if it's an incompatibility problem with device A or esp32.

liuzfesp, 2019-02-15:
Hi @AlexVilella what's your IDF commit ID? Does the esp32 AP has the same subnet with the network? Do you have the debug log?

Moreover, please help to do following test:
1.  Set esp32 to WIFI_MODE_APSTA
2. Connects esp32 station to network
3. esp32 station successfully gets the IP address, then connect the device A to esp32 AP

AlexVilella, 2019-02-15:
Hi again,

Thank you for the fast response and sorry due to forgot include this information.
- I'm using ESP IDF v3.1.1 release (3b92e85b308578c24f14f57371498303c198c7cb)

On my tests I did a process similar to what you are describing:
- Set the ESP32 to WIFI_MODE_APSTA
- Connect ESP32 station to network
Then:
- I'm able to see the AP from the smartphone (device B) and connect to the AP without problems and with an stable connection
- Device A is not able to see the AP
- Device C (Common WiFi USB Dongle) is not able to see the AP.

Thanks you in advance.

liuzfesp, 2019-02-16:
@AlexVilella is it possible for your do similar test in latest IDF master branch?

AlexVilella, 2019-02-18:
Hi again,
I'm downloaded the version 3.1.3 and get the following:
- Case 1: If I connect esp32 to the mode in APSTA mode. Device "Device A" does not connect to the access point.

- Case 2: If I create an access point on the smartphone, and esp32 connects it to the smartphone, then device "Device A" does connect to the esp32 access point.

Note: Device A is an electrical consumption analyzer (WiBeee Circutor Analyzer single-phase).

Does anyone know why this is happening?

Thanks you in advance.

AlexVilella, 2019-02-19:
It's all solved.
The problem was not of the configuration of the esp32, it was of the channel, above channel 10 the Wibeee was not able to find the network.

Thanks

closed 2019-02-19 by AlexVilella