espressif/esp-idf #3055: netif dhcps_pcb missing (used on dhcpserver)

daniol, 2019-02-15:
Component:  components\lwip\apps\dhcpserver\dhcpserver.c

The DHCP-Server uses the field dhcps_pcb of the struct netif, but it's not defined on components\lwip\lwip\src\include\lwip\netif.h, which causes a compilation error.

> C:\users\XXXX\.platformio\packages\framework-espidf\components\lwip\apps\dhcpserver\dhcpserver.c:1128:16: error: 'struct netif' has no member named 'dhcps_pcb'

It compiles if the following is added to the end of the struct:

```
#if LWIP_DHCP
   struct udp_pcb *dhcps_pcb;  
#endif
```

espxiehang, 2019-02-18:
@daniol  The code I saw didn't have this issue. 
What version of code did you use? Let me check it out.

Alvin1Zhang, 2019-02-22:
@daniol Hi, Daniol, would you help provide the version of code you used, so that we could help analyze, thanks.

daniol, 2019-02-27:
It seems to be missing on the master of esp-lwip.
https://github.com/espressif/esp-lwip/blob/master/src/include/lwip/netif.h

However, it is available on the branch 2.0.3-esp:
https://github.com/espressif/esp-lwip/blob/2.0.3-esp/src/include/lwip/netif.h

espxiehang, 2019-03-04:
@daniol   Thank you for pointing out this issue, we will deal with it as soon as possible.

espxiehang, 2019-03-04:
@daniol  hi，daniol， LWIP master branch  is not maintained by us.

Alvin1Zhang, 2019-03-04:
@daniol Feel free to reopen the issue, thanks.

closed 2019-03-04 by Alvin1Zhang