espressif/esp-idf #3056: send UDP broadcast messages

snahmad, 2019-02-15:
I need to send continuous UDP messages to IPv4 broadcast address : 255.255.255.255
Do I need specify port number.

Do I need to connect first.

Is this UDP client . Do I need UDP server.

Kindly give me example using LWIP.

Can multiple UDP application can received/monitor these messages via UDP.

liuzfesp, 2019-02-16:
@ESP_xiehang please help.

negativekelvin, 2019-02-16:
Don't ignore this, it's rude

> If your issue is a general question, starts similar to **"How do I.."**, or is related to 3rd party development kits/libs, please discuss this on our community **forum at esp32.com instead**.

snahmad, 2019-02-16:
ok, thanks.

espxiehang, 2019-02-20:
Do I need specify port number？ 
Yse.
Do I need to connect first?
No need.
Do I need UDP server？
No need.

These messages can be received as long as the application with the same port number.

snahmad, 2019-02-20:
ok, undertstood.

closed 2019-02-22 by Alvin1Zhang