espressif/esp-idf #3057: pm_esp32.c:426 IntegerDivideByZero

coollofty, 2019-02-16:
occasionally encountered once, then no longer encountered:

Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 0 register dump:
PC      : 0x400821e8  PS      : 0x00060031  A0      : 0x800824df  A1      : 0x3ffb0640
0x400821e8: update_ccompare at E:/esp32/esp-adf/esp-idf/components/esp32/pm_esp32.c:426

A2      : 0x00000003  A3      : 0x00060023  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00060021  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x02741217
A10     : 0x00000028  A11     : 0x7cf898f2  A12     : 0x00060021  A13     : 0x3ffd0d60
A14     : 0x00000003  A15     : 0x00060b23  SAR     : 0x00000013  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x400821e8  EPC2    : 0x400821dc  EPC3    : 0x00000000  EPC4    : 0x00000000
0x400821e8: update_ccompare at E:/esp32/esp-adf/esp-idf/components/esp32/pm_esp32.c:426

0x400821dc: update_ccompare at E:/esp32/esp-adf/esp-idf/components/esp32/pm_esp32.c:426

Backtrace: 0x400821e8:0x3ffb0640 0x400824dc:0x3ffb0660 0x40085934:0x3ffb0680 0x4019a6f7:0x00000000
0x400821e8: update_ccompare at E:/esp32/esp-adf/esp-idf/components/esp32/pm_esp32.c:426

0x400824dc: esp_pm_impl_isr_hook at E:/esp32/esp-adf/esp-idf/components/esp32/pm_esp32.c:462

0x40085934: _xt_lowint1 at E:/esp32/esp-adf/esp-idf/components/freertos/xtensa_vectors.S:1105

0x4019a6f7: esp_pm_impl_waiti at E:/esp32/esp-adf/esp-idf/components/esp32/pm_esp32.c:487

igrr, 2019-02-16:
Could you please attach your sdkconfig and mention the IDF commit ID?

coollofty, 2019-02-16:
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2872101/sdkconfig.zip)

The commit id: 5c88c5996dbde6208e3bec05abc21ff6cd822d26

Francois-Belanger, 2019-04-02:
Hi, just got the same issues.

Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 0 register dump:
PC      : 0x40081d1c  PS      : 0x00060031  A0      : 0x80081f97  A1      : 0x3ffb06a0  
0x40081d1c: update_ccompare at /Users/genie/ICE_Files/Firmware/esp/esp-idf/components/esp32/pm_esp32.c:426

A2      : 0x3ffb6314  A3      : 0x00060023  A4      : 0x80091296  A5      : 0x3ffb06a0  
A6      : 0x3ffb0ce4  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x03695baf  
A10     : 0x000000f0  A11     : 0xed062eaa  A12     : 0x80092fa3  A13     : 0x3ffb0670  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
Core 0 was running in ISR context:
EPC1    : 0x40081d1c  EPC2    : 0x40081d10  EPC3    : 0x00000000  EPC4    : 0x00000000
0x40081d1c: update_ccompare at /Users/genie/ICE_Files/Firmware/esp/esp-idf/components/esp32/pm_esp32.c:426

0x40081d10: update_ccompare at /Users/genie/ICE_Files/Firmware/esp/esp-idf/components/esp32/pm_esp32.c:426

ELF file SHA256: fe84065ad23e6f1d15c876269aab3920a7a1352a2de6695fd66fce7ad66e0bc2

Backtrace: 0x40081d1c:0x3ffb06a0 0x40081f94:0x3ffb06c0 0x40083247:0x3ffb06e0 0x400d2c9d:0x3ffcf4f0 0x400d3e4c:0x3ffcf510 0x40092d91:0x3ffcf530 0x400911b5:0x3ffcf550
0x40081d1c: update_ccompare at /Users/genie/ICE_Files/Firmware/esp/esp-idf/components/esp32/pm_esp32.c:426

0x40081f94: esp_pm_impl_isr_hook at /Users/genie/ICE_Files/Firmware/esp/esp-idf/components/esp32/pm_esp32.c:462

0x40083247: _xt_lowint1 at /Users/genie/ICE_Files/Firmware/esp/esp-idf/components/freertos/xtensa_vectors.S:1154

0x400d2c9d: esp_pm_impl_waiti at /Users/genie/ICE_Files/Firmware/esp/esp-idf/components/esp32/pm_esp32.c:478

0x400d3e4c: esp_vApplicationIdleHook at /Users/genie/ICE_Files/Firmware/esp/esp-idf/components/esp32/freertos_hooks.c:86

0x40092d91: prvIdleTask at /Users/genie/ICE_Files/Firmware/esp/esp-idf/components/freertos/tasks.c:3537

0x400911b5: vPortTaskWrapper at /Users/genie/ICE_Files/Firmware/esp/esp-idf/components/freertos/port.c:403

CPU halted.

igrr, 2019-04-03:
@Francois-Belanger It seems that the issue here is the one which is fixed by #3110 by adding a critical section, which prevents a level2 interrupt from occuring while a level1 interrupt is performing the frequency switch:
```
EPC1 : 0x40081d1c EPC2 : 0x40081d10 EPC3 : 0x00000000 EPC4 : 0x00000000
0x40081d1c: update_ccompare at /Users/ICE_Files/Firmware/esp/esp-idf/components/esp32/pm_esp32.c:426

0x40081d10: update_ccompare at /Users/ICE_Files/Firmware/esp/esp-idf/components/esp32/pm_esp32.c:426
```

Could you please try to apply the patch from #3110 and report if it fixes the issue you are observing?

Francois-Belanger, 2019-04-03:
Hi, I would prefer to wait for the fix to appear in the repo, it only occurred once for now.  When should it be available to the standard dev repo ?

Thanks

François Bélanger, Designer
Solution I.C.E.
450-577-3906

closed 2019-04-26 by igrr