espressif/esp-idf #3059: It is easy to encounter SDMC error

coollofty, 2019-02-16:
`E (5461) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (5461) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (5461) diskio_sdmmc: sdmmc_read_blocks failed (263)
`
my devboard is ESP32-Lyrat.
1. If I mount sdcard after audio library initialized, There will be a certain chance of encountering this error.
2. If I mount sdcard before audio library and then load a file(via fopen), then after a few seconds, when I fopen another file, it always report such a error

What will cause this error?

igrr, 2019-02-18:
Based on the description, two possibilities come to mind:

1) something in the process of initializing audio library reassigns one of the SD interface pins for a different function

2) initializing audio library powers up some extra components on the board (such as audio amplifier) which results in insufficient current available for the SD card, leading to a brownout

In both cases, you can approach troubleshooting by gradually removing some parts of audio library initialization, until the issue disappears. Whatever then makes the issue disappear may be related to cause of the issue.

Alternatively, if you can provide the code to reproduce this, we can run it on a LyraT board here and try to help you.

coollofty, 2019-02-19:
> Based on the description, two possibilities come to mind:
>
> 1. something in the process of initializing audio library reassigns one of the SD interface pins for a different function
> 2. initializing [...]

Yes, my mistake, other code makes pin conflict, thanks :)

Alvin1Zhang, 2019-02-22:
@coollofty Feel free to reopen if your issue still exists. Thanks.

closed 2019-02-22 by Alvin1Zhang