espressif/esp-idf #3060: CMake C++ Hello World

thepian, 2019-02-16:
## Environment

- Development Kit:      ESP32-LyraT
- Module or chip used: ESP32
- IDF version: v3.3-dev-337-g1cffc90e5
- Build System:        CMake
- Compiler version: GNU 5.2.0
- Operating System:    macOS
- Power Supply:  USB

## Problem Description

I'm trying to create a C++ based project.
I've copied hello world and renamed the main to .cpp (also tried .c++)
The regular make goes through fine, but `idf.py build` fails.
I tried deleting `build/main`, no dice.

### Expected Behavior

Build pass for simple case

### Actual Behavior

Fail with output below.

### Steps to repropduce

1. copy hello world example
2. rename main file to .cpp

## Debug Logs

Checking Python dependencies...
Python requirements from /Volumes/Projects/esp-idf/requirements.txt are satisfied.
Running cmake in directory /Volumes/Projects/thepia-all/TheBoot/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 --warn-uninitialized /Volumes/Projects/thepia-all/TheBoot"...
Warn about uninitialized values.
-- Found Git: /usr/local/bin/git (found version "2.11.0") 
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Volumes/Projects/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Volumes/Projects/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Volumes/Projects/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Volumes/Projects/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /Volumes/Projects/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building empty aws_iot component due to configuration
-- Component names: soc log heap xtensa-debug-module app_trace freertos vfs newlib esp_ringbuf driver esp_event ethernet lwip tcpip_adapter partition_table app_update spi_flash mbedtls micro-ecc bootloader_support nvs_flash pthread smartconfig_ack wpa_supplicant esp32 cxx asio jsmn aws_iot bootloader bt coap console nghttp esp-tls esp_adc_cal tcp_transport esp_http_client esp_http_server esp_https_ota openssl esp_https_server esptool_py expat wear_levelling sdmmc fatfs freemodbus idf_test json libsodium mdns mqtt protobuf-c protocomm spiffs ulp unity wifi_provisioning main
-- Component paths: /Volumes/Projects/esp-idf/components/soc;/Volumes/Projects/esp-idf/components/log;/Volumes/Projects/esp-idf/components/heap;/Volumes/Projects/esp-idf/components/xtensa-debug-module;/Volumes/Projects/esp-idf/components/app_trace;/Volumes/Projects/esp-idf/components/freertos;/Volumes/Projects/esp-idf/components/vfs;/Volumes/Projects/esp-idf/components/newlib;/Volumes/Projects/esp-idf/components/esp_ringbuf;/Volumes/Projects/esp-idf/components/driver;/Volumes/Projects/esp-idf/components/esp_event;/Volumes/Projects/esp-idf/components/ethernet;/Volumes/Projects/esp-idf/components/lwip;/Volumes/Projects/esp-idf/components/tcpip_adapter;/Volumes/Projects/esp-idf/components/partition_table;/Volumes/Projects/esp-idf/components/app_update;/Volumes/Projects/esp-idf/components/spi_flash;/Volumes/Projects/esp-idf/components/mbedtls;/Volumes/Projects/esp-idf/components/micro-ecc;/Volumes/Projects/esp-idf/components/bootloader_support;/Volumes/Projects/esp-idf/components/nvs_flash;/Volumes/Projects/esp-idf/components/pthread;/Volumes/Projects/esp-idf/components/smartconfig_ack;/Volumes/Projects/esp-idf/components/wpa_supplicant;/Volumes/Projects/esp-idf/components/esp32;/Volumes/Projects/esp-idf/components/cxx;/Volumes/Projects/esp-idf/components/asio;/Volumes/Projects/esp-idf/components/jsmn;/Volumes/Projects/esp-idf/components/aws_iot;/Volumes/Projects/esp-idf/components/bootloader;/Volumes/Projects/esp-idf/components/bt;/Volumes/Projects/esp-idf/components/coap;/Volumes/Projects/esp-idf/components/console;/Volumes/Projects/esp-idf/components/nghttp;/Volumes/Projects/esp-idf/components/esp-tls;/Volumes/Projects/esp-idf/components/esp_adc_cal;/Volumes/Projects/esp-idf/components/tcp_transport;/Volumes/Projects/esp-idf/components/esp_http_client;/Volumes/Projects/esp-idf/components/esp_http_server;/Volumes/Projects/esp-idf/components/esp_https_ota;/Volumes/Projects/esp-idf/components/openssl;/Volumes/Projects/esp-idf/components/esp_https_server;/Volumes/Projects/esp-idf/components/esptool_py;/Volumes/Projects/esp-idf/components/expat;/Volumes/Projects/esp-idf/components/wear_levelling;/Volumes/Projects/esp-idf/components/sdmmc;/Volumes/Projects/esp-idf/components/fatfs;/Volumes/Projects/esp-idf/components/freemodbus;/Volumes/Projects/esp-idf/components/idf_test;/Volumes/Projects/esp-idf/components/json;/Volumes/Projects/esp-idf/components/libsodium;/Volumes/Projects/esp-idf/components/mdns;/Volumes/Projects/esp-idf/components/mqtt;/Volumes/Projects/esp-idf/components/protobuf-c;/Volumes/Projects/esp-idf/components/protocomm;/Volumes/Projects/esp-idf/components/spiffs;/Volumes/Projects/esp-idf/components/ulp;/Volumes/Projects/esp-idf/components/unity;/Volumes/Projects/esp-idf/components/wifi_provisioning;/Volumes/Projects/thepia-all/TheBoot/main
-- Found PythonInterp: /usr/local/bin/python (found version "2.7.15") 
-- Found Perl: /usr/bin/perl (found version "5.18.2") 
-- Adding linker script /Volumes/Projects/thepia-all/TheBoot/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /Volumes/Projects/esp-idf/components/esp32/ld/esp32.rom.ld
-- Adding linker script /Volumes/Projects/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /Volumes/Projects/esp-idf/components/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Volumes/Projects/esp-idf/components/esp32/ld/esp32.rom.spiram_incompatible_fns.ld
-- Building empty aws_iot component due to configuration
-- Component libraries: 
-- Configuring done
CMake Error at /Volumes/Projects/esp-idf/tools/cmake/components.cmake:61 (add_library):
  Cannot find source file:

    hello_world_main.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
Call Stack (most recent call first):
  main/CMakeLists.txt:4 (register_component)

CMake Error at /Volumes/Projects/esp-idf/tools/cmake/components.cmake:61 (add_library):
  No SOURCES given to target: idf_component_main
Call Stack (most recent call first):
  main/CMakeLists.txt:4 (register_component)

igrr, 2019-02-17:
Hi @thepian, unlike the Make based build system which globs source files by default, in CMake the source files need to be listed in CMakeLists.txt: https://github.com/espressif/esp-idf/blob/abea9e4c02bb17e86298aec4e299780399e4789f/examples/get-started/hello_world/main/CMakeLists.txt#L1.

Globbing is also possible, but not recommended. Please see https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system-cmake.html#file-globbing-incremental-builds for details.

Alvin1Zhang, 2019-03-15:
@thepian Hi, would you help share if any updates for this issue? Thanks.

Alvin1Zhang, 2019-04-01:
Feel free to reopen if your issue still exists. @thepian Thanks.

closed 2019-04-01 by Alvin1Zhang