espressif/esp-idf #3062: TLS failure with multiple connections

gregjesl, 2019-02-17:
## Environment
- IDF Version: v3.3-beta1-333-g66439c9b9
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a5
- Operating System: Windows
- Power Supply: USB

## Problem
We are attempting to conduct an OTA via HTTPS while remaining connected to our MQTT broker via TLS.  We would like to remain connected in order to obtain feedback regarding the OTA process.  

[As stated in the documentation](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/esp_tls.html#_CPPv228esp_tls_init_global_ca_storev), mutliple TLS connections are possible with certificates using the same CA.  In `app_main()`, we added
```cpp
ESP_ERROR_CHECK(esp_tls_init_global_ca_store());
ESP_ERROR_CHECK(esp_tls_set_global_ca_store((const unsigned char *)SERVER_ROOT_PEM, sizeof(SERVER_ROOT_PEM)));
```
before setting up any services.  We then connect to the MQTT server using SSL transport.  

We commented out the following lines in the OTA Updater: 
https://github.com/espressif/esp-idf/blob/abea9e4c02bb17e86298aec4e299780399e4789f/components/esp_https_ota/src/esp_https_ota.c#L38-L43
Instead, we instantiate the OTA HTTPS config with
```cpp
config.transport_type = HTTP_TRANSPORT_OVER_SSL;
```

When we attempt an OTA update, the device returns the following:
```
D (41601) HTTP_CLIENT: Begin connect to: [server]
D (41607) esp-tls: host:[server]: strlen 26
D (41825) esp-tls: handshake in progress...
E (42146) esp-tls: mbedtls_ssl_handshake returned -0x2700
I (42147) esp-tls: Failed to verify peer certificate!
I (42148) esp-tls: verification info:   ! The certificate is not correctly signed by the trusted CA

E (42161) esp-tls: Failed to open new connection
E (42163) TRANS_SSL: Failed to open a new connection
E (42168) HTTP_CLIENT: Connection failed, sock < 0
D (42174) OTA_Update: HTTP_EVENT_DISCONNECTED
E (42179) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
```

The OTA is successful if we do not use SSL transport for our MQTT client, indicating the certificate used in the OTA update is correct.

mahavirj, 2019-02-18:
@gregjesl It seems like `esp_http_client` do not support global ca store yet for tls connections,

https://github.com/espressif/esp-idf/blob/abea9e4c02bb17e86298aec4e299780399e4789f/components/esp-tls/esp_tls.h#L85

We will try to address this soon.

gregjesl, 2019-02-18:
@mahavirj Can you confirm that the ESP32 supports two or more simultaneous TLS connections?

mahavirj, 2019-02-18:
@gregjesl Yes, definitely. Other than memory (can go up-to 35-40K per TLS connection) there should not be any other bottleneck.

gregjesl, 2019-02-18:
@mahavirj Got it, thanks.  I was seeing failures when I specify the `cert_pem` field of the OTA configuration and the MQTT client was using a TLS connection.  I initially thought it was due to an issue with how the IDF handles certificates/multiple TLS connections but I think it may actually be a memory issue.  I will try setting the `cert_pem` field of the OTA configuration and report the result.

mahavirj, 2019-02-18:
@gregjesl You may try attached fix (along with required change in `simple-ota` example) for reported problem and let me know result.

[fix_for_http_client_ca_store.tar.gz](https://github.com/espressif/esp-idf/files/2875668/fix_for_http_client_ca_store.tar.gz)

For second issue, yes having track of free heap might be useful, you may use `esp_get_free_heap_size()` for that.

gregjesl, 2019-02-25:
@mahavirj I was not able to integrate the patch; this may be because I'm working from an IDF fork.  I haven't used patches before so I'm not certain though.

closed 2019-03-04 by igrr