espressif/esp-idf #3063: HID : Is there a way to know if we are connected with an Iphone or an Android phone ?

Marc-Aurele, 2019-02-17:
Hello,

I'm developing an HID device using HID over BLE example stack provided in the esp-idf example and i would like to know if there is a way to know if the device is connected with an iphone or an android. Indeed, it seems that the report map behavior is not interpreted in the same way according to the platform so i need to know with which platform i'm connected to give the right report map.

Regards,

Aurélien

Yulong-espressif, 2019-02-21:
@Marc-Aurele  Iphone mobile phone has a dedicated server, such as ANCS.

Marc-Aurele, 2019-02-21:
Hello @Yulong-espressif ,
Thank you very much for the reply. I fanally make all i want except i have some unexpected behavior with iphone. First of all, for launching the voice assitant, i send an AC_HOME event for 1 second but when my phone is asleep; it wakes up the phone but then all event that i will send won't be interpreted anymore by the iphone. 
My second issue occurs sometime at pairing. Most of the time all is working fine but sometime pairing seems to be ok but every event i send to the iphone are not interpreted. I have noticed in that case that i don't see the following event "ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT" in the gap_event_handler function whereas when all is working fine, this event seems to be propagated.
Regards,

Aurélien

closed 2019-02-21 by Marc-Aurele