espressif/esp-idf #3066: esp-http-client breaks presignedURL when uploading to S3 with non zero filesize

buckybots, 2019-02-18:
Below, I show the presignedURL working with cURL, and then failing performing the same operation using esp-http-client.  Output is delineated with “//********”, my comments are in between each output section.  First cURL, then the modified code, lastly output of the test run on esp32.

I created a presigned URL, and uploaded a file to S3 using this URL and a test file.  Testing in cURL (and also in postman), the upload works, no problem.

Here is the run in cURL showing the successful upload :
note , all the lines beginning with “>” show the details of the actual HTTPS PUT to S3, and the “<” show the responses back from the server.

//***********************************************************************************************************

$ curl -v --request PUT --upload-file test.txt "https://iotvid-images-dev.s3.amazonaws.com/cam09/testpresign?AWSAccessKeyId=ASIAYSA5KIE4RLSHU2QJ&Expires=1550767077&Signature=R7lp2l44tggZl4xnrXTJAZLXRXA%3D&x-amz-security-token=FQoGZXIvYXdzEPr%2F%2F%2F%2F%2F%2F%2F%2F%2F%2FwEaDAj5BjtxHBetz3CG5iL9ARclw6FQ3pjvtmaYlVgrhAudiOgGXn2dw1RM17iT8CIKoQC9pz6e3QUEIo%2BxZGzlr6ZUaIB5ELaqMSTqj2hnXx3mbxe0sqRcVj3heKsoFaF%2B%2FmyIM1Uu8neScIGJvriH3sa6Mjvms%2FdVPOZ1U%2F9x6QZroXkhHyM0ZVQ%2FFSt%2FCmgFZXxWTa0DeClEiFYWogcksTQTOoCZ5rQeCMalVMow5OOLd7f%2FP6G%2FzOnPQZrZZetwM2wuyg5ht%2BXNBJC0kt1fr2nHhXJmEXkfM1KWTKiTR0Di9%2FeTyFcFoQ24wPWxOMreY%2FiUFMf30tabubrOdLAegPvULa4HAEcIXEZFP0Qo5b6r4wU%3D"
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying 52.216.20.32...
* TCP_NODELAY set
* Connected to iotvid-images-dev.s3.amazonaws.com (52.216.20.32) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* Cipher selection: ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH
* successfully set certificate verify locations:
*   CAfile: C:/Program Files/Git/mingw64/ssl/certs/ca-bundle.crt
  CApath: none
* TLSv1.2 (OUT), TLS header, Certificate Status (22):
} [5 bytes data]
* TLSv1.2 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
* TLSv1.2 (IN), TLS handshake, Server hello (2):
{ [87 bytes data]
* TLSv1.2 (IN), TLS handshake, Certificate (11):
{ [2902 bytes data]
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
{ [333 bytes data]
* TLSv1.2 (IN), TLS handshake, Server finished (14):
{ [4 bytes data]
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
} [70 bytes data]
* TLSv1.2 (OUT), TLS change cipher, Client hello (1):
} [1 bytes data]
* TLSv1.2 (OUT), TLS handshake, Finished (20):
} [16 bytes data]
* TLSv1.2 (IN), TLS change cipher, Client hello (1):
{ [1 bytes data]
* TLSv1.2 (IN), TLS handshake, Finished (20):
{ [16 bytes data]
* SSL connection using TLSv1.2 / ECDHE-RSA-AES128-SHA
* ALPN, server did not agree to a protocol
* Server certificate:
*  subject: C=US; ST=Washington; L=Seattle; O=Amazon.com Inc.; CN=*.s3.amazonaws.com
*  start date: Nov  7 00:00:00 2018 GMT
*  expire date: Feb  7 12:00:00 2020 GMT
*  subjectAltName: host "iotvid-images-dev.s3.amazonaws.com" matched cert's "*.s3.amazonaws.com"
*  issuer: C=US; O=DigiCert Inc; OU=www.digicert.com; CN=DigiCert Baltimore CA-2 G2
*  SSL certificate verify ok.
} [5 bytes data]
> PUT /cam09/testpresign?AWSAccessKeyId=ASIAYSA5KIE4RLSHU2QJ&Expires=1550767077&Signature=R7lp2l44tggZl4xnrXTJAZLXRXA%3D&x-amz-security-token=FQoGZXIvYXdzEPr%2F%2F%2F%2F%2F%2F%2F%2F%2F%2FwEaDAj5BjtxHBetz3CG5iL9ARclw6FQ3pjvtmaYlVgrhAudiOgGXn2dw1RM17iT8CIKoQC9pz6e3QUEIo%2BxZGzlr6ZUaIB5ELaqMSTqj2hnXx3mbxe0sqRcVj3heKsoFaF%2B%2FmyIM1Uu8neScIGJvriH3sa6Mjvms%2FdVPOZ1U%2F9x6QZroXkhHyM0ZVQ%2FFSt%2FCmgFZXxWTa0DeClEiFYWogcksTQTOoCZ5rQeCMalVMow5OOLd7f%2FP6G%2FzOnPQZrZZetwM2wuyg5ht%2BXNBJC0kt1fr2nHhXJmEXkfM1KWTKiTR0Di9%2FeTyFcFoQ24wPWxOMreY%2FiUFMf30tabubrOdLAegPvULa4HAEcIXEZFP0Qo5b6r4wU%3D HTTP/1.1
> Host: iotvid-images-dev.s3.amazonaws.com
> User-Agent: curl/7.57.0
> Accept: */*
> Content-Length: 348
> Expect: 100-continue
>
{ [5 bytes data]
< HTTP/1.1 100 Continue
} [5 bytes data]
* We are completely uploaded and fine
{ [5 bytes data]
< HTTP/1.1 200 OK
< x-amz-id-2: lZD7dAyKj+H8aYPPHOrPUR0KfPh8Tf0Uy/gigxh4fQ80kODoT+2ekv8FzFgaeDW7Z+rty2FU0dg=
< x-amz-request-id: 74AC90CF78B9C206
< Date: Mon, 18 Feb 2019 16:43:24 GMT
< ETag: "06091e3c4c845d23adf54797e3639985"
[... 573 lines cut ...]
D (17136) HTTP_CLIENT: http_on_headers_complete, status=403, offset=285, nread=285
D (17146) HTTP_CLIENT: content_length = -1
D (17156) HTTP_CLIENT: data_process=0, content_length=-1
D (17166) HTTP_CLIENT: http_on_body 1807
D (17166) HTTP_CLIENT: HTTP_EVENT_ON_DATA, len=1807
D (17166) HTTP_CLIENT: data_process=1807, content_length=-1
D (17176) HTTP_CLIENT: http_on_body 748
D (17176) HTTP_CLIENT: HTTP_EVENT_ON_DATA, len=748
D (17176) HTTP_CLIENT: http_on_chunk_complete
D (17186) HTTP_CLIENT: http_on_chunk_complete
D (17186) HTTP_CLIENT: http_on_message_complete, parser=3ffc5c30
D (17196) HTTP_CLIENT: HTTP_EVENT_ON_FINISH
D (17196) HTTP_CLIENT: Close connection
D (17206) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (17206) HTTP_CLIENT: presignedURL, set_post_field data:HTTPS Status = 403, content_length = -1
D (17216) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (17226) HTTP_CLIENT: Finish http example
I (60196) wifi: state: run -> init (7c0)
I (60196) wifi: pm stop, total sleep time: 0 us / 57362532 us

//*****************************************************************************

You can see the HTTPS Status = 403’s which are the fails.

Thanks for looking at this!

buckybots, 2019-02-20:
How do I see exactly what is being sent out and received by esp-http-client.  I would like to compare that to what is being sent and received by cURL to see what the difference is to help determine the cause of the problem.

buckybots, 2019-02-22:
Could someone let me know your thoughts on this issue.  Is it something being looked at, with feedback expected in a few days, or is this too far down on the priority list to deal with now, and I should look to solve this another way.

Thanks!

buckybots, 2019-02-22:
there was a mismatch in the signature.  Not a problem with esp-http-client.  Closing.

closed 2019-02-22 by buckybots

emhatek, 2020-11-16:
Guys,
Sorry is there any sample code in arduino file?, i'm still confure to post using URL, access key,secret key and bucket name to AWS s3.

me-no-dev, 2020-11-17:
Arduino uses ESP-IDF, so code is the same :)

emhatek, 2020-11-18:
> Arduino uses ESP-IDF, so code is the same :)

Hi me-no-dev,

based on your code could you explain to me, which one the access key and secret key and bucket name?