espressif/esp-idf #3069: Passing a char* to ESP_LOGI

itay-grudev, 2019-02-18:
## Environment

- Module or chip used:  ESP32
- IDF version: master
- Build System: Make
- Compiler version: `(crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0`
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

The following code fails to compile:

```cpp
const char *TAG = "APP", *ssid = "SSID";
ESP_LOGI( TAG, ssid );
```

### Actual Behavior

```
/home/user/Projects/esp32/main/main.cpp: In function 'void app_main()':
/home/user/Projects/esp32/main/main.cpp:341:28: error: expected ')' before 'ssid'
             ESP_LOGI( TAG, ssid );
                            ^
/home/user/esp-idf/components/log/include/esp_log.h:244:79: note: in definition of macro 'LOG_FORMAT'
 #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%d) %s: " format LOG_RESET_COLOR "\n"
                                                                               ^
/home/user/esp-idf/components/log/include/esp_log.h:311:41: note: in expansion of macro 'ESP_LOG_LEVEL'
         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
                                         ^
/home/user/esp-idf/components/log/include/esp_log.h:267:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^
/home/user/Projects/esp32/main/main.cpp:341:13: note: in expansion of macro 'ESP_LOGI'
             ESP_LOGI( TAG, ssid );
             ^
/home/user/Projects/esp32/main/main.cpp:341:28: error: expected ')' before 'ssid'
             ESP_LOGI( TAG, ssid );
                            ^
/home/user/esp-idf/components/log/include/esp_log.h:244:79: note: in definition of macro 'LOG_FORMAT'
 #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%d) %s: " format LOG_RESET_COLOR "\n"
                                                                               ^
/home/user/esp-idf/components/log/include/esp_log.h:311:41: note: in expansion of macro 'ESP_LOG_LEVEL'
         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
                                         ^
/home/user/esp-idf/components/log/include/esp_log.h:267:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^
/home/user/Projects/esp32/main/main.cpp:341:13: note: in expansion of macro 'ESP_LOGI'
             ESP_LOGI( TAG, ssid );
             ^
/home/user/Projects/esp32/main/main.cpp:341:28: error: expected ')' before 'ssid'
             ESP_LOGI( TAG, ssid );
                            ^
/home/user/esp-idf/components/log/include/esp_log.h:244:79: note: in definition of macro 'LOG_FORMAT'
 #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%d) %s: " format LOG_RESET_COLOR "\n"
                                                                               ^
/home/user/esp-idf/components/log/include/esp_log.h:311:41: note: in expansion of macro 'ESP_LOG_LEVEL'
         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
                                         ^
/home/user/esp-idf/components/log/include/esp_log.h:267:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^
/home/user/Projects/esp32/main/main.cpp:341:13: note: in expansion of macro 'ESP_LOGI'
             ESP_LOGI( TAG, ssid );
             ^
/home/user/Projects/esp32/main/main.cpp:341:28: error: expected ')' before 'ssid'
             ESP_LOGI( TAG, ssid );
                            ^
/home/user/esp-idf/components/log/include/esp_log.h:244:79: note: in definition of macro 'LOG_FORMAT'
 #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%d) %s: " format LOG_RESET_COLOR "\n"
                                                                               ^
/home/user/esp-idf/components/log/include/esp_log.h:311:41: note: in expansion of macro 'ESP_LOG_LEVEL'
         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
                                         ^
/home/user/esp-idf/components/log/include/esp_log.h:267:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^
/home/user/Projects/esp32/main/main.cpp:341:13: note: in expansion of macro 'ESP_LOGI'
             ESP_LOGI( TAG, ssid );
             ^
/home/user/Projects/esp32/main/main.cpp:341:28: error: expected ')' before 'ssid'
             ESP_LOGI( TAG, ssid );
                            ^
/home/user/esp-idf/components/log/include/esp_log.h:244:79: note: in definition of macro 'LOG_FORMAT'
 #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%d) %s: " format LOG_RESET_COLOR "\n"
                                                                               ^
/home/user/esp-idf/components/log/include/esp_log.h:311:41: note: in expansion of macro 'ESP_LOG_LEVEL'
         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
                                         ^
/home/user/esp-idf/components/log/include/esp_log.h:267:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^
/home/user/Projects/esp32/main/main.cpp:341:13: note: in expansion of macro 'ESP_LOGI'
             ESP_LOGI( TAG, ssid );
             ^
/home/user/Projects/esp32/main/main.cpp:340:39: warning: unused variable 'ssid' [-Wunused-variable]
             const char *TAG = "APP", *ssid = "SSID";
                                       ^
make[1]: *** [/home/user/esp-idf/make/component_wrapper.mk:292: main.o] Error 1
make: *** [/home/user/esp-idf/make/project.mk:527: component-main-build] Error 2
```

itay-grudev, 2019-02-18:
Got it. The correct syntax is:

```cpp
ESP_LOGI( TAG, "%s", ssid );
```

closed 2019-02-18 by itay-grudev

baldhead69, 2021-08-14:
printf work inn this format.
```
const char str[ ] = "\n**************** %s ****************\n";   
printf(str, info);
```
But ESP_LOGI dont work in this format: 
```
const char str[ ] = "\n**************** %s ****************\n";   
ESP_LOGI( TAG, str, info );
```

baldhead69, 2021-08-14:
How to add a empty line before ESP_LOGI without using printf( "\n" )  ?

How to change text color in ESP_LOGI ?
What commands ?

itay-grudev, 2021-08-14:
New lines are frowned upon in logs as each line in a log should be a self contained message. Nevertheless if you really need new lines around your message you can use:

```cpp
ESP_LOGI( TAG, "\n%s\n", message );
```

Furthermore here is a pattern for displaying different colours: `"\e[%im%s\e[0m"`. It has two parameters (`%i` and `%s`). The first is the colour code and the latter your message. The following is a table of colour codes:

|   Colour   |  Code  |
| ------------ | --------- |
|     red      |   `31`   |
|   green    |   `32`   |
|   yellow   |   `33`   |
|     blue    |   `34`   |

baldhead69, 2021-08-14:
Hi @itay-grudev 

The problem is that the log appear unorganized.

There should be a LOG option for the user (LOGU maybe) so that the user can configure it in a more organized way without needing to use printf.

How to appear in vscode:
```
#define my_log_color "\033[91m" 
    
ESP_LOGI( WIFI_TAG, "\n" );
ESP_LOGI( WIFI_TAG, "%sget_mac[0] = %02X", my_log_color, get_mac[0] );
ESP_LOGI( WIFI_TAG, "%s\nget_mac[1] = %02X", my_log_color, get_mac[1] );
ESP_LOGI( WIFI_TAG, "%sget_mac[2] = %02X", my_log_color, get_mac[2] );
ESP_LOGI( WIFI_TAG, "%sget_mac[3] = %02X", my_log_color, get_mac[3] );
ESP_LOGI( WIFI_TAG, "%sget_mac[4] = %02X", my_log_color, get_mac[4] );
ESP_LOGI( WIFI_TAG, "%sget_mac[5] = %02X", my_log_color, get_mac[5] );
ESP_LOGI( WIFI_TAG, "\n" );
```

[image]

baldhead69, 2021-08-16:
Hi,

I saw the "LOG_FMT" in several esp_idf files, but i'm not finding in my esp-idf installation and no documentation in espressif docs.

Example:
ESP_LOGW(TAG, LOG_FMT("error in %s : %d"), ctx, errno);

My idf-version: v4.4-dev-2359-g58022f859

Thank's.

baldhead69, 2021-08-16:
> printf work in this format.
> 
> ```
> const char str[ ] = "\n**************** %s ****************\n";   
> printf(str, info);
> ```
> 
> But ESP_LOGI dont work in this format:
> 
> ```
> const char str[ ] = "\n**************** %s ****************\n";   
> ESP_LOGI( TAG, str, info );
> ```

If replace "const char str[ ] = "\n**************** %s ****************\n";"
with 
#define str "\n**************** %s ****************\n"
WORKS.

#define str "\n**************** %s ****************\n"   
ESP_LOGI( TAG, str, info );

itay-grudev, 2021-08-16:
That is because the format is supposed to be an argument and not a variable. `printf` uses some pre-compiler magic to achieve the formatting and the formatting is not done at runtime.

baldhead69, 2021-08-22:
How to put a new line in console without using printf("\n") ?

ESP_LOGI dont work.