espressif/esp-idf #3070: https_mbedtls example

myzhang001, 2019-02-19:
我在使用 https_mbedtls example 时发现我在例程的基础上在创建一个任务 程序就无法运行
代码如下
static void zmy_test_task(void *pvParameters)
{
    while(1)
    {
        ESP_LOGI(TAG,"ZMY TEST");
        vTaskDelay(5000 / portTICK_PERIOD_MS);
    }
}

void app_main()
{
    ESP_ERROR_CHECK( nvs_flash_init() );
    initialise_wifi();

    xTaskCreate(&zmy_test_task, "ZMY TEST ", 1024, NULL, 4, NULL);

    xTaskCreate(&https_get_task, "https_get_task", 8192, NULL, 5, NULL);
}
其余代码为做任何修改  程序一直复位

igrr, 2019-02-19:
Try increasing stack size to at least 2048. Stack size of 1024 bytes is not sufficient for printf (which gets called from ESP_LOG.

Also please check IDF Monitor, it will decode backtraces automatically, making it easier to debug crashes: https://docs.espressif.com/projects/esp-idf/zh_CN/latest/get-started/idf-monitor.html

mahavirj, 2019-04-10:
@myzhang001 Could you please update here? Can this issue be closed?

mahavirj, 2019-04-12:
Closing due to lack of response. Please feel free to open new issue in case still problem exists.

closed 2019-04-12 by mahavirj