espressif/esp-idf #3072: Increase the severity of message for whitelist full when reading the (available) size

LaurentLouf, 2019-02-19:
## Environment

- Development Kit: N/A
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used:  N/A
- IDF version (run ``git describe --tags`` to find it): latest as of today : abea9e4
- Build System:  N/A
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): N/A
- Operating System: N/A
- Power Supply: N/A

## Problem Description

The problem only concerns the level of a message, which, in my humble opinion, should be increased. The function concerned is https://github.com/espressif/esp-idf/blob/abea9e4c02bb17e86298aec4e299780399e4789f/components/bt/bluedroid/stack/btm/btm_acl.c#L2067  The current level for the message when the whitelist is full is `DEBUG` while I think it should at least require a level of `WARNING`, since a whitelist full could cause some trouble to operate normally the device.

Note that when trying to add a device but the whitelist is full, the message has a level of `ERROR` which in this case is normal.  

The naming of this function is also a bit misleading since it returns the available size, not the current size of the whitelist. This naming ambiguity is propagated to `btc_get_whitelist_size` and `esp_ble_gap_get_whitelist_size`

Weijian-Espressif, 2019-03-20:
@LaurentLouf, thanks for your reminder. I will modify the level of this message.

Alvin1Zhang, 2020-05-29:
Thanks for reporting and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/5991e7dd7a9f30383cb48a4944f5f7724ba6df23, feel free to reopen if the issue still happens. Thanks.

closed 2020-05-29 by Alvin1Zhang