espressif/esp-idf #3075: Guru Meditation Error: Core  0 -vPortTaskWrapper at .. Every time at this  line components/freertos/port.c:403

jegajith, 2019-02-20:
Hi all , 

We are facing  Guru mediation- /components/freertos/port.c:403 , were we have only One external interrupt as peripheral input and all other actions are output , and we are using UART -1 as our debug log 

We are using ESP-IDF v3.3-dev-264-g8a69ffc36 2nd stage bootloader ( Tip of the tree of 3-2)
Here is back-trace log  , Please have a look and let us know the work around. 

[Tue Feb 19 12:50:46.322 2019] Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[Tue Feb 19 12:50:46.322 2019] Core 0 register dump:
[Tue Feb 19 12:50:46.322 2019] PC      : 0x40173f84  PS      : 0x00060830  A0      : 0x80180b40  A1      : 0x3ffe4fc0  
[Tue Feb 19 12:50:46.322 2019] A2      : 0x00000002  A3      : 0x3ff000d0  A4      : 0x3ffc01a8  A5      : 0x00000000  
[Tue Feb 19 12:50:46.322 2019] A6      : 0x0000002a  A7      : 0x3ffb75d0  A8      : 0xf40300fd  A9      : 0x3ffe4fb0  
[Tue Feb 19 12:50:46.322 2019] A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x00000001  
[Tue Feb 19 12:50:46.322 2019] A14     : 0x0000000a  A15     : 0x3ffb75b0  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
[Tue Feb 19 12:50:46.322 2019] EXCVADDR: 0xf40300fd  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[Tue Feb 19 12:50:46.322 2019] 
[Tue Feb 19 12:50:46.322 2019] Backtrace: 0x40173f84:0x3ffe4fc0 0x40180b3d:0x3ffe4ff0 0x40190e8d:0x3ffe5020 0x40191293:0x3ffe5040 0x4019408d:0x3ffe5060 0x40194119:0x3ffe5080 0x4008ea12:0x3ffe50a0 0x40093da9:0x3ffe50e0

Backtrace: 

ts/esp32/phy_init.c:299 (discriminator 2)
0x40180b3d: wifi_rf_phy_enable at ??:?
0x40190e8d: pm_dream at ??:?
0x40191293: pm_tbtt_process at ??:?
0x4019408d: dbg_lmac_init at ??:?
0x40194119: pp_timer_do_process at ??:?
0x4008ea12: ppTask at ??:?
0x40093da9: vPortTaskWrapper at /home/fxt/esp/IDF_3_2_TipOftheTree/esp-idf-v3.2.beta1.tip/components/freertos/port.c:403
fxt@fxtblrlp46:~/Pro
-------------------------------------

[Tue Feb 19 13:33:29.084 2019] Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[Tue Feb 19 13:33:29.178 2019] Memory dump at 0x40188b20: 2182f03d 00b9590f a400b181
[Tue Feb 19 13:33:29.178 2019] Core 0 register dump:
[Tue Feb 19 13:33:29.178 2019] PC      : 0x40188b27  PS      : 0x00060830  A0      : 0x80189748  A1      : 0x3ffe4f10  
[Tue Feb 19 13:33:29.178 2019] A2      : 0x3ffd61ac  A3      : 0x3fff3a80  A4      : 0x00000080  A5      : 0x3ffd56e8  
[Tue Feb 19 13:33:29.178 2019] A6      : 0x3fff3af4  A7      : 0x0000011c  A8      : 0x00000000  A9      : 0x3fff3c28  
[Tue Feb 19 13:33:29.178 2019] A10     : 0x3ffe4f20  A11     : 0x00000000  A12     : 0x3fff3b0c  A13     : 0x3ffe4f8c  
[Tue Feb 19 13:33:29.178 2019] A14     : 0x3ffd56e8  A15     : 0x00000006  SAR     : 0x00000020  EXCCAUSE: 0x00000000  
[Tue Feb 19 13:33:29.178 2019] EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
[Tue Feb 19 13:33:29.178 2019] 
[Tue Feb 19 13:33:29.178 2019] Backtrace: 0x40188b27:0x3ffe4f10 0x40189745:0x3ffe5030 0x40193095:0x3ffe5050 0x4008ea62:0x3ffe50a0 0x40093da9:0x3ffe50e0

Backtrace 
2:0x3ffe50a0 0x40093da9:0x3ffe50e0
0x40188b27: sta_input at ??:?
0x40189745: sta_rx_cb at ??:?
0x40193095: ppRxPkt at ??:?
0x4008ea62: ppTask at ??:?
0x40093da9: vPortTaskWrapper at /home/fxt/esp/IDF_3_2_TipOftheTree/esp-idf-v3.2.beta1.tip/components/freertos/port.c:403

----------------------------------------

[Tue Feb 19 15:28:25.991 2019] Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
[Tue Feb 19 15:28:25.991 2019] Core 0 register dump:
[Tue Feb 19 15:28:25.991 2019] PC      : 0x84e08c0e  PS      : 0x00060630  A0      : 0x80190815  A1      : 0x3ffe5040  
[Tue Feb 19 15:28:25.991 2019] A2      : 0x00000000  A3      : 0x00000005  A4      : 0x00400000  A5      : 0x00000000  
[Tue Feb 19 15:28:25.991 2019] A6      : 0x3ffe3794  A7      : 0x00000008  A8      : 0x801906c1  A9      : 0x3ffd79a8  
[Tue Feb 19 15:28:25.991 2019] A10     : 0x3ffd1278  A11     : 0x00000000  A12     : 0x3ffd796c  A13     : 0x3ffd1278  
[Tue Feb 19 15:28:25.991 2019] A14     : 0x00000000  A15     : 0x00000100  SAR     : 0x00000020  EXCCAUSE: 0x00000014  
[Tue Feb 19 15:28:25.991 2019] EXCVADDR: 0x84e08c0c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[Tue Feb 19 15:28:25.991 2019] 
[Tue Feb 19 15:28:25.991 2019] Backtrace: 0x44e08c0e:0x3ffe5040 0x40190812:0x3ffe5070 0x4008ea8d:0x3ffe50a0 0x40093da9:0x3ffe50e0

Backtrace 
9:0x3ffe50e0
0x44e08c0e: ?? ??:0
0x40190812: lmacProcessTxComplete at ??:?
0x4008ea8d: ppTask at ??:?
0x40093da9: vPortTaskWrapper at /home/fxt/esp/IDF_3_2_TipOftheTree/esp-idf-v3.2.beta1.tip/components/freertos/port.c:403
[... 48 lines cut ...]

[Wed Feb 20 08:55:56.575 2019] Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
[Wed Feb 20 08:55:56.575 2019] Core 0 register dump:
[Wed Feb 20 08:55:56.575 2019] PC      : 0x90c5b838  PS      : 0x00060a33  A0      : 0x8016e4e1  A1      : 0x3ffe4f30  
[Wed Feb 20 08:55:56.575 2019] A2      : 0x90c5b838  A3      : 0x00000000  A4      : 0x00000050  A5      : 0x00000000  
[Wed Feb 20 08:55:56.575 2019] A6      : 0x7dffffd6  A7      : 0x00000000  A8      : 0x3f453c64  A9      : 0x00060a23  
[Wed Feb 20 08:55:56.575 2019] A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00000001  
[Wed Feb 20 08:55:56.590 2019] A14     : 0x00060a20  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x00000014  
[Wed Feb 20 08:55:56.590 2019] EXCVADDR: 0x90c5b838  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[Wed Feb 20 08:55:56.590 2019] 
[Wed Feb 20 08:55:56.590 2019] Backtrace: 0x50c5b838:0x3ffe4f30 0x4016e4de:0x3ffe4f50 0x4018012c:0x3ffe4f70 0x40190c8f:0x3ffe4fa0 0x40191620:0x3ffe4fc0 0x40191beb:0x3ffe4ff0 0x401922b5:0x3ffe5030 0x401929a5:0x3ffe5050 0x4008ea62:0x3ffe50a0 0x40093da9:0x3ffe50e0

**Backtrace**
0x50c5b838: ?? ??:0
0x4016e4de: periph_module_disable at /home/fxt/esp/IDF_3_2_TipOftheTree/esp-idf-v3.2.beta1.tip/components/driver/periph_ctrl.c:42
0x4018012c: wifi_rf_phy_disable at ??:?
0x40190c8f: lmac_stop_hw_txq at ??:?
0x40191620: pm_rx_beacon_process at ??:?
0x40191beb: pm_on_beacon_rx at ??:?
0x401922b5: ppRxProtoProc at ??:?
0x401929a5: ppRxPkt at ??:?
0x4008ea62: ppTask at ??:?
0x40093da9: vPortTaskWrapper at /home/fxt/esp/IDF_3_2_TipOftheTree/esp-idf-v3.2.beta1.tip/components/freertos/port.c:403

----------------------------------------

jegajith, 2019-02-20:
All Task are Pinned  to Core-1 and Stack (WiFi and BLE )are in  Core-0

Dazza0, 2019-02-22:
The errors above (Load Prohibited, Illegal Instruction, and Instruction Fetch Prohibited) are all indicative of memory corruption. 
- Illegal Instruction means that the instruction at a particular address was not a valid opcode. From the register dump, the CPU was attempting to run an instructions at addresses `0x40188b27`. `0x40150a24`, and `0x40185cf4`. Use GDB to check if that address has a valid instruction or function. If not, it may indicate a corrupt function pointer was called.
- Instruction Fetch Prohibited means that when attempting to call a subroutine, the address of that subroutine is not valid (out of bounds).  This could also be caused by corrupt function pointers.
- The Load Prohibited was attempting to load a value form memory address `0xf40300fd` which is not a valid address on the ESP32. Data is usually stored in address starting with `0x3xxxxxxx` and instructions are stored in addresses starting with `0x4xxxxxx`.

Try setting your Heap Corrupt Detection to Comprehensive in `make menuconfig`, and double checking to see if all your memory allocation and accesses are within bounds.

jegajith, 2019-02-25:
@Dazza0  Thank you for your comment , Let me setup Heap Corrupt Detection in menu-config and Let you know if i have any  observation.

Alvin1Zhang, 2019-03-14:
@jegajith Hi, would you help share if any updates for the issue? Thanks.

projectgus, 2019-08-19:
Closing due to lack of updates. Please open a new issue if still experiencing problems, and you think it's a bug in ESP-IDF.

closed 2019-08-19 by projectgus