espressif/esp-idf #3076: step by step to failure

braveheart007, 2019-02-20:
when i tried to prepare the ubuntu environment of development for ESP32, i jsut found there is so many unexpected failure. 
Could you help to prepare one copy for us to download. 
Thanks for your time

Alvin1Zhang, 2019-02-20:
@braveheart007 Hi, could you help specify which version you are using, what failures you have, and details like this? Thanks. You are also welcome to explore/post a topic in our forum https://esp32.com/index.php. Thanks.

braveheart007, 2019-02-20:
sure! I tried 14.04 and 16.04 ubuntu both.  the downloading speed is too low to fetch IDF from git hub when I use 14.04;  situation improves a lot when I use 16.04 ubuntu, but new issue come alone with the 16.04 ubuntu.  for example:Project is not inside a git repository, will not use 'git describe' to determine PROJECT_VER.

Alvin1Zhang, 2019-02-20:
@braveheart007 Hi, thanks for your inputs, we will look into this issue and provide feedbacks if we have updates, thanks for posting this issue.

braveheart007, 2019-02-20:
make menuconfig works fine. the step of "make all" got some wrong:
esp32@esp32-VirtualBox:/mnt/share/ESP32/esp32proj/hello_world$ make all -j8
Toolchain path: /mnt/share/ESP32/toolchain/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
AR build/aws_iot/libaws_iot.a
Project is not inside a git repository, will not use 'git describe' to determine PROJECT_VER.
App "hello-world" version: 1
CC build/bt/bt.o
CXX build/asio/asio/asio/src/asio.o
make[1]: xtensa-esp32-elf-c++: Command not found
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'asio/asio/src/asio.o' failed
make[1]: *** [asio/asio/src/asio.o] Error 127
/mnt/share/ESP32/IDF/esp-idf/make/project.mk:527: recipe for target 'component-asio-build' failed
make: *** [component-asio-build] Error 2
make: *** Waiting for unfinished jobs....
CC build/app_trace/app_trace_util.o
CC build/app_update/esp_ota_ops.o
CC build/bootloader_support/src/bootloader_flash.o
CC build/app_trace/host_file_io.o
CC build/coap/libcoap/src/address.o
Python requirements from /mnt/share/ESP32/IDF/esp-idf/requirements.txt are satisfied.
CC build/app_trace/app_trace.o
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'src/bootloader_flash.o' failed
make[1]: *** [src/bootloader_flash.o] Error 1
/mnt/share/ESP32/IDF/esp-idf/make/project.mk:527: recipe for target 'component-bootloader_support-build' failed
make: *** [component-bootloader_support-build] Error 2
CC build/app_trace/gcov/gcov_rtio.o
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'host_file_io.o' failed
make[1]: *** [host_file_io.o] Error 1
make[1]: *** Waiting for unfinished jobs....
CC build/app_update/esp_app_desc.o
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'app_trace_util.o' failed
make[1]: *** [app_trace_util.o] Error 1
CC build/coap/libcoap/src/async.o
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'bt.o' failed
make[1]: *** [bt.o] Error 1
/mnt/share/ESP32/IDF/esp-idf/make/project.mk:527: recipe for target 'component-bt-build' failed
make: *** [component-bt-build] Error 2
CC build/coap/libcoap/src/block.o
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'esp_ota_ops.o' failed
make[1]: *** [esp_ota_ops.o] Error 1
make[1]: *** Waiting for unfinished jobs....
CC build/coap/libcoap/src/coap_time.o
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'app_trace.o' failed
make[1]: *** [app_trace.o] Error 1
CC build/coap/libcoap/src/debug.o
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'libcoap/src/address.o' failed
make[1]: *** [libcoap/src/address.o] Error 1
make[1]: *** Waiting for unfinished jobs....
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'esp_app_desc.o' failed
make[1]: *** [esp_app_desc.o] Error 1
/mnt/share/ESP32/IDF/esp-idf/make/project.mk:527: recipe for target 'component-app_update-build' failed
make: *** [component-app_update-build] Error 2
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'gcov/gcov_rtio.o' failed
make[1]: *** [gcov/gcov_rtio.o] Error 1
/mnt/share/ESP32/IDF/esp-idf/make/project.mk:527: recipe for target 'component-app_trace-build' failed
make: *** [component-app_trace-build] Error 2
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'libcoap/src/async.o' failed
make[1]: *** [libcoap/src/async.o] Error 1
[... 46 lines cut ...]
make[1]: *** [component-log-build] Error 2
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'bootloader_start.o' failed
make[2]: *** [bootloader_start.o] Error 1
/mnt/share/ESP32/IDF/esp-idf/make/project.mk:527: recipe for target 'component-main-build' failed
make[1]: *** [component-main-build] Error 2
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'src/efuse.o' failed
make[2]: *** [src/efuse.o] Error 1
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'src/bootloader_flash.o' failed
make[2]: *** [src/bootloader_flash.o] Error 1
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'src/bootloader_init.o' failed
make[2]: *** [src/bootloader_init.o] Error 1
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'src/bootloader_common.o' failed
make[2]: *** [src/bootloader_common.o] Error 1
as: unrecognized option '--longcalls'
/mnt/share/ESP32/IDF/esp-idf/make/component_wrapper.mk:291: recipe for target 'src/bootloader_utility.o' failed
make[2]: *** [src/bootloader_utility.o] Error 1
/mnt/share/ESP32/IDF/esp-idf/make/project.mk:527: recipe for target 'component-bootloader_support-build' failed
make[1]: *** [component-bootloader_support-build] Error 2
/mnt/share/ESP32/IDF/esp-idf/components/bootloader/Makefile.projbuild:41: recipe for target '/mnt/share/ESP32/esp32proj/hello_world/build/bootloader/bootloader.bin' failed
make: *** [/mnt/share/ESP32/esp32proj/hello_world/build/bootloader/bootloader.bin] Error 2

closed 2019-02-20 by braveheart007