espressif/esp-idf #3079: Secure boot (One-time flash) issue

MaxHrytsak, 2019-02-20:
## Environment

- Module ESP32-WROOM-32
- IDF version v3.2-dev-1148-g96cd3b75c
- Compiler version  1.22.0-80-g6c4433a5
- Esptools version 2.5..0
- Operating System:     Windows 10 Pro
- Power Supply:         USB
- IDE: Eclipse IDE for C/C++ Developers , Version: 2018-12 (4.10.0), Build id: 20181214-0600

## Problem Description

We have a problem with enabling secure boot. We use the msys2 toolchain. All commands run in the mingw32. 

We follow step by step as discribed in [Reference document(RD), "How to enable secure boot"](https://docs.espressif.com/projects/esp-idf/en/latest/security/secure-boot.html#secure-boot-resume-normal-flashing) and the chip die after power reset (step 8-9 in RD). 
##### Step 1

Enable secure boot in menuconfig

command: make menuconfig

keyfile: sg_secure_boot.pem

```
Max@DESKTOP-PFGLJT2 MINGW32 /d/firmware/secure_test-firware/code
# make menuconfig
MENUCONFIG
configuration written to /d/firmware/secure_test-firware/code/sdkconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.
```

[image]

##### Step 2

Build project

command: make all app

```
Max@DESKTOP-PFGLJT2 MINGW32 /d/firmware/secure_test-firware/code
# make all app
GENCONFIG
Building partitions from /d/firmware/secure_test-firware/code/sg_pt.csv...
espsecure.py v2.5.0
Signed 3072 bytes of data from D:/firmware/secure_test-firware/code/build/sg_pt-unsigned.bin with key D:/firmware/secure_test-firware/code/sg_secure_boot.pem

.........................

CC build/bootloader_support/src/bootloader_flash.o
espsecure.py v2.5.0
D:/firmware/secure_test-firware/code/sg_secure_boot.pem public key extracted to D:/firmware/secure_test-firware/code/build/bootloader_support/signature_verification _key.bin
EMBED build/bootloader_support/signature_verification_key.bin.bin.o
AR build/bootloader_support/libbootloader_support.a
CC build/bt/bt.o

......................

AR build/xtensa-debug-module/libxtensa-debug-module.a
LD build/secure_test.elf
esptool.py v2.5.0
espsecure.py v2.5.0
Signed 1245104 bytes of data from D:/firmware/secure_test-firware/code/build/secure_test-unsigned.bin with key D:/firmware/secure_test-firware/code/sg_secure_boot.pem
(Secure boot enabled, so bootloader not flashed automatically. See 'make bootloader' output)
To flash app & partition table, run 'make flash' or:
python /d/firmware/secure_test-firware/components/esptool_py/esptool/esptool.py --chip esp32 --port COM6 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xd000 /d/firmware/secure_test-firware/code/build/ota_data_initial.bin 0x10000 /d/firmware/secure_test-firware/code/build/secure_test.bin 0x8000 /d/firmware/secure_test-firware/code/build/sg_pt.bin
App built. Default flash app command is:
python /d/firmware/secure_test-firware/components/esptool_py/esptool/esptool.py --chip esp32 --port COM6 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x10000 /d/firmware/secure_test-firware/code/build/secure_test.bin
[... 284 lines cut ...]
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Identity fuses:
MAC                    MAC Address
  = cc:50:e3:8d:27:90 (CRC 7a OK) R/W
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VERSION           Reserved for future chip versions                 = 2 R/W (0x2)
CHIP_PACKAGE           Chip package identifier                           = 0 R/W (0x0)

Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1107 R/W (0x1)

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
```

Most interesting that we have enabled secure boot early on another machine.

projectgus, 2019-02-21:
Hi @MaxHrytsak ,

Thanks for the very detailed bug report. It makes it very straightforward to diagnose the issue.

It looks like you've hit a nasty bug which was fixed in 4faa2a8. Your ESP-IDF master branch version is slightly older than this. Your bootloader.bin length happens to be 20368 bytes and `(20368 % 128) < 32`.

Because of the bug, the secure boot digest that the bootloader generated on first boot (for hardware secure boot protection) is invalid and doesn't match the actual bootloader.bin.  Because secure boot was configured in "one time flash" mode, there's no way to recover the chip after the digest was written. I'm terribly sorry about that.

If you can, I'd recommend checking out the v3.2 release branch (`release/v3.2`) and fast-forwarding your IDF to the latest commit on that branch. This is a beta quality release branch and I expect the final v3.2 release will be made in then ext couple of weeks. It will have all of the features on the master branch revision you have, plus a good number of fixes (including that one). Some description of the different IDF version options can be found here: https://docs.espressif.com/projects/esp-idf/en/latest/versions.html

projectgus, 2019-04-30:
Hi @MaxHrytsak , 

Have you had an opportunity to try this fix out yet?

Angus

Alvin1Zhang, 2019-05-06:
@MaxHrytsak Hi, would you help share if any updates for the issue? Thanks.

MaxHrytsak, 2019-05-06:
@projectgus @Alvin1Zhang Hi. Sorry for silent. 
Yes, the problem was in the bootloader. But I corrected that part in my IDF(older version).

projectgus, 2019-05-13:
Hi @MaxHrytsak ,

It sounds like you found a solution, so I'm going to close this issue. Feel free to comment if there's still a problem.

Angus

closed 2019-05-13 by projectgus