espressif/esp-idf #3080: aws-iot last will

chegewara, 2019-02-20:
I am fighting for some time with aws-iot last will. When i set plain text in last will there is no issue, but when i am trying to set simple json string then most of the time data is corrupted. Usually its missing 1 or 2 leading chars or sometimes leading bytes (1-2) are replaced with other random char.
I could add leading space if not the problem that sometimes 2 chars are corrupted and in very rare case there is no corruption in string.

Any advice or help will be appreciated because its in production app.

mahavirj, 2019-02-27:
@chegewara Do you suspect this to be IDF side issue? If yes can you please provide sample code to reproduce it?

chegewara, 2019-02-27:
Hi @mahavirj 
to be honest i dont know if this is esp-idf issue or aws broker itself.
This is code i am using to create last will message and few other topics. With regular topics i dont see this issue, it is only last will problem:
```
    char* name = get_fanet_addr(); // name = "1005";
    ESP_LOGE(__func__, "%s", name);
    char l_will[50];
    sprintf(l_will, "  {\"data\":\"%s\"}", name);

    IoT_MQTT_Will_Options will = iotMqttWillOptionsDefault;
    will.pTopicName = "/last_will";
    will.topicNameLen = strlen("/last_will");
    will.pMessage = l_will;
    will.msgLen = strlen(l_will);
```

I would suspect my error if this would be trailing character, but with leading characters corrupted it has to be something in library or aws.
Thank You for help.

chegewara, 2019-02-28:
@mahavirj What i can say in addition is:
when i perform test from aws IoT core and publish to /last_will topic then all works fine. This is suggesting that message sent to aws broker from esp32 aws-iot component is corrupted (some memory allocation problem maybe?).

mahavirj, 2019-04-18:
@chegewara Sorry for late followup, did you analyze this further? Interim we moved `aws-iot` to independent repository at https://github.com/espressif/esp-aws-iot, in case this still exists I will move it over there.

chegewara, 2019-04-18:
Hi @mahavirj 
since there was no response i left it with settings that worked in my app which is sending just simple plain text, one word or to be accurate numeric value as a string (ie 1005).
Because my project is in advanced stage, some devices are working now, i cant move to aws-iot repository, but thank you for info.