espressif/esp-idf #3081: ESP WPS makes software hang.

FlorisColpaert, 2019-02-21:
I'm working with the v3.1.3. Trying the WPS. The heapy flow works fine (start wps of router before wps of the ESP32). But no WPS time-out, calling wps manually, causes no respons in the main freertos task, reset chip doesn't work,...

FlorisColpaert, 2019-02-21:
/* WiFi Connection Example using WPS

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

/*
   Showing how to use WPS.

   WPS_TYPE_PBC: Start esp32 and it will enter wps PBC mode. Then push the button of wps on router down. The esp32 will connected to the router.
   WPS_TYPE_PIN: Start esp32, You'll see PIN code which is a eight-digit number showing on COM. Enter the PIN code in router and then the esp32 will connected to router.
*/

#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_log.h"
#include "esp_wps.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"

/*set wps mode via "make menuconfig"*/
#if CONFIG_EXAMPLE_WPS_TYPE_PBC
#define WPS_TEST_MODE WPS_TYPE_PBC
#elif CONFIG_EXAMPLE_WPS_TYPE_PIN
#define WPS_TEST_MODE WPS_TYPE_PIN
#else
#define WPS_TEST_MODE WPS_TYPE_DISABLE
#endif /*CONFIG_EXAMPLE_WPS_TYPE_PBC*/

#ifndef PIN2STR
#define PIN2STR(a) (a)[0], (a)[1], (a)[2], (a)[3], (a)[4], (a)[5], (a)[6], (a)[7]
#define PINSTR "%c%c%c%c%c%c%c%c"
#endif

static const char *TAG = "example_wps";
static esp_wps_config_t config = WPS_CONFIG_INIT_DEFAULT(WPS_TEST_MODE);

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
	ESP_LOGI(TAG, "SYSTEM_EVENT_STA_START");
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
	ESP_LOGI(TAG, "SYSTEM_EVENT_STA_GOT_IP");
	ESP_LOGI(TAG, "got ip:%s\n",
		ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
	ESP_LOGI(TAG, "SYSTEM_EVENT_STA_DISCONNECTED");
	ESP_ERROR_CHECK(esp_wifi_connect());
        break;
    case SYSTEM_EVENT_STA_WPS_ER_SUCCESS:
	/*point: the function esp_wifi_wps_start() only get ssid & password
	 * so call the function esp_wifi_connect() here
	 * */
	ESP_LOGI(TAG, "SYSTEM_EVENT_STA_WPS_ER_SUCCESS");
	ESP_ERROR_CHECK(esp_wifi_wps_disable());
	ESP_ERROR_CHECK(esp_wifi_connect());
	break;
    case SYSTEM_EVENT_STA_WPS_ER_FAILED:
	ESP_LOGI(TAG, "SYSTEM_EVENT_STA_WPS_ER_FAILED");
	ESP_ERROR_CHECK(esp_wifi_wps_disable());
	ESP_ERROR_CHECK(esp_wifi_wps_enable(&config));
	ESP_ERROR_CHECK(esp_wifi_wps_start(0));
	break;
    case SYSTEM_EVENT_STA_WPS_ER_TIMEOUT:
	ESP_LOGI(TAG, "SYSTEM_EVENT_STA_WPS_ER_TIMEOUT");
	ESP_ERROR_CHECK(esp_wifi_wps_disable());
	ESP_ERROR_CHECK(esp_wifi_wps_enable(&config));
	ESP_ERROR_CHECK(esp_wifi_wps_start(0));
 	break;
    case SYSTEM_EVENT_STA_WPS_ER_PIN:
	ESP_LOGI(TAG, "SYSTEM_EVENT_STA_WPS_ER_PIN");
	/*show the PIN code here*/
	ESP_LOGI(TAG, "WPS_PIN = "PINSTR, PIN2STR(event->event_info.sta_er_pin.pin_code));
	break;
    default:
        break;
    }
    return ESP_OK;
}

/*init wifi as sta and start wps*/
static void start_wps(void)
{
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_start());
    
    ESP_LOGI(TAG, "start wps...");
    
       
    ESP_ERROR_CHECK(esp_wifi_wps_enable(&config));
    ESP_ERROR_CHECK(esp_wifi_wps_start(1000));
}

void app_main()
{
    /* Initialize NVS — it is used to store PHY calibration data */
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );

    start_wps();

	vTaskDelay(1000);
	
	esp_wifi_wps_disable();
	
	ESP_LOGI(TAG, "wps stopped");
}

------------------------
I (162) wifi: wifi driver task: 3ffc0a40, prio:23, stack:3584, core=0
I (162) wifi: wifi firmware version: d5da5a5
I (162) wifi: config NVS flash: enabled
I (172) wifi: config nano formating: disabled
I (172) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (182) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (202) wifi: Init dynamic tx buffer num: 32
I (202) wifi: Init data frame dynamic rx buffer num: 32
I (202) wifi: Init management frame dynamic rx buffer num: 32
I (212) wifi: Init static rx buffer size: 1600
I (212) wifi: Init static rx buffer num: 10
I (212) wifi: Init dynamic rx buffer num: 32
I (282) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
I (292) wifi: mode : sta (30:ae:a4:7b:48:fc)
I (292) example_wps: start wps...
I (292) example_wps: SYSTEM_EVENT_STA_START
I (292) wpa: manufacturer: ESPRESSIF, model number: ESP32, model name: ESPRESSIF IOT, device name: ESP STATION
I (302) wpa: wifi_wps_enable

I (11162) wpa: wifi_wps_disable

--------------

Wps Stopped is never printed in the example

FlorisColpaert, 2019-02-21:
When the submodule esp32 in components is updated to branch releasev3.2, it seems to work all fine.

liuzfesp, 2019-02-23:
@FlorisColpaert do you mean the WPS doesn't work in v3.1.3, but it works in 3.2?

FlorisColpaert, 2019-02-25:
@liuzfesp, first I checked out the tag v3.1.3. Than i go to components eps32 (lib) and there i checked out the branch releasev3.2 of the submodule eps32 in the components.
If I build the example and my code and compile it, the wps succeeds. Also the reconnect method in my project was fixed again.

liuzfesp, 2019-02-25:
HI @FlorisColpaert, so you use the IDF v3.1.3 + esp32/lib v3.2. I think the correct way should be:

- If you want to use IDF v3.1.3,  cd $IDF_PATH, checkout IDF v3.1.3, then run 'git submodule update'
- If you want to use IDF v3.2,  cd $IDF_PATH, checkout IDF v3.2, then run 'git submodule update'

You may encounter other issues if you use IDF v3.1.3 + esp32/lib v3.2.

FlorisColpaert, 2019-03-01:
@liuzfesp, If i build (even) the example in the v3.1.3 the wps hangs when you add wps disable after x time you started wps. There is no WPS time out in the wifi event handler. I builded it with another module and added integration testing for or interface just so we could continue with or project. The device has integration tests and field test.

liuzfesp, 2019-03-04:
Hi @FlorisColpaert, I hope to replicate this issue myself, could you help to describe the test "when you add wps disable after x time you started wps" in detailed?  Does it means call esp_wifi_wps_disable() after calling esp_wifi_wps_start()? Or power off the router after esp_wifi_start() is called? Another question is how long time disable the WPS after wps start?

Alvin1Zhang, 2019-03-14:
@FlorisColpaert Hi, would you help provide if any updates for the issue? Thanks.

Alvin1Zhang, 2019-06-29:
@FlorisColpaert Thanks for reporting the issue. Will close the issue for the time being due to lack of feedback, feel free to reopen the issue if your issue still happens. Thanks.

closed 2019-06-29 by Alvin1Zhang