espressif/esp-idf #3082: esp_https_server: missing initializer for member 'httpd_config::uri_match_fn

chrissyheb, 2019-02-21:
Hi guys,
I'm using current master branch of esp-idf under linux with cmake.
When compiling my program using `esp_https_server` I get the following warning

`warning: missing initializer for member 'httpd_config::uri_match_fn' [-Wmissing-field-initializers]
[build]     conf = HTTPD_SSL_CONFIG_DEFAULT();`

I think within the header `esp_https_server.h` inside the macro `HTTPD_SSL_CONFIG_DEFAULT()` the initializer for `httpd_config::uri_match_fn` is missing. 
Could you please add this?
Thanks
Chrissy

mahavirj, 2019-02-27:
@chrissyheb It is already present or maybe you are referring to some different IDF version?
https://github.com/espressif/esp-idf/blob/140b6e3893317228debf5815b790e967c8562c62/components/esp_https_server/include/esp_https_server.h#L98

chrissyheb, 2019-02-27:
Seems it has been added two days ago:
https://github.com/espressif/esp-idf/commit/1af5d0d87cc2cf2f259ebd949da706f0d454d6bb
Thanks

closed 2019-02-27 by chrissyheb