espressif/esp-idf #3083: How do I connect to WEP as a station

rtmistler, 2019-02-21:
I have ESP32 and am using the esp-idf code.

I am attaching to WIFI as a station.

I can use Open, some of WPA, but I cannot use WEP or certain WPA configurations.

For WEP: There is no place in any structures where I can give a key.

For WPA: When it uses Enterprise, there is a Radius server required to configure and there is no way in the structures to specify this.

Are these modes supported for Station Mode?

Best Regards,

- RT

itay-grudev, 2019-02-21:
Since WEP is officially deprecated, the developers might not be interested in implementing it.

liuzfesp, 2019-02-23:
HI @rtmistler, the configuration of SSID/PASSWORD is same for WEP/WPA.

I think the simplest way is run the station example:

- Configure the router to WEP mode
- cd $IDF_PATH/examples/wifi/getting_started/station
- run make menuconfig => Example Configuration  to configure the SSID/PASSWORD of the AP
- build and run 

Then the ESP32 can connect to the router. You can refer to the implementation in the example.

rtmistler, 2019-02-25:
Thank you for the inputs.

I've been informed that WEP is deprecated, and therefore the requirement to
have it working is not so pressing.

Meanwhile for WPA2 Enterprise there are additional attributes beyond SSID
and password which are required, such as the address of the RADIUS server.

WPA2 Personal mode does work.

My original question had to do with where the attributes would be placed
for the RADIUS server.  I do not see any location in the data structures or
any arguments in API functions which would allow me to pass the RADIUS
server hostname or address.

I am familiar with the cited example and have used to to complete our
existing code application.

Best Regards,

- RT

liuzfesp, 2019-02-25:
@ rtmistler currently WPA2 enterprise doesn't support to configure RADIUS server hostname/address.

rtmistler, 2019-02-25:
Hi,

Thank you for the reply.

Correct me if I'm wrong, however I feel that a more accurate statement
would be that you presently do support WPA-Personal, but presently do not
support WPA-Enterprise,
https://en.wikipedia.org/wiki/Wi-Fi_Protected_Access#WPA_terminology.

Best,

- RT

liuzfesp, 2019-02-26:
Hi @rtmistler, ESP32 supports WPA-Enterprise, but it can only support TLS, PEAP-MSCHAPv2 and TTLS-MSCHAPv2 method. Moreover, we doesn't provide API to configure all attributes of RADIUS. 
Maybe you can try WPA2-Enterprise example: $IDF_PATH/examples/wifi/wpa2_enterprise

rtmistler, 2019-02-26:
I'll give it a try.
-RT

rtmistler, 2019-02-26:
Thank you.  That example does explain what is supported versus not.
Best,
- RT

Alvin1Zhang, 2019-03-14:
@rtmistler Hi, could you help share if any updates? Thanks.

Alvin1Zhang, 2019-03-19:
Feel free to reopen, thanks. @rtmistler

closed 2019-03-19 by Alvin1Zhang