espressif/esp-idf #3084: HID : sometimes ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT not received

Marc-Aurele, 2019-02-21:
Hello,

Sometimes, when i connect the device to an Iphone, i don't receive this event : ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT in my gap_event_handler callback. Then, when this happens, every event i send to the phone seems to not be interpreted.

If you have any idea, don't hesistate to share.

Regards,

Aurélien

Weijian-Espressif, 2019-02-22:
@Marc-Aurele please provide your test code, thanks.

Marc-Aurele, 2019-02-22:
Hi @Weijian-Espressif 

Here is the case when it works : 

```
I (1704) HID_LE_PRF: HID service added to the stack database. hid svc handle = 32
I (1714) BLE_TASK: gap_event_handler (event : 0)
I (1724) HID_LE_PRF: SERVICE_START_EVT, status 0, service_handle 40
I (1724) HID_LE_PRF: SERVICE_START_EVT, status 0, service_handle 45
I (1734) HID_LE_PRF: SERVICE_START_EVT, status 0, service_handle 50
I (1734) BLE_TASK: gap_event_handler (event : 6)
I (1744) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 
I (1754) BLE_TASK: GPIO BJB task started !
I (2254) HID_LE_PRF: HID connection establish, conn_id = 0
I (2254) BLE_TASK: ESP_HIDD_EVENT_BLE_CONNECT - 0
I (2314) BLE_TASK: update connection params status = 15, min_int = 6, max_int = 6,conn_int = 24,latency = 0, timeout = 72
I (2324) BLE_TASK: gap_event_handler (event : 20)
I (3684) BLE_TASK: gap_event_handler (event : 9)
I (3684) BLE_TASK: gap_event_handler (event : 9)
I (3784) BLE_TASK: gap_event_handler (event : 9)
I (3994) BLE_TASK: gap_event_handler (event : 9)
I (3994) BLE_TASK: remote BD_ADDR: 6a166b75d50a
I (3994) BLE_TASK: address type = 1
I (4004) BLE_TASK: pair status = success
I (4004) BLE_TASK: gap_event_handler (event : 8)
I (4054) BLE_TASK: update connection params status = 15, min_int = 24, max_int = 24,conn_int = 24,latency = 0, timeout = 72
I (4054) BLE_TASK: gap_event_handler **(event : 20)**
I (4394) BLE_TASK: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 12,latency = 4, timeout = 100
I (4394) BLE_TASK: gap_event_handler **(event : 20)**
```
We can see the event 20 (ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT). In that case all is working and the phone is interpreting each event sent. 

Here are the logs when the issue occurs : 

```
I (1644) BLE_TASK: gap_event_handler (event : 0)
I (1654) HID_LE_PRF: SERVICE_START_EVT, status 0, service_handle 40
I (1654) HID_LE_PRF: SERVICE_START_EVT, status 0, service_handle 45
I (1664) HID_LE_PRF: SERVICE_START_EVT, status 0, service_handle 50
I (1674) BLE_TASK: gap_event_handler (event : 6)
I (1674) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 
I (1684) BLE_TASK: GPIO BJB task started !
I (19254) HID_LE_PRF: HID connection establish, conn_id = 0
I (19254) BLE_TASK: ESP_HIDD_EVENT_BLE_CONNECT - 0
I (21184) BLE_TASK: gap_event_handler (event : 9)
I (21184) BLE_TASK: gap_event_handler (event : 9)
I (21464) BLE_TASK: gap_event_handler (event : 9)
I (21464) BLE_TASK: gap_event_handler (event : 9)
I (21464) BLE_TASK: remote BD_ADDR: 69578bd8d722
I (21474) BLE_TASK: address type = 1
I (21474) BLE_TASK: pair status = success
I (21484) BLE_TASK: gap_event_handler (event : 8)
```
(nothing more if i do nothing)

In that case, each event that will be sent to the phone won't be interpreted.

I have noticed that i can easily reproduce the issue by pushing the Boot button. I'm using this button to send events to the phone once the connection is done. On an iphone i launch a pairing process, when i get the popup message on the phone, i push the boot button, and in that case , i have the issue. Once i have clicked "pair" on the popup, all the events that follow and that are sent  to the phone won't be interpreted !

Regards,

Marc-Aurele, 2019-02-26:
Hi,

Any help with this issue

Weijian-Espressif, 2019-02-27:
@Marc-Aurele, We found two bugs:
1.  Update parameters are not allowed during Apple HID encryption. 
`I (2314) BLE_TASK: update connection params status = 15, min_int = 6, max_int = 6,conn_int = 24,latency = 0, timeout = 72`
`I (4054) BLE_TASK: update connection params status = 15, min_int = 24, max_int = 24,conn_int = 24,latency = 0, timeout = 72`

These two parameter updates are initiated by our slave side, we can close these two connection parameter updates.

2. We found a bug in Apple. After Apple HID is successfully encrypted, it will automatically initiate update parameters , then salve can send events to iphone, and can be interpreted by iphone .
you can see `I (4394) BLE_TASK: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 12,latency = 4, timeout = 100`

When the Apple mobile phone pairing box pops up, you click slowly. Even if the encryption is successful, Apple will not actively update the parameters, and then all the events that sent to the iphone won't be interpreted !

This patch is to fix the first bug. The second bug is an Apple phone issue, and we haven't found a solution right now. When second bug occurs, you can only reconnect.

[hid.patch.zip](https://github.com/espressif/esp-idf/files/2908398/hid.patch.zip)

Marc-Aurele, 2019-02-27:
Hello @Weijian-Espressif 

Thank you very much for these informations. I applied your patch but i'm still having a lot of instability at connection and i guess it is related to the point 2) in your previous post.

But, i have made a modification that seems to have solve the issue (or maybe hidden). Indeed, in the current HID example project, in the following file : hid_device_le_prf.c i have changed every ESP_GATT_PERM_READ in ESP_GATT_PERM_READ_ENCRYPTED and every ESP_GATT_PERM_WRITE in ESP_GATT_PERM_WRITE_ENCRYPTED. 
I only see the following error in the log :

>  E (60127) BT_GATT: GATT_INSUF_ENCRYPTION

but now, each time i try to connect i can see the ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT event to come and everything seems to work.

Weijian-Espressif, 2019-02-28:
Hi @Marc-Aurele 
You only need to modify `// Report KEY INPUT Characteristic - Client Characteristic Configuration Descriptor
    [HIDD_LE_IDX_REPORT_CC_IN_CCC]              = {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_client_config_uuid,
                                                                      (ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE_ENCRYPTED),
                                                                      sizeof(uint16_t), 0,
                                                                      NULL}},`

Weijian-Espressif, 2019-02-28:
After the HID is connected, the iPhone writes 1 to the Report Characteristic Declaration, even if the encryption is not completed. This should actually be written  1 after the encryption is completed, that is the root cause. So you can modify the permissions of the Report Characteristic Declaration to `
ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE_ENCRYPTED`.

Marc-Aurele, 2019-02-28:
Hi @Weijian-Espressif 

Indeed, only modifying HIDD_LE_IDX_REPORT_CC_IN_CCC write permission seems to have solved my issue butnow in the log i can see : 

`E (39907) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_ENCRYPTION`

Do i need to care about this error ?

Regards,

Weijian-Espressif, 2019-02-28:
@Marc-Aurele 
Don't need to care about this error, this means that the HID encryption has not been completed when iPhone writing data.

Marc-Aurele, 2019-02-28:
Ok, thank you very much for your help. i will test this modification with all iphone we have and hope it will be more stable, at least for the iphone 5 it works very well !

Weijian-Espressif, 2019-02-28:
@Marc-Aurele,
Win10 does not support vendor report, you need to delete the vendor report related code.
I am modifying these codes, them will be merged into IDF, master , v3.2 and v3.1.

Marc-Aurele, 2019-02-28:
Thx again for the support, for now we only want a compatiblity with as many phone as possible. But i will follow your advise and remove the vendor report.

Marc-Aurele, 2019-03-01:
> @Marc-Aurele,
> Win10 does not support vendor report, you need to delete the vendor report related code.
> I am modifying these codes, them will be merged into IDF, master , v3.2 and v3.1.

@Weijian-Espressif Will you merge in v3.2 and v3.1 the patch you gave me in your previous post ? Do you have a date scheduled ? (this is just to not forget to rebase our github fork)

Weijian-Espressif, 2019-03-04:
@Marc-Aurele, they will be  merged into IDF this week.

Alvin1Zhang, 2019-03-06:
Will keep this open until our fix is merged.

Marc-Aurele, 2019-03-12:
hi @Weijian-Espressif 
I saw this patch : https://github.com/espressif/esp-idf/commit/1b5a77eba9432cc25c0cf5e95cdd3659a8f23c72
has been merged into v3.1 but not in v3.2. On our side we are working on v3.1 but we are planing to move to v3.2. Is this patch will be merged into v3.2 also ?

Regards,

Aurélien

Weijian-Espressif, 2019-03-12:
@Marc-Aurele, 
It will be merged into v3.2, but still need to wait a few days.

Alvin1Zhang, 2019-05-17:
@Marc-Aurele Thanks for reporting the issue. The fix has been available. Feel free to reopen if your issue still exists. Thanks

closed 2019-05-17 by Alvin1Zhang