espressif/esp-idf #3085: aws_iot: failed! mbedtls_net_connect returned -0x52 on ESP-IDF

zafersn, 2019-02-21:
Hi. I've looked at questions like this, but it's not a solution for me. 

let me explain briefly.

I'm using ESP32 in apsta mode. So both AP mode and STA mode together. I checked the internet connection on esp32 and successfully send and receive data. But when I try to link to aws_iot, I get the following error.
I mentioned the same issue in this link: https://github.com/tonyp7/esp32-wifi-manager/issues/31
I have successfully tested the same aws_iot codes in another project before. I made the change only in wifimanger codes.
Thanks for your help.
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6264
load:0x40078000,len:11164
load:0x40080400,len:6672
entry 0x40080764
I (29) boot: ESP-IDF v3.3-beta1-328-gabea9e4c0-dirty 2nd stage bootloader
I (29) boot: compile time 19:51:45
I (39) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2b230 (176688) map
I (156) esp_image: segment 1: paddr=0x0003b258 vaddr=0x3ffb0000 size=0x0300c ( 12300) load
I (161) esp_image: segment 2: paddr=0x0003e26c vaddr=0x40080000 size=0x00400 (  1024) load
I (163) esp_image: segment 3: paddr=0x0003e674 vaddr=0x40080400 size=0x0199c (  6556) load
I (174) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x88240 (557632) map
I (376) esp_image: segment 5: paddr=0x000c8260 vaddr=0x40081d9c size=0x0f184 ( 61828) load
I (412) boot: Loaded app from partition at offset 0x10000
I (412) boot: Disabling RNG early entropy source...
I (412) cpu_start: Pro cpu up.
I (416) cpu_start: Application information:
I (421) cpu_start: Project name:     tcp_client
I (426) cpu_start: App version:      1
I (430) cpu_start: Compile time:     19:52:01
I (435) cpu_start: Compile date:     Feb 15 2019
I (441) cpu_start: ESP-IDF:          v3.3-beta1-328-gabea9e4c0-dirty
I (448) cpu_start: Starting app cpu, entry point is 0x40081028
I (0) cpu_start: App cpu up.
I (458) heap_init: Initializing. RAM available for dynamic allocation:
I (465) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (471) heap_init: At 3FFB9F80 len 00026080 (152 KiB): DRAM
I (477) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (484) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (490) heap_init: At 40090F20 len 0000F0E0 (60 KiB): IRAM
I (496) cpu_start: Pro cpu start user code
I (179) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
http_server: waiting for start bit
wifi_manager_fetch_wifi_sta_config: ssid:SUPERONLINE-WiFi_0668 password:FTYFRMHNXYLU
wifi_manager_fetch_wifi_settings: SoftAP_ssid:esp32
wifi_manager_fetch_wifi_settings: SoftAP_pwd:esp32pwd
wifi_manager_fetch_wifi_settings: SoftAP_channel:5
wifi_manager_fetch_wifi_settings: SoftAP_hidden (1 = yes):0
wifi_manager_fetch_wifi_settings: SoftAP_bandwidth (1 = 20MHz, 2 = 40MHz)1
wifi_manager_fetch_wifi_settings: sta_only (0 = APSTA, 1 = STA when connected):1
wifi_manager_fetch_wifi_settings: sta_power_save (1 = yes):1
wifi_manager_fetch_wifi_settings: sta_static_ip (0 = dhcp client, 1 = static ip):0
wifi_manager_fetch_wifi_settings: sta_static_ip_config: IP: 255.255.255.0 , GW: 255.255.255.0 , Mask: 255.255.255.0
wifi_manager_fetch_wifi_settings: sta_ip_addr: 192.168.0.10
wifi_manager_fetch_wifi_settings: sta_gw_addr: 192.168.0.1
wifi_manager_fetch_wifi_settings: sta_netmask: 255.255.255.0
wifi_manager: saved wifi found on startup
wifi_manager: Start DHCP client for STA interface. If not already running
AWS IoT SDK Version 3.0.1-
AWS init ...

free heap: 243340
I (482) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (482) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1592) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0
wifi_manager: starting softAP with ssid esp32
wifi_manager: starting softAP with 20 MHz bandwidth
wifi_manager: starting softAP on channel 5
wifi_manager: STA power save enabled
wifi_mamager: softAP started, starting http_server
http_server: received start bit, starting server
HTTP Server listening...
I (3862) event: sta ip: 192.168.1.24, mask: 255.255.255.0, gw: 192.168.1.1
wifi_manager: About to save config to flash
wifi_manager_wrote wifi_sta_config: ssid:SUPERONLINE-WiFi_0668 password:FTYFRMHNXYLU
wifi_manager_wrote wifi_settings: SoftAP_ssid: esp32
wifi_manager_wrote wifi_settings: SoftAP_pwd: esp32pwd
wifi_manager_wrote wifi_settings: SoftAP_channel: 5
wifi_manager_wrote wifi_settings: SoftAP_hidden (1 = yes): 0
wifi_manager_wrote wifi_settings: SoftAP_bandwidth (1 = 20MHz, 2 = 40MHz): 1
wifi_manager_wrote wifi_settings: sta_only (0 = APSTA, 1 = STA when connected): 1
wifi_manager_wrote wifi_settings: sta_power_save (1 = yes): 1
wifi_manager_wrote wifi_settings: sta_static_ip (0 = dhcp client, 1 = static ip): 0
wifi_manager_wrote wifi_settings: sta_ip_addr: 192.168.0.10
wifi_manager_wrote wifi_settings: sta_gw_addr: 192.168.0.1
wifi_manager_wrote wifi_settings: sta_netmask: 255.255.255.0
aws_start_Bit
Connecting to AWS...
free heap: 207976

free heap: 207972

free heap: 207976
E (17362) aws_iot: failed! mbedtls_net_connect returned -0x52
Error(-23) connecting to xxxxxx-ats.iot.us-east-2.amazonaws.com:8883
```

mahavirj, 2019-02-22:
@zafersn At-least from error `-0x52` it appears to be host name resolution issue, I wonder if it is temporary failure and or is it reproducible across multiple attempts to connect to aws iot endpoint?

zafersn, 2019-02-22:
hi @mahavirj thanks for answer.  "or is it reproducible across multiple attempts to connect to aws iot endpoint?" <== here I do not fully understand what you mean.  I'm getting this error when I try to use esp in "apsta" mode. only sta mode is successfully connecting to aws.

mahavirj, 2019-02-23:
@zafersn Yes, I understand that. What I was thinking, with APSTA mode, for both interfaces channel also  needs to be same, if not then it will internally get adjusted (softAP moved to sta channel), so maybe host name resolution failure happening during this adjustment period (guess though).

Anyways you may provide with patch on `aws-iot` examples in IDF that can reproduce this issue, and we will take a look.

zafersn, 2019-02-23:
@mahavirj . did you have the opportunity to review the wifimanager project on this link? : https://github.com/tonyp7/esp32-wifi-manager

Using the wifimanager in my project I want to do the following:
 esp32-wifi-manager is an esp32 program that enables easy management of wifi networks through a web application.

zafersn, 2019-02-25:
Can you help me? @mahavirj

zafersn, 2019-02-26:
Hi, I'm still looking for a solution to the problem.  as I've made some progress. like I'm getting a little closer to the solution. I've read and applied a solution for the "mbedtls_net_connect returned -0x52" error. 
https://stackoverflow.com/a/46749602

It is said that the problem may be the static IP configuration and the missing or incorrect setting of the DNS settings. When I examined the wifimanager configuration, I observed that there was a static IP configuration available.

I have no knowledge about network configuration, so I tried to switch off and try static ip configurations in wifi configuration settings in wifimanager.c and the result was successful. I am no longer receiving the error "aws_iot: failed! Mbedtls_net_connect returned -0x52" . But now after receiving the message "autoreconnect ... Subscribing ... Subscribing Success", I get an error like "MQTT DisconnectAuto Reconnect is Enabled," when I wait before publishing a message. If I want to post a message, I get an error such as "E (58050) aws_iot: failed! Mbedtls_ssl_write returned -0x50".

- How do you think this static ip and dns configuration should be?
- static ip and dns configuration (delete for this project) how it solved this error: "aws_iot: failed! Mbedtls_net_connect returned -0x52" .
- What are your suggestions for the "E (58050) aws_iot: failed! Mbedtls_ssl_write returned -0x50" error?

Alvin1Zhang, 2022-10-10:
Thanks for reporting and sorry for the slow turnaround, please try on esp-aws-iot and report if its still valid. Feel free to reopen.

closed 2022-10-10 by Alvin1Zhang