espressif/esp-idf #3086: Build System: Process file before embeding

itay-grudev, 2019-02-21:
I would like to embed a file, but I would like for it to be pre-processed first.

For example before embedding an HTML file one might want to minify it first. Or maybe concatenate several files into one before embedding them.

I imagine it to work sth like this:

```make
COMPONENT_EMBED_FILES := html/index.html
html/index.html:
    some_preprocessor_with_params
```

Is that possible with the current build system?

krzychb, 2019-02-22:
Hi @itay-grudev,

Have you checked [Source Code Generation](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system-cmake.html#source-code-generation) under **Build System (CMake)** documentation?

itay-grudev, 2019-02-23:
Yes I have. There just isn't an easy way at the moment I think. I will try extending the `esp-idf`Makefile to see if I can add the functionality.

atanisoft, 2020-01-31:
@itay-grudev were you able to find a way to pre-process the files before embedding? I'm considering how to do this to gzip compress the html/css/js files before embedding (it saves ~75kb roughly)

atanisoft, 2020-02-01:
While it would be great if the ```target_add_binary_data``` utility would allow pre-processing there is a way to do this using the following:
```
FIND_PROGRAM(GZIP_TOOL
  NAMES gzip
  PATHS /bin
        /usr/bin
        /usr/local/bin
)

IF(NOT GZIP_TOOL)
  MESSAGE(FATAL_ERROR "Unable to find 'gzip' program")
ENDIF(NOT GZIP_TOOL)

add_custom_command(OUTPUT "${CMAKE_CURRENT_SOURCE_DIR}/data/index.html.gz"
    COMMAND ${GZIP_TOOL} -fk ${CMAKE_CURRENT_SOURCE_DIR}/data/index.html
    VERBATIM)
set_property(TARGET ${CMAKE_PROJECT_NAME}.elf APPEND PROPERTY ADDITIONAL_MAKE_CLEAN_FILES
    "${CMAKE_CURRENT_SOURCE_DIR}/data/index.html.gz")

target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "${CMAKE_CURRENT_SOURCE_DIR}/data/index.html.gz" BINARY)
```

itay-grudev, 2020-02-02:
@atanisoft Nope. I ended up flashing assets them in an uncompressed state just because it's not easy. Your solution looks interesting and If I ever revise my project I'd consider it.

renzbagaporo, 2020-04-22:
Hi @itay-grudev, we'll take this into consideration. Note that @atanisoft's suggestion is pretty much how you would do it right now.

dobairoland, 2023-08-27:
This has probably been solved by aed204ce6f6e3924dcda0507b87af18eb17ffea0 a long time ago.

closed 2023-08-27 by dobairoland