espressif/esp-idf #3087: HID : W (6663) BT_L2CAP: l2cble_start_conn_update, the last connection update command still pending.

Marc-Aurele, 2019-02-22:
Hello,

With one particular phone i get these kind of message when i use the HID example project : 

```
I (5093) BLE_TASK: ESP_HIDD_EVENT_BLE_DISCONNECT
I (5103) BLE_TASK: ------> 6
I (6303) HID_LE_PRF: HID connection establish, conn_id = 0
I (6303) BLE_TASK: ESP_HIDD_EVENT_BLE_CONNECT - 0
W (6663) BT_L2CAP: l2cble_start_conn_update, the last connection update command still pending.
I (6673) BLE_TASK: remote BD_ADDR: 418679e02776
I (6673) BLE_TASK: address type = 1
I (6673) BLE_TASK: pair status = fail
E (6673) BLE_TASK: fail reason = 0x51
I (6683) BLE_TASK: ------> 8
I (6893) BLE_TASK: update connection params status = 0, min_int = 6, max_int = 6,conn_int = 6,latency = 0, timeout = 500
I (6893) BLE_TASK: ------> 20
I (7893) BLE_TASK: update connection params status = 16, min_int = 36, max_int = 36,conn_int = 6,latency = 0, timeout = 500
I (7893) BLE_TASK: ------> 20
W (9723) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB

E (9723) BT_BTM: Device not found

W (9723) BT_APPL: BTA got unregistered event id 31

W (9723) BT_APPL: BTA got unregistered event id 31
```
On the phone side, i get this message : Coudn't pair with device because of an incorrect PIN or passkey.

If you have any idea, feel free to share.

Regards,

Aurélien

Weijian-Espressif, 2019-02-25:
@Marc-Aurele, your pairing was not successful. The function of HID must be used after pairing is successful.

Marc-Aurele, 2019-02-25:
@Weijian-Espressif In that case, i do not use any function, i just get this message but it happened only once on a particular phone. I think you make reference to this other issue : https://github.com/espressif/esp-idf/issues/3084

Weijian-Espressif, 2019-02-27:
please refer #3084

closed 2019-03-13 by Alvin1Zhang