espressif/esp-idf #3088: esp-mqtt: receiving messages with empty payload

tsichevski, 2019-02-22:
I know how to fix the problem, so I skipped some forensics below.

The cure is:

mqttclient.c:L552: the following condition must be removed, and the code block should be executed unconditionally:
if (total_mqtt_len != 0) {
   ... do dispatching
}

## Environment

- Development Kit:      [Whatever]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): 
v3.3-beta1-328-gabea9e4c0
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Linux]
- Power Supply:         [external 3.3V]

## Problem Description

Library ignores incoming messages with empty payload, but it should not.

### Expected Behavior

Should dispatch all incoming messages.

### Actual Behavior

Ignores all incoming messages if their payload is empty.

### Steps to repropduce

1. Subscribe to some topic
2. Send a message by any means with a matching topic and no (zero-length) payload

### Code to reproduce this issue

Not needed.

gregjesl, 2019-02-25:
Thanks for posting a fix @tsichevski; I was having a problem with this as well.

david-cermak, 2019-03-04:
Hi @tsichevski 

Thank you for reporting this and posting your fix. Will check if it doesn't break anything and apply the changes.
Thanks!

gregjesl, 2019-03-29:
@david-cermak Has the EspressIf team made any progress on this?  If not, let me know and I can submit a pull request.

david-cermak, 2019-03-29:
I'm sorry @gregjesl, no update from Espressif team. Please feel free (anybody) to file a PR.

gregjesl, 2019-03-29:
@david-cermak You added
```cpp
if (total_mqtt_len != 0) {
... do dispatching
}
```
in https://github.com/espressif/esp-mqtt/commit/cf5b8eda89057f5eb49654d2eaf4d1a209af2d11.  It looks like you added it for a reason; will moving the contents of the `if` block outside of the `if` block cause the issue you patched to reoccur?

david-cermak, 2019-03-30:
@gregjesl Yes, indeed, removing that condition would break the fix. It was actually oversight from my side to (mis)use `total_mqtt_len` as an indication of correct (complete) mqtt message in recv buffer. Correct fix should employ a separate state variable indicating both `mqtt_get_publish_topic` and `mqtt_get_publish_data` returned non-nullptr.

gregjesl, 2019-04-01:
@david-cermak Something like this? ([here](https://github.com/espressif/esp-mqtt/blob/b1aa1d444ef4a0538758269f1ede2673944783c4/mqtt_client.c#L531-L601))
```cpp
#define MESSAGE_STATE_INCOMPLETE 0
#define MESSAGE_STATE_COMPLETE 1

static void deliver_publish(esp_mqtt_client_handle_t client, uint8_t *message, int length)
{
    ...
    int message_state = MESSAGE_STATE_INCOMPLETE;
    ...
    do
    {
        if (message_state == MESSAGE_STATE_INCOMPLETE) {
            /* any further reading only the underlying payload */
            transport = esp_transport_get_payload_transport_handle(transport);
            mqtt_data_length = mqtt_topic_length = length;
            if (NULL == (mqtt_topic = mqtt_get_publish_topic(message, &mqtt_topic_length)) ||
                NULL == (mqtt_data = mqtt_get_publish_data(message, &mqtt_data_length)) ) {
                /* Message is still incomplete */
                ...
            } else {
                /* Message is complete */
                message_state = MESSAGE_STATE_COMPLETE;
                ...
            }
        } else {
            ...
        }

        if (message_state == MESSAGE_STATE_COMPLETE) {
            ...
            esp_mqtt_dispatch_event(client);
        }
        ...
    } while (1);

}
```

david-cermak, 2019-04-01:
@gregjesl Yes, this should work

gregjesl, 2019-04-03:
@david-cermak I've submitted [a pull request](https://github.com/espressif/esp-mqtt/pull/110)

david-cermak, 2019-10-10:
Resolved in https://github.com/espressif/esp-idf/commit/6289a26596309f3aa4761d383203dfdc74a8eb50 (which included mqtt commit https://github.com/espressif/esp-mqtt/commit/0450bd0093978b8e6a7bd78e5b278dea9bbe212b)

Alvin1Zhang, 2019-10-10:
@david-cermak Thanks for the updates, and @tsichevski thanks for reporting, feel free to reopen if the issue still exists. Thanks.

closed 2019-10-10 by Alvin1Zhang