espressif/esp-idf #3090: BUG: Mesh and smart config doesn't work

leenowell, 2019-02-23:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):     * v3.3-beta1-223-ga62cbfe
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):     * 1.22.0-73-ge28a011
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

There appears to be an interoperability issue when using smart config with Mesh.  When the mesh detects "no parent" (due to incorrect SSID) I start the smart config process in a new task as per the smart config example.  When it receives the SSID and password from smart config (SC_STATUS_LINK event) I call esp_mesh_set_router with the new details and it appears to hang (no further output)

I raised this issue on the forum, together with a workaround here
https://www.esp32.com/viewtopic.php?f=21&t=9092

### Expected Behavior
Mesh to connect to the router with the new credentials

### Actual Behavior
Appears to hang with no further output
### Steps to repropduce

1. Start the smart config process in the NoParent event callback
2. Call the set router when smart config gets the

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue
Code is included in the forum discussion here
https://www.esp32.com/viewtopic.php?f=21&t=9092

```cpp
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs
In the forum discussion here
https://www.esp32.com/viewtopic.php?f=21&t=9092
```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2019-03-18:
@Yulong-espressif Any updates? Thanks.

Alvin1Zhang, 2019-03-18:
@dongdong004 Any updates? Thanks.

jack0c, 2022-01-05:
This bug should be fixed after version v3.3

closed 2022-01-05 by jack0c