espressif/esp-idf #3092: BUG In coap server example

leenowell, 2019-02-23:
## Environment
* Development Kit:      ESP32-DevKitC

* Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]

* Module or chip used:  [ESP32-WROOM-32|

* IDF version (run `git describe --tags` to find it): v3.3-beta1-223-ga62cbfe

* Build System:         [Make]

* Compiler version (run `xtensa-esp32-elf-gcc --version` to find it): 1.22.0-73-ge28a011

* Operating System:     [Linux]

* Power Supply:         [USB]

## Problem Description

The message received by the coap client has random characters after the "Hello World!" message sent because the /0 was not sent by the server

### Expected Behavior
Should send "Hello World!\0"

### Actual Behavior
Sends a non terminated string

### Steps to reproduce

### Code to reproduce this issue

```cpp

the line 
    coap_add_data  (response, strlen(response_data), (unsigned char *)response_data);

should be replaced by
    coap_add_data  (response, strlen(response_data)+1, (unsigned char *)response_data);

```

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

mahavirj, 2019-04-05:
@leenowell Can you please check once on latest `master`? Commit https://github.com/espressif/esp-idf/commit/77e7af4e902b2777ff3521c554fadd1b4a25583e should address the issue.

mahavirj, 2019-05-15:
@leenowell Do you have any update here? (CC @jitin17)

mahavirj, 2019-05-17:
Closing this issue, please feel free to reopen in case it is still present with latest IDF.

closed 2019-05-17 by mahavirj