espressif/esp-idf #3099: I2C interrupt

timkoers, 2019-02-25:
This is the code

```
/* i2c - Example

   For other examples please check:
   https://github.com/espressif/esp-idf/tree/master/examples

   See README.md file to get detailed usage of this example.

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "esp_log.h"
#include "driver/i2c.h"
#include "soc/i2c_reg.h"

#include "sdkconfig.h"

static const char *TAG = "i2c-example";

#define _I2C_NUMBER(num) I2C_NUM_##num
#define I2C_NUMBER(num) _I2C_NUMBER(num)

#define DATA_LENGTH 512                  /*!< Data buffer length of test buffer */
#define RW_TEST_LENGTH 128               /*!< Data length for r/w test, [0,DATA_LENGTH] */
#define DELAY_TIME_BETWEEN_ITEMS_MS 1000 /*!< delay time between different test items */

#define I2C_SLAVE_SCL_IO CONFIG_I2C_SLAVE_SCL               /*!< gpio number for i2c slave clock */
#define I2C_SLAVE_SDA_IO CONFIG_I2C_SLAVE_SDA               /*!< gpio number for i2c slave data */
#define I2C_SLAVE_NUM I2C_NUMBER(CONFIG_I2C_SLAVE_PORT_NUM) /*!< I2C port number for slave dev */
#define I2C_SLAVE_TX_BUF_LEN (2 * DATA_LENGTH)              /*!< I2C slave tx buffer size */
#define I2C_SLAVE_RX_BUF_LEN (2 * DATA_LENGTH)              /*!< I2C slave rx buffer size */

#define ESP_SLAVE_ADDR CONFIG_I2C_SLAVE_ADDRESS /*!< ESP32 slave address, you can set any 7bit value */
SemaphoreHandle_t print_mux = NULL;

static xQueueHandle i2c_event_queue = NULL;

intr_handle_t *i2c_slave_intr_handle = NULL;

/**
 * @brief test function to show buffer
 */
static void disp_buf(uint8_t *buf, int len)
{
    int i;
    for (i = 0; i < len; i++) {
        printf("%02x ", buf[i]);
        if ((i + 1) % 16 == 0) {
            printf("\n");
        }
    }
    printf("\n");
}

static void i2c_handle_interrupt(void *arg){
	
	while(1){
	
		xSemaphoreTake(print_mux, portMAX_DELAY);
	
		int arg_pointer = 0;
	
		BaseType_t ret = xQueueReceive(i2c_event_queue, &arg_pointer, 1000 / portTICK_RATE_MS);
		
		if(ret != pdTRUE){
			continue;
		}
		
		ESP_LOGI(TAG, "Found new I2C event to handle");
		
		int size;
		
		uint8_t *data = (uint8_t *)malloc(RW_TEST_LENGTH);
		
		size = i2c_slave_read_buffer(I2C_SLAVE_NUM, data, 16, 1000 / portTICK_RATE_MS);
		
		if(size == -1){
			ESP_LOGW(TAG, "i2c_slave_read_buffer returned -1");
		}else{
			ESP_LOGI(TAG, "Master told me that there are a few bytes comming up");
		}
		
		disp_buf(data, size);
		
		// This is the data length
		int data_length = atoi((char*)data);
		
		size = i2c_slave_read_buffer(I2C_SLAVE_NUM, data, data_length, 1000 / portTICK_RATE_MS);
		
		if(size != data_length){
			ESP_LOGW(TAG, "I2C expected data length vs read data length does not match");
		}else{
			disp_buf(data, size);
		}
		
		xSemaphoreGive(print_mux);
		
		vTaskDelay((DELAY_TIME_BETWEEN_ITEMS_MS * (1000 + 1)) / portTICK_RATE_MS);
	}
}

void IRAM_ATTR i2c_interrupt(){
	ets_printf("i2c_interrupt has been triggered\n");
	ets_printf("interupt number %d", esp_intr_get_intno(i2c_slave_intr_handle));
	uint32_t unusedVar = 0; 	
	xQueueSendFromISR(i2c_event_queue, &unusedVar, NULL);	
}

/**
 * @brief i2c slave initialization
 */
static esp_err_t i2c_slave_init()
{
	i2c_event_queue = xQueueCreate(2, sizeof(uint32_t));
	
	xSemaphoreTake(print_mux, portMAX_DELAY);
    int i2c_slave_port = I2C_SLAVE_NUM;
    i2c_config_t conf_slave;
    conf_slave.sda_io_num = I2C_SLAVE_SDA_IO;
    conf_slave.sda_pullup_en = GPIO_PULLUP_ENABLE;
    conf_slave.scl_io_num = I2C_SLAVE_SCL_IO;
    conf_slave.scl_pullup_en = GPIO_PULLUP_ENABLE;
    conf_slave.mode = I2C_MODE_SLAVE;
    conf_slave.slave.addr_10bit_en = 0;
    conf_slave.slave.slave_addr = ESP_SLAVE_ADDR;
    i2c_param_config(i2c_slave_port, &conf_slave);
	esp_err_t isr_register_ret = i2c_isr_register(i2c_slave_port, i2c_interrupt, 0,0,i2c_slave_intr_handle);
	
	if(isr_register_ret == ESP_OK){
		ESP_LOGI(TAG, "Registered interrupt handler");
	}else{
		ESP_LOGW(TAG, "Failed to register interrupt handler");
	}
	
	xSemaphoreGive(print_mux);
	
    return i2c_driver_install(i2c_slave_port, conf_slave.mode,
                              I2C_SLAVE_RX_BUF_LEN,
                              I2C_SLAVE_TX_BUF_LEN, 0);
}
void app_main()
{
    print_mux = xSemaphoreCreateMutex();
    ESP_ERROR_CHECK(i2c_slave_init());
    //ESP_ERROR_CHECK(i2c_master_init());
    xTaskCreate(i2c_handle_interrupt, "i2c_handle_interrupt", 1024 * 2, (void *)0, 10, NULL);
    //xTaskCreate(i2c_test_task, "i2c_test_task_1", 1024 * 2, (void *)1, 10, NULL);
}
```
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.3-beta1-223-ga62cbfec9
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

Please provide a correct and complete documentation set (including interrupts and such) about I2C, not only how to use some little project that includes both a slave and master.
Having said that, I'm running into the following problems; the ESP-32 crashes when running the code below code.

### Expected Behavior

I am trying to get an interrupt when there is I2C data that needs to be processed (so a **full, ready to use I2C command, that has been fully handled by the system**).

### Actual Behavior

However, I have the feeling that the interrupts trigger on any I2C event, because the monitor get's spammed with 

> i2c_interrupt has been triggered

When the `ets_printf("interupt number %d", esp_intr_get_intno(i2c_slave_intr_handle));` is commented out, since that function call crashes the ESP

### Steps to repropduce

Upload the program and send an I2C request to address 0x28.

### Program

See my [gist](https://gist.github.com/timkoers/d5384da410abec0bbfca8029003ef92b)

## Debug Logs

```bash
i2c_interrupt has been triggered
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400e3caf  PS      : 0x00060031  A0      : 0x80082714  A1      : 0x3ffb0640
0x400e3caf: esp_intr_get_intno at C:/esp-idf/components/esp32/intr_alloc.c:775

A2      : 0x00000000  A3      : 0x00000021  A4      : 0x00000004  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x3ffb5470  A8      : 0x8000814b  A9      : 0x3ffb05b0
A10     : 0x00000000  A11     : 0x00060023  A12     : 0x800869f9  A13     : 0x3ffb5440
A14     : 0x00000003  A15     : 0x00060123  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x400e3caf  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000
0x400e3caf: esp_intr_get_intno at C:/esp-idf/components/esp32/intr_alloc.c:775

Backtrace: 0x400e3caf:0x3ffb0640 0x40082711:0x3ffb0660 0x40082165:0x3ffb0690 0x400e3cbb:0x00000000
0x400e3caf: esp_intr_get_intno at C:/esp-idf/components/esp32/intr_alloc.c:775

0x40082711: i2c_interrupt at C:/Users/Tim/Documents/Projecten/Prive/TINES01/esp32/I2C_Test_Code/main/i2c_example_main.c:252

0x40082165: _xt_lowint1 at C:/esp-idf/components/freertos/xtensa_vectors.S:1105

0x400e3cbb: esp_pm_impl_waiti at C:/esp-idf/components/esp32/pm_esp32.c:487
```

costaud, 2019-02-26:
After a brief glance at your code, I think the basic problem is that you should use `&i2c_slave_intr_handle` instead of i2c_slave_intr_handle in i2c_isr_register().

Besides, I thing you should not post everything from an ISR to a task handler. At least you need to clear the hardware interrupt before exiting ISR(only to handle item that can be delayed in task handler), otherwise, the code would get back in ISR again and again, which would trigger interrupt watchdog.

timkoers, 2019-02-26:
> After a brief glance at your code, I think the basic problem is that you should use `&i2c_slave_intr_handle` instead of i2c_slave_intr_handle in i2c_isr_register().

That is fixed with the latest version of my gist! 
Should I make a call to `i2c_isr_free` everytime a ISR is triggered and re-register the interrupt when the subroutine finishes?

negativekelvin, 2019-02-27:
i2c_driver_install already configures a built in isr handler so I don't think you should try to configure your own.  All you have to do is have a high priority task blocked on i2c_slave_read_buffer and the built in isr will unblock it for you when a transaction is complete or the buffer is full.

timkoers, 2019-02-28:
But what if I want to do it non-blocking, by the usage of Queues as shown in the gist?

negativekelvin, 2019-02-28:
You're still blocking on a queue rather than a ringbuffer but if it is useful to you then copy the i2c driver to your project and modify it.

timkoers, 2019-02-28:
So how would one implement blocking on a ringbuffer? Call the i2c_slave_read_buffer from the task and set the wait time to a value greater than 0?

closed 2023-10-19 by costaud