espressif/esp-idf #3100: Partition API

pavelskirko, 2019-02-25:
## Environment

- Development Kit:  custom
- Module or chip used:  ESP32-WROOM-32
- IDF version:  v3.2 stable
- Build System:  Make
- Compiler version:n 1.22.0-80-g6c4433a5
- Operating System:   Windows 10
- Power Supply: external 5V

## Problem Description

I cannot write to flash memory using the same code, that worked earlier. Investigating into problem show me that behavior actually very unpredictable: sometimes it can write several time, sometimes it even cannot find partition.

It worked without any problems earlier.

Does it mean flash memory get damaged? How is it reliable?

### Expected Behavior

1) esp_partition_find_first() returns a pointer to a structure with real properties of my custom partition in flash memory. (I'm using custom partition table with additional data 2M) 
2) data are written to this partition using esp_partition_write()
3) data are read using esp_partition_read()

### Actual Behavior

1) sometimes esp_partition_find_first() returns a pointer to a structure with size = 0 and other default properties:
[image]
2) sometimes writing is successful but I get the same empty partition pointer after several cycles
And then It crashes because (size=0) < size of my buffer.
3) reading just isn't happening

### Steps to repropduce

actually don't know what to put here...

### Code to reproduce this issue

```cpp
void app_main()
{
    uint8_t buf[60];
    partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_ANY, "storage");
	esp_partition_erase_range(partition, 0, partition->size);
    for(uint8_t i = 0; i<10; i++)
    {
        esp_partition_write(partition, (sizeof(buf))*i, buf, sizeof(buf));
    }
}

```
[here](github.com/pavelskirko/frog/tree/latest) is git of this project(this issue corresponds to spi.c file, lines 182 and 259)

partition table:
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,        data, nvs,      0x9000,  0x6000,
phy_init,   data, phy,      0xf000,  0x1000,
factory,    app,  factory,  0x10000, 1M,
storage,    data, ,             , 2M,
```

## Debug Logs

attached:
[gdb_log.txt](https://github.com/espressif/esp-idf/files/2902297/gdb_log.txt)

KonstantinKondrashov, 2019-02-26:
Hi @pavelskirko. Probably you have corruption of `partition`. I build your project and look at `Frog3.map`:
```
 COMMON         0x3ffb6b20       0x1c C:/msys32/home/virtpc/esp/frog/build/main\libmain.a(main.o)
                0x3ffb6b20                wifi_event_group
                0x3ffb6b24                xTaskToNotify
                0x3ffb6b28                SpiEventGroup
                0x3ffb6b2c                button_task_handle
                0x3ffb6b30                data_size   < - Can rewrite the partition.
                0x3ffb6b38                partition
```

`data_size` is array uint8_t data_size[6]; 

Inside your code `pb_get_encoded_size(&data_size[NUM_OF_FIELDS + num2 + i], Accel_fields, &a1[i]);`
where `data_size[3+2+3]`. I think this place of over writing the next located in memory the `partition` varible.

Thanks. Check this point.

pavelskirko, 2019-02-27:
Thank you! It helped.

closed 2019-02-27 by pavelskirko