espressif/esp-idf #3103: cc.h File location guide

Goldenstarc, 2019-02-27:
Hi,
in ⁨esp-idf⁩ ▸ ⁨components⁩ ▸ ⁨lwip⁩ ▸ ⁨port⁩ ▸ ⁨esp32⁩ ▸ ⁨include⁩ ▸ ⁨arch⁩ ▸ cc.h
Line 22 there is 
> #include "ports/unix/port/include/arch/cc.h"

where is this location ?!

projectgus, 2019-03-12:
Hi @Goldenstarc ,

The [LWIP repo](http://github.com/espressif/esp-lwip/) under `components/lwip/lwip` is mostly a clone of the [upstream LWIP project](http://savannah.nongnu.org/projects/lwip/). So we don't modify all of the non-core pieces of code in this repo, only the parts we use.

In this case, if updating the test code for ESP-IDF you can probably change this to something like "#include "lwip/port/esp32/include/arch/cc.h" or similar to include this port-specific file: https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/include/arch/cc.h

projectgus, 2019-03-12:
BTW, the best place to ask questions like this is the esp32.com forum: https://esp32.com

closed 2019-03-12 by projectgus