espressif/esp-idf #3105: NVS Partition Generator KeyError: 'value'

asweeney99, 2019-02-27:
## Environment
- IDF version:  v3.2-beta1-182-g4c881708d
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: Linux

## Problem Description
I'm following the recommendation [here](https://github.com/espressif/esp-idf/issues/1497#issuecomment-432106269) to use NVS storage to store AWS certificates in flash.
The NVS partition generator returns an error when checking the value of the file to be encoded. 
I think my certificate file is not a valid String. How can I make the utility accept the certificate file format?

### Expected Behavior
The utility should generate a bin file with the AWS certificate which can be flashed to an esp32 partition.

### Actual Behavior
AN empty bin file is generated and a stack trace with the error message:
KeyError: 'value'

### Steps to reproduce
1. Generate AWS certificate.
2. Null terminate the content of the certificate with '\0'
```
-----BEGIN CERTIFICATE-----
NOTARealCertIfICAtEAMPD7uM9E2FC7WcXkVQvjJ818Nr+MA0GCSqGSIb3DQEB
CwUAME0xSzBJBgNVBAsMQkFtYXpvbiBXZWIgU2VydmljZXMgTz1BbWF6b24uY29t
IEluYy4gTD1TZWF0dGxlIFNUPVdhc2hpbmd0b24gQz1VUzAeFw0xOTAyMTgxMTEx
NTFaFw00OTEyMzEyMzU5NTlaMB4xHDAaBgNVBAMME0FXUyBJb1QgQ2VydGlmaWNh
dGUwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDFhVNrOPnq3TKNA7X7
LCeQGjD6ec0QxuHs+eRFGJtIgv5G0mM/KYOEw4n0D5JjRlbu22aSi9NAf5BejRDo
1HtcrNUvhmvNDSli4NX+yqA34PgQNUXs1d5GOxO9K/uXlhjBuBO7O54zTtfhN1ZP
8Gd20XC/X6TOZZ0ltx7pmvPH64ZKNW461k7MTGZZEqWlYndsjizNODePlc5rAUuE
09A2YEourhkKRiOAHQk3DLtuyfR/wQlLmcnhOlDDKA8IzeYE+/v4eFkQtc5lzLu4
Q+5z3ozsnPOJ3lnMiQ2NXUUBuMJY+v6Bb98GyOgY7ay4Uzl2lw8ScwyHhawiq2TF
od6VAgMBAAGjYDBeMB8GA1UdIwQYMBaAFPE649b3WKoPCg9tIjnyZnYpxA4AMB0G
A1UdDgQWBBSbBSBvOks3f4oaWDqEpWTGLXpjUjAMBgNVHRMBAf8EAjAAMA4GA1Ud
DwEB/wQEAwIHgDANBgkqhkiG9w0BAQsFAAOCAQEAYMb71nZbaz7VAlpWkZDJju5r
ckyb4CPpH9KQykVnXuYN1sDGdNPFnF8wnpowRRiuQn5cpZHwc0HNchFZ4caIyksF
favEXD/GPpNnkCK+03YDTsTG69cXFbrWDKwyqj6hKR6dkQxBkic5feq2rQ+bav3H
Rju8UbuvrtE6aJTz3IjPqVvbbzTcMeZLY4iEpA2W6Tky8u0Kttz6mIb9GAW5dcx8
s+6ONomRML/FzJQYIqrIjSPIhGwyN6TVzBr9AAmm+d+9mfsv9xIf9H1z1AkyJv9O
CLaxqeNOEtxriSV49bf1e1ccNu8qRsnHH973U/KheqOZ5ch1D9+4c+0RlI3eNg==
-----END CERTIFICATE-----\0
```
3. Create the NVS input file certs.csv
```key,type,encoding,value    
certs,namespace,,  
certificate_pem_crt,file,string,~/esp/esp-idf/examples/protocols/aws_iot/certs/certificate_pem_crt
private_pem_key,file,string,~/esp/esp-idf/examples/protocols/aws_iot/certs/private_pem_key
```
4. Run the NVS Partition Generator command below.

### Code to reproduce this issue
```python ~/esp/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py ~/esp/esp-idf/examples/protocols/aws_iot/certs.csv certs.bin 8192```

## Debug Logs
```Traceback (most recent call last):
  File "/home/andrew/esp/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 775, in <module>
    main()
  File "/home/andrew/esp/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 770, in main
    nvs_part_gen(input_filename, output_filename, input_size, key_gen, is_encrypt_data, key_file, version_no)
  File "/home/andrew/esp/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 692, in nvs_part_gen
    write_entry(nvs_obj, row["key"], row["type"], row["encoding"], row["value"])
KeyError: 'value'
```

asweeney99, 2019-02-28:
@mahavirj Any thoughts on this?

Alvin1Zhang, 2019-02-28:
@asweeney99 Hi, we have started to look into this issue, @shivani-tipnis please help share if any updates, thanks.

shivani-tipnis, 2019-02-28:
@asweeney99 Hi, 
1. Could you try this on master branch, there have been a couple of fixes on it. 
2. Could you provide the absolute path in the certs.csv for the certs value instead of the tilde sign (`~`).

asweeney99, 2019-02-28:
Superb. Thank you. Removing the space worked.

asweeney99, 2019-02-28:
Maybe the documentation could be updated? Thanks again.
[image]

closed 2019-02-28 by asweeney99

mahavirj, 2019-03-01:
@asweeney99 Interim, please note that you may not null terminate certificate file (as you had mentioned in step 2), it will be taken care of being `string` type in NVS configuration.