espressif/esp-idf #3107: http OTA Bug

Andy001847, 2019-02-28:
When ESP32 at OTA process, it would occur some mistake. The message is "ASSERT_PARAM(-218959118 0), in arch_main.c at line 325", but file arch_main.c cannot find in esp-idf.

TianHao-Espressif, 2019-03-01:
Hi , @Andy001847 ，
Thank you for bug reporting. This problem is due to the memory heap pool of bluetooth controller is not enough。

could you describe something more, such as following:
1. What's OTA based on ?  BLE or BR/EDR or WiFi?
2. Does your project use bluetooth and wifi coexistence?
3. When OTA is precessing, what is bluetooth or Wifi doing?

Andy001847, 2019-03-01:
My project base on BLE and WiFi coexistence, the ble is scanning device when ota process, also, the project used mqtt.

closed 2019-03-01 by Andy001847