espressif/esp-idf #3108: firmware Crashing  occasionally while using gpio_intr_disable or pcnt_counter_pause as   Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)

sijokwork, 2019-02-28:
## Environment

- Development Kit:      [ESP32-Custom Board]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-S]
- IDF version (run ``git describe --tags`` to find it): 3.1
    
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [external 5V]

## Problem Description

firmware Crashing  occasionally while using gpio_intr_disable or pcnt_counter_pause as 
 Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
//Detailed problem description goes here.

I am working on medium level a project which enabled timer and  gpio interrupt /pulse counter, here i used init timer to start gpio interrupt / pulse counter for 0.4 sec  and stop ,but  once in a while the firmware crashing with .  "Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)"  error.Timer handler uses IRAM_ATTR attribute too.
i have searched related crash issues but could not solve mine.

with gpio interrupt error 
-------------------------------

Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x400f3734  PS      : 0x00060034  A0      : 0x40082808  A1      : 0x3ffb1038  
0x400f3734: gpio_intr_disable at esp-idf/components/driver/gpio.c:416
0x40082808: _xt_medint2 at /esp-idf/components/freertos/xtensa_vectors.S:1185

A2      : 0x00000001  A3      : 0x00000001  A4      : 0x3ffb6048  A5      : 0x40087230  
0x40087230: _frxt_int_enter at /esp-idf/components/freertos/portasm.S:119
A6      : 0x00000000  A7      : 0x13080248  A8      : 0x80083200  A9      : 0x3ffe3b20  
A10     : 0x00000019  A11     : 0x00000002  A12     : 0x3ffb4ecc  A13     : 0x00063004  
A14     : 0x3ffb4ecc  A15     : 0xfff9cffb  SAR     : 0x0000001d  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x400855a1  EPC2    : 0x400838f8  EPC3    : 0x00000000  EPC4    : 0x400f3734
0x400855a1: xQueueGenericReceive at /esp-idf/components/freertos/queue.c:2037

0x400838f8: spi_flash_op_block_func at /esp-idf/components/spi_flash/cache_utils.c:82 (discriminator 1)

0x400f3734: gpio_intr_disable at /esp-idf/components/driver/gpio.c:416

Backtrace: 0x400f3734:0x3ffb1038
0x400f3734: gpio_intr_disable at /esp-idf/components/driver/gpio.c:416

with pulse counter error 
---------------------------------------

Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x400f40fc  PS      : 0x00060034  A0      : 0x4008277c  A1      : 0x3ffb1028  
0x400f40fc: pcnt_counter_pause at /home/workspace/esp-idf/components/driver/pcnt.c:281

0x4008277c: _xt_lowint1 at /home/workspace/esp-idf/components/freertos/xtensa_vectors.S:1105

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x3ff42000  A5      : 0x400871ac  
0x400871ac: _frxt_int_enter at /home/workspace/esp-idf/components/freertos/portasm.S:119

A6      : 0x000420bc  A7      : 0x3ffc2f00  A8      : 0x80083204  A9      : 0x3ffb4efc  
A10     : 0x00000000  A11     : 0xff000000  A12     : 0x8008b5d1  A13     : 0x3ffcf560  
A14     : 0x00000000  A15     : 0x3ffcf590  SAR     : 0x00000011  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x4008b69e  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400f40fc
0x4008b69e: esp_rom_spiflash_read_data at /home/workspace/esp-idf/components/spi_flash/spi_flash_rom_patch.c:412

0x400f40fc: pcnt_counter_pause at /home/workspace/esp-idf/components/driver/pcnt.c:281

Backtrace: 0x400f40fc:0x3ffb1028
0x400f40fc: pcnt_counter_pause at /home/workspace/esp-idf/components/driver/pcnt.c:281

### Expected Behavior
no crashing 
### Actual Behavior
crashing once in a while 

## Debug Logs

Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x400f40fc  PS      : 0x00060034  A0      : 0x4008277c  A1      : 0x3ffb1028  
0x400f40fc: pcnt_counter_pause at /home/workspace/esp-idf/components/driver/pcnt.c:281

0x4008277c: _xt_lowint1 at /home/workspace/esp-idf/components/freertos/xtensa_vectors.S:1105

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x3ff42000  A5      : 0x400871ac  
0x400871ac: _frxt_int_enter at /home/workspace/esp-idf/components/freertos/portasm.S:119

A6      : 0x000420bc  A7      : 0x3ffc2f00  A8      : 0x80083204  A9      : 0x3ffb4efc  
A10     : 0x00000000  A11     : 0xff000000  A12     : 0x8008b5d1  A13     : 0x3ffcf560  
A14     : 0x00000000  A15     : 0x3ffcf590  SAR     : 0x00000011  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x4008b69e  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400f40fc
0x4008b69e: esp_rom_spiflash_read_data at /home/workspace/esp-idf/components/spi_flash/spi_flash_rom_patch.c:412

0x400f40fc: pcnt_counter_pause at /home/workspace/esp-idf/components/driver/pcnt.c:281

Backtrace: 0x400f40fc:0x3ffb1028
0x400f40fc: pcnt_counter_pause at /home/workspace/esp-idf/components/driver/pcnt.c:281

----------------------------------------------

I (75313) [MQTT_APP_TASK]: mqtt_client_connect check
D (77363) RTC_MODULE: adc mode takes adc1 lock.
D (77363) RTC_MODULE: returns adc1 lock.
D (77363) RTC_MODULE: adc mode takes adc1 lock.
D (77363) RTC_MODULE: returns adc1 lock.
Raw: 1398	Voltage: 408mV
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x400f3734  PS      : 0x00060034  A0      : 0x40082808  A1      : 0x3ffb1038  
0x400f3734: gpio_intr_disable at  /home/workspace/esp-idf/components/driver/gpio.c:416

0x40082808: _xt_medint2 at  /home/workspace//esp-idf/components/freertos/xtensa_vectors.S:1185

A2      : 0x00000001  A3      : 0x00000001  A4      : 0x3ffb6048  A5      : 0x40087230  
0x40087230: _frxt_int_enter at  /home/workspace//esp-idf/components/freertos/portasm.S:119

A6      : 0x00000000  A7      : 0x13080248  A8      : 0x80083200  A9      : 0x3ffe3b20  
A10     : 0x00000019  A11     : 0x00000002  A12     : 0x3ffb4ecc  A13     : 0x00063004  
A14     : 0x3ffb4ecc  A15     : 0xfff9cffb  SAR     : 0x0000001d  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x400855a1  EPC2    : 0x400838f8  EPC3    : 0x00000000  EPC4    : 0x400f3734
0x400855a1: xQueueGenericReceive at  /home/workspace//esp-idf/components/freertos/queue.c:2037

0x400838f8: spi_flash_op_block_func at  /home/workspace//esp-idf/components/spi_flash/cache_utils.c:82 (discriminator 1)

0x400f3734: gpio_intr_disable at  /home/workspace//esp-idf/components/driver/gpio.c:416

Backtrace: 0x400f3734:0x3ffb1038
0x400f3734: gpio_intr_disable at  /home/workspace//esp-idf/components/driver/gpio.c:416

koobest, 2019-04-01:
Hi,@sijokwork
Since some data is stored in flash(e.g. GPIO_IS_VALID_GPIO() -> GPIO_PIN_MUX_REG), we cannot access this data when the cache is disabled. Therefore, you'd better not call this API in the ISR context.

Alvin1Zhang, 2019-07-15:
@sijokwork Thanks for reporting the issue, will close this issue due to few updates. Feel free to reopen if you still have the issue, thanks.

closed 2019-07-15 by Alvin1Zhang